ghdl/ghdl #2752: Can't find entity or configuration when running cocotb simulation

EmilioPeJu, 2024-09-17:
**Description**
I am testing a very simple entity with cocotb (v1.9.1) and it can't run the simulation when using ghdl built from the master branch(v5.0.0-dev), however, if I use an old ghdl version (v3.0.0) from the fedora repository, it works as expected.

The DUT is:
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity eth_rx is
    port (
        clk_i : in std_logic;
        rx_tlast_i : in std_logic;
        rx_tdata_i : in std_logic_vector(7 downto 0);
        rx_tvalid_i : in std_logic;
        recv_stb_o : out std_logic
    );
end;

architecture rtl of eth_rx is
begin
    process (clk_i)
    begin
        if rising_edge(clk_i) then
            recv_stb_o <= '0';
            if rx_tlast_i then
                recv_stb_o <= '1';
            end if;
        end if;
    end process;
end;
```
The cocotb test is:
```python
#!/usr/bin/env python
import cocotb

from cocotb.clock import Clock
from cocotb.triggers import RisingEdge
from cocotb.runner import get_runner

@cocotb.test()
async def recv_strobe_initial_value(dut):
    cocotb.start_soon(Clock(dut.clk_i, 1, 'ns').start(start_high=False))
    await RisingEdge(dut.clk_i)
    assert dut.recv_stb_o.value == 'U'
    await RisingEdge(dut.clk_i)
    assert dut.recv_stb_o.value == '0'

def test_eth_rx():
    runner = get_runner('ghdl')
    runner.build(sources=['./eth_rx.vhd'],
                 hdl_toplevel='eth_rx', build_args=['--std=08'], always=True)
    runner.test(hdl_toplevel='eth_rx', test_module='test_eth_rx')

if __name__ == '__main__':
    test_eth_rx()
```
This is the output when the test is run:
```sh
$ python test_eth_rx.py
/dls/science/users/hir12111/src/mlatency2/test/test_eth_rx.py:6: UserWarning: Python runners and associated APIs are an experimental feature and subject to change.
  from cocotb.runner import get_runner
INFO: Running command ghdl -i --work=top --std=08 eth_rx.vhd in directory sim_build
INFO: Running command ghdl -m --work=top --std=08 eth_rx in directory sim_build
INFO: Running command ghdl -r --work=top eth_rx --vpi=~/venv/lib/python3.12/site-packages/cocotb/libs/libcocotbvpi_ghdl.so in directory sim_build
ghdl:error: cannot find entity or configuration eth_rx
Process 'ghdl' terminated with error 1
```

Unfortunately I couldn't isolate what's special about this entity, I have run many tests for other entities and didn't get this problem.

**Expected behaviour**
GHDL should find the entity. This is the output using ghdl v3.0.0
```sh
analyze eth_rx.vhd
elaborate eth_rx
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:79   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on GHDL version 3.0.0 (tarball) [Dunoon edition]
     0.00ns INFO     cocotb                             Running tests with cocotb v1.9.1 from /usr/local/lib64/python3.12/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1726575171
     0.00ns INFO     cocotb.regression                  pytest not found, install it to enable better AssertionError messages
     0.00ns INFO     cocotb.regression                  Found test test_eth_rx.recv_strobe_initial_value
     0.00ns INFO     cocotb.regression                  running recv_strobe_initial_value (1/1)
     1.50ns INFO     cocotb.regression                  recv_strobe_initial_value passed
     1.50ns INFO     cocotb.regression                  ***********************************************************************************************
                                                        ** TEST                                   STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
                                                        ***********************************************************************************************
                                                        ** test_eth_rx.recv_strobe_initial_value   PASS           1.50           0.00       1883.11  **
                                                        ***********************************************************************************************
                                                        ** TESTS=1 PASS=1 FAIL=0 SKIP=0                           1.50           0.02         80.79  **
                                                        ***********************************************************************************************
                                                        
INFO: Running command ghdl -i --work=top --std=08 eth_rx.vhd in directory sim_build
INFO: Running command ghdl -m --work=top --std=08 eth_rx in directory sim_build
INFO: Running command ghdl -r --work=top eth_rx --vpi=/usr/local/lib64/python3.12/site-packages/cocotb/libs/libcocotbvpi_ghdl.so in directory sim_build
INFO: Results file: results.xml
```

**How to reproduce?**
- Clone, build and install ghdl from the master branch
- Copy the vhdl code above to a file called `eth_rx.vhd`
- Copy the python code above to a file called `test_eth_rx.py`
- `pip install cocotb` if needed
- Run the python script: `python test_eth_rx.py`

tgingold, 2024-09-18:
Option `--std=08` must be specified to `ghdl -r`.

EmilioPeJu, 2024-09-18:
That worked, thanks a lot for you help, I fixed it by calling the test function specifying the `test_args` argument

closed 2024-09-18 by EmilioPeJu