ghdl/ghdl #2763: Memory Leak?

markuspg, 2024-09-26:
**Description**

I'm working on a project. Long running (> 1h) test runs crash because my machine runs out of memory (32 GB). Roughly estimated a GHDL process is eating about 5 to 10 MB per second.

The issue is only with `Run_Mode : Run_Mode_Kind := Run_Jit;` in `ghdlrun.adb`. If I revert [0e3ac92a](https://github.com/ghdl/ghdl/commit/0e3ac92a434756b2e5c9a54215ed9f476bf73b5c) the memory usage stays constant. The issue is the same with `v4.1.0` or current `master` (`92a4fc97`). I found this because with `v3.0.0` there is no issue, so a `git bisect` led me to commit [0e3ac92a](https://github.com/ghdl/ghdl/commit/0e3ac92a434756b2e5c9a54215ed9f476bf73b5c).

I'm puzzled, because the issue seems to go back to September 2023 but no one else seems to have come across this.

**Expected behaviour**

Before the commit introducing the issue the memory usage of a GHDL process on my machine is constant at about 23 MB.

**How to reproduce?**

So far I couldn't reproduce this with an exemplary testbench apart from my codebase, I'm sorry. Do you know any bigger example or test projects I could try to run?

**Context**
Please, provide the following information:

- OS: Fedora 40 on x86_64
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `v4.1.0` and `92a4fc97` (_mcode_ backend)

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**

I'm running my tests with VUnit (which I faultily suspected at first (see [VUnit issue 1062](https://github.com/VUnit/vunit/pull/1062)).

Unfortunately I'm more a C/C++ guy than Ada or VHDL. If you have any hints how I can support in tracking this down I'd do so gladly.

tgingold, 2024-09-26:
The commit you spot is not very useful as it changes the elaboration mode.

Is it possible to file your project, when if it is large ?

tgingold, 2024-09-27:
Another possibility is to run with valgrind and to display the leaks.

markuspg, 2024-10-03:
My coworker tested the issue on his machine (Ubuntu) additionally. With 4.1.0 and the _LLVM_ backend everything is fine, with the _mcode_ backend he's leaking memory too.

Unfortunately I cannot file the project. On some of our testbenches we see the issue and on others not. I'm trying to find the pivotal component to maybe create a reduced example.

I made two test runs on current _master_ (`6a9c70dd`). With `ghdl_not_leaking.txt` I was using `Run_Mode` `Run_Elab_Jit` and with `ghdl_leaking.txt` `Run_Jit`.

[ghdl_not_leaking.txt](https://github.com/user-attachments/files/17247685/ghdl_not_leaking.txt)
[ghdl_leaking.txt](https://github.com/user-attachments/files/17247686/ghdl_leaking.txt)

tgingold, 2024-10-04:
I haven't spotted an important source of leaks.
Are you using `new` or `textio` ?  If you do not deallocate, this creates leaks in vhdl.

markuspg, 2026-10-31:
I lost access to the project causing the memory leak due to a job change and could not reproduce the issue anymore with the current versions of GHDL and VUnit. I'm closing this issue for this reason.

closed 2026-10-31 by markuspg

Paebbels, 2026-10-31:
@markuspg sorry to hear this. Also thanks for closing this stale issue.