ghdl/ghdl #2764: Error "cannot associate individually with open" is annoyingly restrictive

Araneidae, 2024-09-27:
Here is an entity with unconstrained signals:
```vhdl
entity test is
    port (
        input_i : in std_ulogic_vector;
        output1_o : out std_ulogic_vector;
        output2_o : out std_ulogic_vector
    );
end;
```
(The body is immaterial; for sake of argument let it be `output1_o <= input_i; output2_o <= input_i`.)

Let's say I have signals `my_input, my_output : std_ulogic` and I want to use `test` with `output2_o` unused, then I'd like to write this:
```vhdl
thing : entity work.test port map (
    input_i => my_input,
    output1_o => my_output,
    output2_o(0) => open
);
```
With GHDL this fails with the error
```
error: cannot associate individually with open
```
Ok, I get it, this is in the language standard specification ... but there isn't a clean workaround.  I can't just write:
```vhdl
thing : entity work.test port map (
    input_i => my_input,
    output1_o => my_output,
    output2_o => open
);
```
because of course this now fails with the perfectly reasonable error
```
error: unconstrained port "output2_o" must be connected
```
In the end the only workaround is to declare and assign to dummy signals ... or to rewrite the entity to use explicit signal ranges, which I'd also rather not do.

I raise this because both ModelSim and Vivado accept the first construction without demur, it'd be nice to have the option of doing this with GHDL.

I am using
```
$ ghdl --version
GHDL 5.0.0-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 12.4.0
 static elaboration, mcode JIT code generator
```
with command line options
```
 -a --std=08 -frelaxed -Wno-hide -Wno-specs 
```
(I might raise issues regarding the two `-Wno-` options if that would be helpful...)

Paebbels, 2024-09-27:
All ports must be constrained at latest at instantiation. By using open, no constraints are given to the output port.

Araneidae, 2024-09-27:
Well, in Vivado and ModelSim the elaborator is able to constrain the output port from the explicitly given range.  In GHDL there seems to be an adjacent bug reported in issue #2765.

closed 2024-09-29 by tgingold

Araneidae, 2024-09-30:
Much appreciated.

commit cf00e853 (2024-09-29): Allow open individual association in relaxed mode

Fix #2764