ghdl/ghdl #2766: Explain better what `--vendor-library=...` does

filmil, 2024-09-28:
**Description**
At  https://ghdl.github.io/ghdl/using/Synthesis.html#cmdoption-ghdl-vendor-library, the `--vendor-library=NAME` option is described as:  "Any unit from library NAME is a black box." I suppose this description is trivially true, but the issue here is that reading it I have no idea what that actually means for the synthesized output.  Is there any chance to explain what this is supposed to do, because at least in combination with `--out=verilog`, it seems to do nothing.

**Expected behaviour**
I would have expected that a declaration like:

`ibuf_inst: ibufds ... ` 

gets synthesized as a reference to some unknown `ibufds`. Insetad, it is fully instantiated.  I did have to substitute a definition for `ibufds` to get the synthesis to work, since if not all components can be instantiated, the compilation may fail.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: ent.vhd
library unisim;
use unisim.vcomponents.all;

entity ent is
end entity;

architecture a of ent is
begin

  ibuf_inst: ibufds generic map(...) port map (...);

end;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --synth --out=verilog ent
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2024-09-29:
The reproducer is incomplete and does not use the `--vendor-library=` option.

filmil, 2024-09-29:
Excuse me, are you saying that it is not possible to improve the docs on a command line option without a repro case? Do I get that right?

tgingold, 2024-09-29:
I need to understand why the documentation is not clear.  And the best way is to know how the documentation is currently understood.

filmil, 2024-09-29:
ok, I will try to provide a repro.

tgingold, 2024-09-29:
The most important part is how are you using the option.

filmil, 2024-09-29:
Code here: https://github.com/filmil/bugreports/tree/main/ghdl-vendor-library

Output here:

First, this:
```
[nix-shell:~/tmp/bugreports/ghdl-vendor-library]$ make v
ghdl -a --work=unisim fake_unisim.vhdl
ghdl -a --work=work file.vhdl
ghdl --synth --out=verilog --work=work --vendor-library=unisim demo
module ibufds_df5af4b053d1d36dd66fd8d62c2f9b4f1a3f74a7
  (input  i,
   input  ib,
   output o);
  assign o = i;
endmodule

module demo
  (input  i,
   input  ib,
   output o);
  wire c_o;
  assign o = c_o;
  /* file.vhdl:17:9  */
  ibufds_df5af4b053d1d36dd66fd8d62c2f9b4f1a3f74a7 c (
    .i(i),
    .ib(ib),
    .o(c_o));
endmodule
```

Then this:
```
[nix-shell:~/tmp/bugreports/ghdl-vendor-library]$ make v2
ghdl -a --work=unisim fake_unisim.vhdl
ghdl -a --work=work file.vhdl
ghdl --synth --out=verilog --work=work demo
module ibufds_df5af4b053d1d36dd66fd8d62c2f9b4f1a3f74a7
  (input  i,
   input  ib,
   output o);
  assign o = i;
endmodule

module demo
  (input  i,
   input  ib,
   output o);
  wire c_o;
  assign o = c_o;
  /* file.vhdl:17:9  */
  ibufds_df5af4b053d1d36dd66fd8d62c2f9b4f1a3f74a7 c (
    .i(i),
    .ib(ib),
    .o(c_o));
endmodule
```

Note the `make v` invocation uses `--vendor-library=unisim`, but `make v2` does not.

Outputs are identical. Hence my question: what is `--vendor-library=unisim` supposed to be doing, if it does not cause the outputs of these two invocations to be different?

I would have expected something like:

```
module demo
  (input  i,
   input  ib,
   output o);
  wire c_o;
  assign o = c_o;
  /* file.vhdl:17:9  */
  ibufds c (
    .i(i),
    .ib(ib),
    .o(c_o));
endmodule
```

Where the instantiation of ibufds would have been omitted. 

This would have allowed me to actually use a vendor library in synthesis so that it can put the `ibufds` in its rightful spot and synthesize the circuit. This, however does not happen, so I have to do guesswork to figure out how to produce an instance of `ibufds_df5af4b053d1d36dd66fd8d62c2f9b4f1a3f74a7`, which is supposed to instantiate a device primitive.

The instructions at: https://ghdl.github.io/ghdl/using/Synthesis.html#cmdoption-ghdl-vendor-library say:

```
--vendor-library=NAME
Any unit from library NAME is a black box.

Example:

$ ghdl --synth --std=08 --vendor-library=vendorlib [library.]top_unit [arch]
```

With the outputs of `make v` and `make v2` being identical, and the text "Any unit from library NAME is a black box." not explaining why the two should be identical, I am at a loss as to what `--vendor-library` is supposed to do. 

Hence the bug report.

filmil, 2024-09-29:
I also tried adding `--std=08` to all relevant command lines to match the website example, without any effect.

```

[nix-shell:~/tmp/bugreports/ghdl-vendor-library]$ make v
ghdl -a --std=08 --work=unisim fake_unisim.vhdl
ghdl -a --std=08 --work=work file.vhdl
ghdl --synth --std=08 --out=verilog --work=work --vendor-library=unisim demo
module ibufds_df5af4b053d1d36dd66fd8d62c2f9b4f1a3f74a7
  (input  i,
   input  ib,
   output o);
  assign o = i;
endmodule

module demo
  (input  i,
   input  ib,
   output o);
  wire c_o;
  assign o = c_o;
  /* file.vhdl:17:9  */
  ibufds_df5af4b053d1d36dd66fd8d62c2f9b4f1a3f74a7 c (
    .i(i),
    .ib(ib),
    .o(c_o));
endmodule

[nix-shell:~/tmp/bugreports/ghdl-vendor-library]$ make v2
ghdl --synth --std=08 --out=verilog --work=work demo
module ibufds_df5af4b053d1d36dd66fd8d62c2f9b4f1a3f74a7
  (input  i,
   input  ib,
   output o);
  assign o = i;
endmodule

module demo
  (input  i,
   input  ib,
   output o);
  wire c_o;
  assign o = c_o;
  /* file.vhdl:17:9  */
  ibufds_df5af4b053d1d36dd66fd8d62c2f9b4f1a3f74a7 c (
    .i(i),
    .ib(ib),
    .o(c_o));
endmodule

[nix-shell:~/tmp/bugreports/ghdl-vendor-library]$ 
```

* Could it be that I am using the option wrong. Yes.
* Does the documentation actually tell me how to use it correctly? Not that I could see.

HTH,
F

tgingold, 2024-09-29:
This is not the output expected,  so there might be a bug.  Let me try to fix it.

tgingold, 2024-09-29:
In the initial issue, you mentioned component instantiations while in the reproducer you are using direct entity instantiation.   Which one are you using ?

filmil, 2024-09-29:
Indeed, I was imprecise with terminology. I prefer direct instantiation as it ends up being way less verbose than component instantiation.

filmil, 2024-09-29:
2nd attempt: https://github.com/filmil/bugreports/tree/main/ghdl-vendor-library-attempt-2

Using component instantiation, instead of direct instantiation. This is more promising, see below.

```
[nix-shell:~/tmp/bugreports/ghdl-vendor-library-attempt-2]$ make v
ghdl -a --std=08 --work=unisim fake_unisim.vhdl
ghdl -a --std=08 --work=work file.vhdl
ghdl --synth --std=08 --out=verilog --work=work --vendor-library=unisim demo
file.vhdl:30:9:warning: instance "c" of component "ibufds" is not bound [-Wbinding]
        c: ibufds
        ^
file.vhdl:14:14:warning: (in default configuration of demo(demo))
module demo
  (input  i,
   input  ib,
   output o);
  wire c_o;
  assign o = c_o;
  /* file.vhdl:30:9  */
  ibufds #(
    .DIFF_TERM(1'b1),
    .IOSTANDARD(48'b010011000101011001000100010100110011001000110101))
    c (
    .i(i),
    .ib(ib),
    .o(c_o));
endmodule

[nix-shell:~/tmp/bugreports/ghdl-vendor-library-attempt-2]$ make v2
ghdl --synth --std=08 --out=verilog --work=work demo
file.vhdl:30:9:warning: instance "c" of component "ibufds" is not bound [-Wbinding]
        c: ibufds
        ^
file.vhdl:14:14:warning: (in default configuration of demo(demo))
module demo
  (input  i,
   input  ib,
   output o);
  wire c_o;
  assign o = c_o;
  /* file.vhdl:30:9  */
  ibufds #(
    .DIFF_TERM(1'b1),
    .IOSTANDARD(48'b010011000101011001000100010100110011001000110101))
    c (
    .i(i),
    .ib(ib),
    .o(c_o));
endmodule
```

* This output is more like it. Though it took me a double-take to realize that the message printed is a *warning* not an error.
* Documentation for `--vendor-library=....` says: "Any unit from library NAME is a black box.".   Can you help me understand how the word "any" in that sentence excludes direct instantiation?  (Or, it could be that it somehow does. But in that case I would contend that this particular intention of the option is too hidden for mere mortals.) Perhaps being more explicit about what the option does, and doesn't do, could help?

tgingold, 2024-09-29:
So, this now also applies to direct instantiations.  And the documentation is improved to explain how this option changes warnings.

closed 2024-09-29 by tgingold

filmil, 2024-09-29:
Thanks for fixing this. Synthesis-to-verilog enables some really interesting things, many thanks for making ghdl happen.

tgingold, 2024-09-30:
You're welcome.

commit e7e4c79e (2024-09-29): synth-vhdl_insts: entities of vendor libs are black boxes

Fix #2766