ghdl/ghdl #2767: GHDL and GUIX: ghdl_llvm hardlinked to /lib64

csantosb, 2024-09-30:
**Description**

I try to package GHDL under GUIX package manager.

When I build GHDL I get the output of

    strings -a ghdl1-llvm | grep ld-linux

as 

    //gnu/store/zvlp3n8iwa1svxmwv4q22pv1pb1c9pjq-glibc-2.39/lib/ld-linux-x86-64.so.2

which is the expected behaviour; while

    strings -a ghdl_llvm | grep ld-linux

gives 

    /lib64/ld-linux-x86-64.so.2

With `ghdl_llvm` hardlinked to `/lib64` for some reason, and `/lib64` is not present under GUIX.

**Expected behaviour**

I would expect that both `ghdl1-llvm` and `ghdl_llvm` are linked against

    //gnu/store/zvlp3n8iwa1svxmwv4q22pv1pb1c9pjq-glibc-2.39/lib/ld-linux-x86-64.so.2

**How to reproduce?**

```
# clone repo
git clone --depth=1 https://github.com/ghdl/ghdl.git ghdl-src
cd ghdl-src

# create a shell container with dependencies
guix shell -C gnat coreutils make clang grep diffutils sed bash glibc

# set necessary env variables
# libraries inside the container are at $LIBRARY_PATH
export LDFLAGS=-no-pie
export LD_LIBRARY_PATH=$LIBRARY_PATH

# build 
mkdir build
cd build
../configure --with-llvm-config
make -j $(nproc)

# check
strings -a ghdl1-llvm | grep ld-linux
# returns: //gnu/store/zvlp3n8iwa1svxmwv4q22pv1pb1c9pjq-glibc-2.39/lib/ld-linux-x86-64.so.2

strings -a ghdl_llvm | grep ld-linux
# returns: /lib64/ld-linux-x86-64.so.2
```

**Context**

Please, provide the following information:

- OS: GUIX shell container

**Additional context**

Full guix package is here: https://git.sr.ht/~csantosb/ip.channel/tree/master/item/ip_test.scm

The error obtained when building the package is to be reproduced inside a guix shell container.

tgingold, 2024-09-30:
I have never used guix so I am not sure I am the best placed to investigate.

How is the dynamic linker specified ?  Are there any wrappers ?

You should start by inspecting the build log and search for the difference between the two builds.

csantosb, 2024-10-01:
>lun. 30 sept. 2024 at 11:10, tgingold ***@***.***> wrote:

> I have never used guix so I am not sure I am the best placed to investigate.

To my understanding, Guix is not that relevant here, other that not
including /lib64 in its package building context.

Guix just provides an isolated context where to build ghdl. I guess same
behaviour occurs under any other environment. For example, under
archlinux, I get

    > strings -a ghdl_llvm | grep ld-linux
    /lib64/ld-linux-x86-64.so.2

    > strings -a ghdl1-lvm | grep ld-linux
    /lib64/ld-linux-x86-64.so.2

> How is the dynamic linker specified ? Are there any wrappers ?

Just usual

    ../configure --with-llvm-config
    make -j $(nproc)

> You should start by inspecting the build log and search for the difference between the two
> builds.

The (attached) build log is a bit opaque to me, I’m afraid. I see no difference on
how the two binary products are produced.

starting phase `separate-from-pid1'
build process now running as PID 10
phase `separate-from-pid1' succeeded after 0.0 seconds
starting phase `set-SOURCE-DATE-EPOCH'
phase `set-SOURCE-DATE-EPOCH' succeeded after 0.0 seconds
starting phase `set-paths'
environment variable `PATH' set to `/gnu/store/rnicfsxlymi8alp6xbngqwzjfyc985mk-gnat-14.2.0-1/bin:/gnu/store/c9hb3j8ajfavvgciy00nrr4v29963z19-clang-13.0.1/bin:/gnu/store/j5zgzgsmbjgywr67r86h1n6s4qiabv5q-tar-1.34/bin:/gnu/store/0r2fx1lr1h2i3cl1x5fw4s4ly95qspya-gzip-1.13/bin:/gnu/store/56aq6sdx35f7rsxq8jq9ypafk0dhd3p3-bzip2-1.0.8/bin:/gnu/store/544ga5n6nq8y2961ihigra6n6a0r99nn-file-5.45/bin:/gnu/store/swmflx2bmqpf0drm167pxfpgyr7ckcam-diffutils-3.10/bin:/gnu/store/4jhlsg65s1zx90gnnfmaax52i8prnl45-patch-2.7.6/bin:/gnu/store/b5c62i99dsfa5j0j3f8g2j1xdi6fih22-findutils-4.9.0/bin:/gnu/store/vrl2r0ps24k1a6yap2chjylv5rhbndvr-gawk-5.3.0/bin:/gnu/store/3f0fjq25n2kwcmi0zyrf216wa6g8xpjw-zstd-1.5.2/bin:/gnu/store/3i4fr8riqw9azh1p4mqggl5vdz23kyr8-sed-4.8/bin:/gnu/store/7k8b93779dqpwcg2qjdvnf4nl43jv7hf-grep-3.11/bin:/gnu/store/fbaw0sb21gv02qq7gs9wg5y5wlpdgzih-xz-5.4.5/bin:/gnu/store/7h0mnlwalw23j9jmvz5n1i5mqkgb4d06-coreutils-9.1/bin:/gnu/store/lc6r1mh0jlcwixjziiw1zsx1my478rv1-make-4.4.1/bin:/gnu/store/3jhfhxdf6v5ms10x5zmnl166dh3yhbr1-bash-minimal-5.1.16/bin:/gnu/store/3k6lyifyg024lk3gncsgf2gzjzvc70gv-ld-wrapper-0/bin:/gnu/store/ylgkrq9j1gyfig2y66srmgsbyh9w20d9-binutils-2.41/bin:/gnu/store/86fc8bi3mciljxz7c79jx8zr4wsx7xw8-gcc-11.4.0/bin:/gnu/store/zvlp3n8iwa1svxmwv4q22pv1pb1c9pjq-glibc-2.39/bin:/gnu/store/zvlp3n8iwa1svxmwv4q22pv1pb1c9pjq-glibc-2.39/sbin:/gnu/store/8mfcc1h0a8v81mp58msbdnrjqcsm2pyz-clang-runtime-13.0.1/bin:/gnu/store/46vsk7fkcrf1c91j7d8w9i8pyvhw2w0m-llvm-13.0.1/bin'
environment variable `C_INCLUDE_PATH' set to `/gnu/store/c9hb3j8ajfavvgciy00nrr4v29963z19-clang-13.0.1/include:/gnu/store/56aq6sdx35f7rsxq8jq9ypafk0dhd3p3-bzip2-1.0.8/include:/gnu/store/544ga5n6nq8y2961ihigra6n6a0r99nn-file-5.45/include:/gnu/store/vrl2r0ps24k1a6yap2chjylv5rhbndvr-gawk-5.3.0/include:/gnu/store/fbaw0sb21gv02qq7gs9wg5y5wlpdgzih-xz-5.4.5/include:/gnu/store/lc6r1mh0jlcwixjziiw1zsx1my478rv1-make-4.4.1/include:/gnu/store/ylgkrq9j1gyfig2y66srmgsbyh9w20d9-binutils-2.41/include:/gnu/store/86fc8bi3mciljxz7c79jx8zr4wsx7xw8-gcc-11.4.0/include:/gnu/store/zvlp3n8iwa1svxmwv4q22pv1pb1c9pjq-glibc-2.39/include:/gnu/store/8mfcc1h0a8v81mp58msbdnrjqcsm2pyz-clang-runtime-13.0.1/include:/gnu/store/46vsk7fkcrf1c91j7d8w9i8pyvhw2w0m-llvm-13.0.1/include:/gnu/store/svghsdn1x3fyili8ij791nn25zba9bpf-linux-libre-headers-5.15.49/include:/gnu/store/1prv14v6jfnzzg7szm57690b7fr6sx33-zlib-1.3/include'
environment variable `CPLUS_INCLUDE_PATH' set to `/gnu/store/c9hb3j8ajfavvgciy00nrr4v29963z19-clang-13.0.1/include:/gnu/store/56aq6sdx35f7rsxq8jq9ypafk0dhd3p3-bzip2-1.0.8/include:/gnu/store/544ga5n6nq8y2961ihigra6n6a0r99nn-file-5.45/include:/gnu/store/vrl2r0ps24k1a6yap2chjylv5rhbndvr-gawk-5.3.0/include:/gnu/store/fbaw0sb21gv02qq7gs9wg5y5wlpdgzih-xz-5.4.5/include:/gnu/store/lc6r1mh0jlcwixjziiw1zsx1my478rv1-make-4.4.1/include:/gnu/store/ylgkrq9j1gyfig2y66srmgsbyh9w20d9-binutils-2.41/include:/gnu/store/86fc8bi3mciljxz7c79jx8zr4wsx7xw8-gcc-11.4.0/include/c++:/gnu/store/86fc8bi3mciljxz7c79jx8zr4wsx7xw8-gcc-11.4.0/include:/gnu/store/zvlp3n8iwa1svxmwv4q22pv1pb1c9pjq-glibc-2.39/include:/gnu/store/8mfcc1h0a8v81mp58msbdnrjqcsm2pyz-clang-runtime-13.0.1/include:/gnu/store/46vsk7fkcrf1c91j7d8w9i8pyvhw2w0m-llvm-13.0.1/include:/gnu/store/svghsdn1x3fyili8ij791nn25zba9bpf-linux-libre-headers-5.15.49/include:/gnu/store/1prv14v6jfnzzg7szm57690b7fr6sx33-zlib-1.3/include'
environment variable `OBJC_INCLUDE_PATH' set to `/gnu/store/c9hb3j8ajfavvgciy00nrr4v29963z19-clang-13.0.1/include:/gnu/store/56aq6sdx35f7rsxq8jq9ypafk0dhd3p3-bzip2-1.0.8/include:/gnu/store/544ga5n6nq8y2961ihigra6n6a0r99nn-file-5.45/include:/gnu/store/vrl2r0ps24k1a6yap2chjylv5rhbndvr-gawk-5.3.0/include:/gnu/store/fbaw0sb21gv02qq7gs9wg5y5wlpdgzih-xz-5.4.5/include:/gnu/store/lc6r1mh0jlcwixjziiw1zsx1my478rv1-make-4.4.1/include:/gnu/store/ylgkrq9j1gyfig2y66srmgsbyh9w20d9-binutils-2.41/include:/gnu/store/86fc8bi3mciljxz7c79jx8zr4wsx7xw8-gcc-11.4.0/include:/gnu/store/zvlp3n8iwa1svxmwv4q22pv1pb1c9pjq-glibc-2.39/include:/gnu/store/8mfcc1h0a8v81mp58msbdnrjqcsm2pyz-clang-runtime-13.0.1/include:/gnu/store/46vsk7fkcrf1c91j7d8w9i8pyvhw2w0m-llvm-13.0.1/include:/gnu/store/svghsdn1x3fyili8ij791nn25zba9bpf-linux-libre-headers-5.15.49/include:/gnu/store/1prv14v6jfnzzg7szm57690b7fr6sx33-zlib-1.3/include'
[... cut ...]
`/gnu/store/bgxl6mq4p4qblspx57zkbb68xm2qw3y9-ghdl-3.0-2.0afa60d-checkout/src/synth/synth-verilog_exprs.ads' -> `./src/synth/synth-verilog_exprs.ads'
`/gnu/store/bgxl6mq4p4qblspx57zkbb68xm2qw3y9-ghdl-3.0-2.0afa60d-checkout/src/synth/synth-vhdl_eval.adb' -> `./src/synth/synth-vhdl_eval.adb'
`/gnu/store/bgxl6mq4p4qblspx57zkbb68xm2qw3y9-ghdl-3.0-2.0afa60d-checkout/src/synth/synth-flags.ads' -> `./src/synth/synth-flags.ads'
`/gnu/store/bgxl6mq4p4qblspx57zkbb68xm2qw3y9-ghdl-3.0-2.0afa60d-checkout/src/synth/netlis

tgingold, 2024-10-02:
Well, guix is very relevant because it modifies the sources and install wrappers.

ghdl1-llvm is linked with clang++ while ghdl_llvm is linked with gcc.
Can you rebuild using `make LDFLAGS="-no-pie -v -v"` to see how the linker is invoked ?

csantosb, 2024-10-02:
Thanks a lot for your support. It would be great if we could build and distribute
ghdl under guix.

> Can you rebuild using make LDFLAGS="-no-pie -v -v" to see how the linker is invoked ?

Sure, here you have:

    https://paste.sr.ht/~csantosb/9ed9a08013bc72e33210f9488a099b9d0c134915

civodul, 2024-10-02:
Hello!

@csantosb The `gcc` package in Guix is modified so that programs it links use `/gnu/store/.../lib/ld-linux.so` as they dynamic linker rather than the default `/lib64/ld-linux.so`. It [does so](https://git.savannah.gnu.org/cgit/guix.git/tree/gnu/packages/gcc.scm#n259) by changing the value of the `GLIBC_DYNAMIC_LINKER` constant under `gcc/config`.

Most likely, the `gnat` program you're running does not have this modification.  Thus, when running `gnat -o exe ...`, it creates `exe` with `/lib64/ld-linux.so` as its dynamic linker, which won't work in a pure Guix environment.

A short-term hack would be to somehow invoke `patchelf --set-interpreter` on the `ghdl_llvm` executable.

A longer-term solution would be to have a `gnat` package built from source: it would inherit from `gcc` and have the pre-built `gnat` you're currently using as an input.

(This is really a Guix packaging issue, not GHDL's fault at all. :-))

csantosb, 2024-10-02:
>mer. 02 oct. 2024 at 06:36, Ludovic Courtès ***@***.***> wrote:

> Most likely, the gnat program you're running does not have this modification.

I’m using a patchelf version of gnat (to my understanding).

So:

    ldd /gnu/store/4wcrg2x649fx66j4qg8c0hhf2zi31k4y-profile/bin/gnat

gives:

    linux-vdso.so.1 (0x00007ffc3a1a7000)
    libdl.so.2 => /gnu/store/zvlp3n8iwa1svxmwv4q22pv1pb1c9pjq-glibc-2.39/lib/libdl.so.2 (0x00007bbc0adfa000)
    libm.so.6 => /gnu/store/zvlp3n8iwa1svxmwv4q22pv1pb1c9pjq-glibc-2.39/lib/libm.so.6 (0x00007bbc0ad1a000)
    libc.so.6 => /gnu/store/zvlp3n8iwa1svxmwv4q22pv1pb1c9pjq-glibc-2.39/lib/libc.so.6 (0x00007bbc0ab3c000)
    /gnu/store/zvlp3n8iwa1svxmwv4q22pv1pb1c9pjq-glibc-2.39/lib/ld-linux-x86-64.so.2 =>
    /gnu/store/sz2byfjk3jmzadp8qasz77qawxr1q820-glibc-2.39/lib/ld-linux-x86-64.so.2

and

    strings -a /gnu/store/4wcrg2x649fx66j4qg8c0hhf2zi31k4y-profile/bin/gnat | grep ld-linux

gives

    /gnu/store/zvlp3n8iwa1svxmwv4q22pv1pb1c9pjq-glibc-2.39/lib/ld-linux-x86-64.so.2
    ld-linux-x86-64.so.2

Similar behaviour with gnatmake.

For as long as I know, this is the correct behaviour.

> Thus, when
> running gnat -o exe ..., it creates exe with /lib64/ld-linux.so as its dynamic linker,
> which won't work in a pure Guix environment.

Anyway, the problematic binary is ghdl_llvm, linked with gcc.

This is reason of this issue: I don’t fully understand where the /lib64
comes from exactly, and if this is what is expected from ghdl side.

csantosb, 2024-10-02:
Finally, in the case of ghdl_llvm:

***@***.*** ~/ghdl-src/build [env]$ ldd ghdl_llvm

        linux-vdso.so.1 (0x00007fffaeb23000)
        libm.so.6 => /gnu/store/4wcrg2x649fx66j4qg8c0hhf2zi31k4y-profile/lib/libm.so.6 (0x000
        libc.so.6 => /gnu/store/4wcrg2x649fx66j4qg8c0hhf2zi31k4y-profile/lib/libc.so.6 (0x000
        /lib64/ld-linux-x86-64.so.2 => /gnu/store/sz2byfjk3jmzadp8qasz77qawxr1q820-glibc-2.39

***@***.*** ~/ghdl-src/build [env]$ ./ghdl_llvm

sh: ./ghdl_llvm: No such file or directory

***@***.*** ~/ghdl-src/build [env]$ strings -a ghdl_llvm | grep ld-linux

/lib64/ld-linux-x86-64.so.2
ld-linux-x86-64.so.2

Whereas in the case of ghdl1-llvm:

***@***.*** ~/ghdl-src/build [env]$ ldd ghdl1-llvm

        linux-vdso.so.1 (0x00007ffdcdb81000)
        libLLVM.so.18.1 => /gnu/store/4wcrg2x649fx66j4qg8c0hhf2zi31k4y-profile/lib/libLLVM.so.18.1 (0x000071abbfe00000)
        libm.so.6 => /gnu/store/4wcrg2x649fx66j4qg8c0hhf2zi31k4y-profile/lib/libm.so.6 (0x000071abc7e60000)
        libstdc++.so.6 => /gnu/store/4wcrg2x649fx66j4qg8c0hhf2zi31k4y-profile/lib64/libstdc++.so.6 (0x000071abbfa00000)
        libgcc_s.so.1 => /gnu/store/4wcrg2x649fx66j4qg8c0hhf2zi31k4y-profile/lib64/libgcc_s.so.1 (0x000071abc7e32000)
        libc.so.6 => /gnu/store/4wcrg2x649fx66j4qg8c0hhf2zi31k4y-profile/lib/libc.so.6 (0x000071abbf822000)
        //gnu/store/zvlp3n8iwa1svxmwv4q22pv1pb1c9pjq-glibc-2.39/lib/ld-linux-x86-64.so.2 => /gnu/store/sz2byfjk3jmzadp8qasz77qawxr1q820-glibc-2.39/lib/ld-linux-x86-64.so.2 (0x000071abc7f42000)
        libffi.so.8 => /gnu/store/s6iqwc5sqjrk76kzslqc1n1wlcvfyqkw-libffi-3.4.4/lib/libffi.so.8 (0x000071abc7e20000)
        libz.so.1 => /gnu/store/4wcrg2x649fx66j4qg8c0hhf2zi31k4y-profile/lib/libz.so.1 (0x000071abc7e02000)

***@***.*** ~/ghdl-src/build [env]$ strings -a ghdl1-llvm | grep ld-linux

//gnu/store/zvlp3n8iwa1svxmwv4q22pv1pb1c9pjq-glibc-2.39/lib/ld-linux-x86-64.so.2
ld-linux-x86-64.so.2

tgingold, 2024-10-02:
@civodul, @csantosb  it might be interesting to check if `gnat` is correctly configured by building a simple hello world program.  But as a linker, it uses `gcc`.

Probably, a work-around is to use `LDFLAGS=--LINK=clang++`

civodul, 2024-10-03:
> Anyway, the problematic binary is ghdl_llvm, linked with gcc.

@csantosb Could it be that the `gcc` command being used here also comes from your `gnat` package (made from pre-built binaries) and not from the `gcc` package?

If that is the case, one way to work around it might be to add `gcc` to `native-inputs`, which should have the effect of putting that `gcc` first in `$PATH` (better double-check though by looking at the environment variable dump at the beginning of the build log).

csantosb, 2024-10-03:
> @civodul, @csantosb it might be interesting to check if gnat is correctly configured by
> building a simple hello world program.

Good point. I get this:

/gnu/store/y6jwyhsk28r5wns06rr1g9v7zr71zapi-gnat-14.2.0-1/bin/../lib/gcc/x86_64-pc-linux-gnu/14.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find crt1.o: No such file or directory

So most probably, the problem comes from the way I’m building gnat.

So not related to ghdl, sorry.

tgingold, 2024-10-04:
So I close this issue, but do not hesitate to re-open it if you have a related question.

closed 2024-10-04 by tgingold

csantosb, 2024-10-14:
Issue is fixed now. It is then possible to install and use ghdl (in its clang variant) from a dedicated guix channel.

    https://git.sr.ht/~csantosb/guix.channel-electronics

Thanks for this great tool !

tgingold, 2024-10-14:
Great!