ghdl/ghdl #2768: GHDL fuzzing revived

avelure, 2024-10-01:
I have some interest in looking into fuzzing of VHDL parsers and since @sudden6 provided a nice startingpoint in #2076, I have been playing around a bit with GHDL as a target.
I have in the process found a few new crashes. 
[crashes_24-10-01.zip](https://github.com/user-attachments/files/17212204/crashes_24-10-01.zip)

The zip will have some duplicates as there are several code paths to the same crash.
The following seem to be the unique once based on the error message alone.
```
./0665f547d08f1194038c1ceab233f59dee306de978d0a8b526d301255c0ecbcc:1:8: an identifier is expected instead of 'function'
C:\msys64\opt\ghdl_llvm\bin\ghdl.exe: error limit reached

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r152.g20e0931ec) [Dunoon edition]
Compiled with GNAT Version: 14.2.0
Target: x86_64-w64-mingw32
C:\proj\inventas\fuzz\ghdl\ghdl_afl_fuzzing\afl_work\crashes_new\
Command line:
C:\msys64\opt\ghdl_llvm\bin\ghdl.exe -s --std=08 -fsynopsys -frelaxed -fno-diagnostics-show-option -fno-caret-diagnostics -fpsl --mb-comments -fmax-errors=1 -Wno-library -Wno-port -Wno-pragma -Wno-specs -Wno-runtime-error -Wno-missing-wait -Wno-shared -Wno-hide -Wno-pure -Wno-analyze-assert -Wno-attribute -Wno-useless -Wno-conformance -Wno-elaboration ./0665f547d08f1194038c1ceab233f59dee306de978d0a8b526d301255c0ecbcc
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : vhdl-nodes.adb:862
******************************************************************
./06cdd8f55a5f69cd11c26771b9490d770826b94a1c78d8133d59fce0c94cb07c:1:37: no declaration for "e"
C:\msys64\opt\ghdl_llvm\bin\ghdl.exe: error limit reached
is_expression: cannot handle IIR_KIND_INTERFACE_TYPE_DECLARATION (./06cdd8f55a5f69cd11c26771b9490d770826b94a1c78d8133d59fce0c94cb07c:1:26)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r152.g20e0931ec) [Dunoon edition]
Compiled with GNAT Version: 14.2.0
Target: x86_64-w64-mingw32
C:\proj\inventas\fuzz\ghdl\ghdl_afl_fuzzing\afl_work\crashes_new\
Command line:
C:\msys64\opt\ghdl_llvm\bin\ghdl.exe -s --std=08 -fsynopsys -frelaxed -fno-diagnostics-show-option -fno-caret-diagnostics -fpsl --mb-comments -fmax-errors=1 -Wno-library -Wno-port -Wno-pragma -Wno-specs -Wno-runtime-error -Wno-missing-wait -Wno-shared -Wno-hide -Wno-pure -Wno-analyze-assert -Wno-attribute -Wno-useless -Wno-conformance -Wno-elaboration ./06cdd8f55a5f69cd11c26771b9490d770826b94a1c78d8133d59fce0c94cb07c
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
******************************************************************
./16cb82d1bb82c64e97cb23aeca26d23287e21c2f3fb5d8416479470e714c84a8:1:8: an identifier is expected instead of 'function'
C:\msys64\opt\ghdl_llvm\bin\ghdl.exe: error limit reached

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r152.g20e0931ec) [Dunoon edition]
Compiled with GNAT Version: 14.2.0
Target: x86_64-w64-mingw32
C:\proj\inventas\fuzz\ghdl\ghdl_afl_fuzzing\afl_work\crashes_new\
Command line:
C:\msys64\opt\ghdl_llvm\bin\ghdl.exe -s --std=08 -fsynopsys -frelaxed -fno-diagnostics-show-option -fno-caret-diagnostics -fpsl --mb-comments -fmax-errors=1 -Wno-library -Wno-port -Wno-pragma -Wno-specs -Wno-runtime-error -Wno-missing-wait -Wno-shared -Wno-hide -Wno-pure -Wno-analyze-assert -Wno-attribute -Wno-useless -Wno-conformance -Wno-elaboration ./16cb82d1bb82c64e97cb23aeca26d23287e21c2f3fb5d8416479470e714c84a8
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : no field Identifier
******************************************************************
./57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed:6:10: signal "s" is not visible here
C:\msys64\opt\ghdl_llvm\bin\ghdl.exe: error limit reached

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r152.g20e0931ec) [Dunoon edition]
Compiled with GNAT Version: 14.2.0
Target: x86_64-w64-mingw32
C:\proj\inventas\fuzz\ghdl\ghdl_afl_fuzzing\afl_work\crashes_new\
Command line:
C:\msys64\opt\ghdl_llvm\bin\ghdl.exe -s --std=08 -fsynopsys -frelaxed -fno-diagnostics-show-option -fno-caret-diagnostics -fpsl --mb-comments -fmax-errors=1 -Wno-library -Wno-port -Wno-pragma -Wno-specs -Wno-runtime-error -Wno-missing-wait -Wno-shared -Wno-hide -Wno-pure -Wno-analyze-assert -Wno-attribute -Wno-useless -Wno-conformance -Wno-elaboration ./57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed
Exception STORAGE_ERROR raised
Exception information:
raised STORAGE_ERROR : EXCEPTION_STACK_OVERFLOW
******************************************************************
```

I need to accumulate a better corpus of quality VHDL code and a dictionary of words and the quality of the bugs should improve.
There are also several hangs reported (where the parsing takes more than 5s) that may be interesting to look at.

closed 2024-10-02 by tgingold

avelure, 2024-10-04:
Thanks.
I see case 2 and 4 in my overview still cause a crash, I assume they are so obscure they are not worth fixing?

tgingold, 2024-10-05:
The tests with:
./06cdd8f55a5f69cd11c26771b9490d770826b94a1c78d8133d59fce0c94cb07c
or
./57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed
?

I cannot make those crash (on linux).

avelure, 2024-10-05:
Yes, I tested with both msys and Ubuntu 22.04 in docker. On Ubuntu I use the mcode backend.
I can check again on Monday.

avelure, 2024-10-07:
57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed crashes on latest master on msys ucrt64 and ubuntu 22.04 (in docker on windows) with both llvm and mcode backend.

```
$ /opt/ghdl_llvm/bin/ghdl -a --std=08 57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed
57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed:6:10:error: signal "s" is not visible here
signal s:s(0 to 0);signal s:s(0 downto 0);begin
         ^
57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed:6:27:error: identifier "s" already used for a declaration
signal s:s(0 to 0);signal s:s(0 downto 0);begin
                          ^
57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed:6:8:error: previous declaration: signal "s"
57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed:6:29:error: a type mark must denote a type or a subtype
signal s:s(0 to 0);signal s:s(0 downto 0);begin
                            ^
57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed:6:29:error: (type mark denotes signal "s")
57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed:9:1:error: no declaration for "w"
w;report to_string(0.0,50000000);end process;t(0);end architecture;
^

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r253.geeab69c29) [Dunoon edition]
Compiled with GNAT Version: 14.2.0
Target: x86_64-w64-mingw32
C:\proj\inventas\fuzz\ghdl\ghdl_afl_fuzzing\afl_work\crashes_new_2\
Command line:
C:\msys64\opt\ghdl_llvm\bin\ghdl1-llvm.exe --std=08 -PC:\msys64\opt\ghdl_llvm\lib/ghdl\ieee\v08\ -PC:\msys64\opt\ghdl_llvm\lib/ghdl\ -c -o 57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed.o 57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed
Exception STORAGE_ERROR raised
Exception information:
raised STORAGE_ERROR : EXCEPTION_STACK_OVERFLOW
Load address: 0x7ff7236a0000
Call stack traceback locations:
0x7ff723b16ac8 0x7ff80c9f51dd 0x7ff80c96e864 0x7ff80c9f41cc 0x7ff723b23014 0x2faf1be 0x2385a1b0436 0x7ff7239569f1 0x7ff72393ac80 0x7ff723940030 0x7ff7239450a6 0x7ff72394518e 0x7ff72394523d 0x7ff7239457ac 0x7ff7239fca65 0x7ff7239ea078 0x7ff7239eb488 0x7ff723a01a85 0x7ff723a01acb 0x7ff723a0366c 0x7ff723a03747 0x7ff723a04492 0x7ff723a04c58 0x7ff723a04e8d 0x7ff723a04ff1 0x7ff7239c39b7 0x7ff7239caefe 0x7ff72399f921 0x7ff723ad129e 0x7ff723ad2c76 0x7ff723ad3ff0 0x7ff7236a3224 0x7ff7236a12e7 0x7ff7236a13f4 0x7ff80a85257b 0x7ff80c9aaf26
******************************************************************
C:\msys64\opt\ghdl_llvm\bin\ghdl.exe:error: compilation error
```

The **vhdl-errors.adb:30** error now happens on 22c24209078be30d8c79bb4002f1da2bd376743896a30e21267619c84b55b612 instead, so maybe it is implementation specific.
```
$ /opt/ghdl_llvm/bin/ghdl -a --std=08 22c24209078be30d8c79bb4002f1da2bd376743896a30e21267619c84b55b612
22c24209078be30d8c79bb4002f1da2bd376743896a30e21267619c84b55b612:1:45:error: signal "r" is not visible here
entity g is generic(type e);type t;signal r:r range 0 to 0;signal r:r range 0 to 0;signal m:n;begin o<='0'when(0)and 0 else'0'when(0)and'0';m(0);process(a)begin if e(0)then
                                            ^
22c24209078be30d8c79bb4002f1da2bd376743896a30e21267619c84b55b612:1:67:error: identifier "r" already used for a declaration
entity g is generic(type e);type t;signal r:r range 0 to 0;signal r:r range 0 to 0;signal m:n;begin o<='0'when(0)and 0 else'0'when(0)and'0';m(0);process(a)begin if e(0)then
                                                                  ^
22c24209078be30d8c79bb4002f1da2bd376743896a30e21267619c84b55b612:1:43:error: previous declaration: signal "r"
22c24209078be30d8c79bb4002f1da2bd376743896a30e21267619c84b55b612:1:69:error: a type mark must denote a type or a subtype
entity g is generic(type e);type t;signal r:r range 0 to 0;signal r:r range 0 to 0;signal m:n;begin o<='0'when(0)and 0 else'0'when(0)and'0';m(0);process(a)begin if e(0)then
                                                                    ^
22c24209078be30d8c79bb4002f1da2bd376743896a30e21267619c84b55b612:1:69:error: (type mark denotes signal "r")
22c24209078be30d8c79bb4002f1da2bd376743896a30e21267619c84b55b612:1:93:error: no declaration for "n"
entity g is generic(type e);type t;signal r:r range 0 to 0;signal r:r range 0 to 0;signal m:n;begin o<='0'when(0)and 0 else'0'when(0)and'0';m(0);process(a)begin if e(0)then
                                                                                            ^
22c24209078be30d8c79bb4002f1da2bd376743896a30e21267619c84b55b612:1:101:error: signal assignment forbidden in entity
entity g is generic(type e);type t;signal r:r range 0 to 0;signal r:r range 0 to 0;signal m:n;begin o<='0'when(0)and 0 else'0'when(0)and'0';m(0);process(a)begin if e(0)then
                                                                                                    ^
22c24209078be30d8c79bb4002f1da2bd376743896a30e21267619c84b55b612:1:101:error: no declaration for "o"
entity g is generic(type e);type t;signal r:r range 0 to 0;signal r:r range 0 to 0;signal m:n;begin o<='0'when(0)and 0 else'0'when(0)and'0';m(0);process(a)begin if e(0)then
                                                                                                    ^
22c24209078be30d8c79bb4002f1da2bd376743896a30e21267619c84b55b612:1:114:error: no function declarations for operator "and"
[... cut ...]
0x7ff7238642b0 0x7ff7239f83b0 0x7ff7239fbdb7 0x7ff7239e9fb4 0x7ff7239eb138 0x7ff7239ebcf8 0x7ff723a00a16 0x7ff723a03556 0x7ff723a03747 0x7ff723a04492 0x7ff723a04592 0x7ff723a04c38 0x7ff723a04e8d 0x7ff723a04ff1 0x7ff7239c355e 0x7ff7239caeef 0x7ff72399f921 0x7ff723ad129e 0x7ff723ad2c76 0x7ff723ad3ff0 0x7ff7236a3224 0x7ff7236a12e7 0x7ff7236a13f4 0x7ff80a85257b 0x7ff80c9aaf26
******************************************************************
C:\msys64\opt\ghdl_llvm\bin\ghdl.exe:error: compilation error
```

tgingold, 2024-10-07:
For the first one (57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed) I cannot reproduce on Linux.  But it might be related to the stack size.  Can you show the crash in docker ?

For the second one, I am testing a fix

avelure, 2024-10-08:
Here is the backtrace with mcode on ubuntu docker

```
Starting program: /usr/bin/ghdl -a --std=08 57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed:6:10:error: signal "s" is not visible here
signal s:s(0 to 0);signal s:s(0 downto 0);begin
         ^
57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed:6:27:error: identifier "s" already used for a declaration
signal s:s(0 to 0);signal s:s(0 downto 0);begin
                          ^
57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed:6:8:error: previous declaration: signal "s"
57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed:6:29:error: a type mark must denote a type or a subtype
signal s:s(0 to 0);signal s:s(0 downto 0);begin
                            ^
57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed:6:29:error: (type mark denotes signal "s")
57e7202889b50c3a95bd8cb03348f7e431c30eaea06a7b87ab67d316bae02bed:9:1:error: no declaration for "w"
w;report to_string(0.0,50000000);end process;t(0);end architecture;
^

Program received signal SIGSEGV, Segmentation fault.
0x0000555555933df2 in synth.vhdl_eval.eval_static_predefined_function_call ()
(gdb) bt
#0  0x0000555555933df2 in synth.vhdl_eval.eval_static_predefined_function_call ()
#1  0x0000555555916319 in vhdl.evaluation.eval_predefined_call ()
#2  0x000055555591bc57 in vhdl.evaluation.eval_dyadic_operator ()
#3  0x00005555559212bc in vhdl.evaluation.eval_static_expr_orig ()
#4  0x00005555559213ca in vhdl.evaluation.eval_static_expr ()
#5  0x0000555555921474 in vhdl.evaluation.eval_expr_keep_orig ()
#6  0x00005555559219f0 in vhdl.evaluation.eval_expr_if_static ()
#7  0x00005555559e173d in vhdl.sem_names.name_to_expression ()
#8  0x00005555559ce212 in vhdl.sem_expr.sem_expression_ov ()
#9  0x00005555559cf6fb in vhdl.sem_expr.sem_expression ()
#10 0x00005555559e71e0 in vhdl.sem_stmts.sem_report_expression ()
#11 0x00005555559e7223 in vhdl.sem_stmts.sem_report_statement ()
#12 0x00005555559e8f8d in vhdl.sem_stmts.sem_sequential_statements_internal ()
#13 0x00005555559e9073 in vhdl.sem_stmts.sem_sequential_statements ()
#14 0x00005555559e9e56 in vhdl.sem_stmts.sem_process_statement ()
#15 0x00005555559ea645 in vhdl.sem_stmts.sem_concurrent_statement ()
#16 0x00005555559ea89e in vhdl.sem_stmts.sem_concurrent_statement_chain ()
#17 0x00005555559eaa0b in vhdl.sem_stmts.sem_block ()
#18 0x00005555559a6ca5 in vhdl.sem.sem_architecture_body ()
#19 0x00005555559ae5f3 in vhdl.sem.semantic ()
#20 0x00005555559803e2 in vhdl.sem_lib.finish_compilation ()
#21 0x0000555555a0b14e in ghdlcomp.analyze_file ()
#22 0x0000555555a0b408 in ghdlcomp.perform_action ()
#23 0x00005555559f1ed4 in ghdlmain.main ()
#24 0x0000555555b21c6c in ghdl_jit ()
```
The version is the ZIP from the nightly build
```
root@bcb30cd8955a:/tmp/crashes_new# ghdl --version
GHDL 5.0.0-dev (4.1.0.r253.geeab69c29) [Dunoon edition]
 Compiled with GNAT Version: 10.5.0
 static elaboration, mcode JIT code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2024 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

commit de5a1b11 (2024-10-02): vhdl-parse.adb: avoid a crash after errors

Fix #2768