ghdl/ghdl #2771: Add to winget

ecstrema, 2024-10-03:
The current installation workflow for windows is not ideal (download an archive, unpack and add bin to path).

Since we already have a scoop version (see #2680), can we get a winget version? ([how to submit a package to winget](https://learn.microsoft.com/en-us/windows/package-manager/package/))

tgingold, 2024-10-04:
I have no idea on how winget works (as I am not a Windows user).  Feel free to create a pull request.

ecstrema, 2024-11-18:
I was looking into this, and realized that for this we'd first need a windows installer. Having no idea how to create windows installer, I've looked into it, and it looks like there unfortunately no automatic _give-me-your-files-i-make-an-installer_ solution.

InnoSetup seems the simplest option (I don't think anything fancy is needed for GHDL). A simple configuration file would be

```INI
; -- installer.iss --

[Setup]
AppName=GHDL
AppVersion=*****
DefaultDirName={pf}\GHDL
DefaultGroupName=GHDL
Compression=lzma2
SolidCompression=yes
OutputBaseFilename=GHDLInstaller
OutputDir=.
UninstallDisplayIcon={app}\GHDL.exe
UninstallDisplayName=GHDL

[Files]
Source: "bin/ghdl.exe"; DestDir: "{app}/bin"
Source: "bin/ghwdump.exe"; DestDir: "{app}/bin"

Source: "include/GHDL/libghw.h"; DestDir: "{app}/include"
; ...

[Icons]
Name: "{group}\GHDLInstaller"; Filename: "{app}\GHDL.exe"
Name: "{group}\Uninstall"; Filename: "{uninstallexe}"
```

Xcodo, 2024-11-19:
I believe in the new release of WinGet an portable executable inside a zip file is now supported. I can submit a request to the WinGet community repo.

I'm unclear with v4.1 which version is preferred, ghdl-MINGW32 or ghdl-UCRT64, for 64-bit systems. Are they both the same backend? @tgingold is there any documentation here? I can see a note on backend support in the NEWS file, but nothing in the docs pages.

ecstrema, 2024-11-20:
Having a real installer would be nice for other things than just winget though.

Dragon1573, 2024-12-11:
> I'm unclear with v4.1 which version is preferred, ghdl-MINGW32 or ghdl-UCRT64, for 64-bit systems.

For most packages in the `winget-pkgs`, I would prefer **MSVC** version if there exists one since `Microsoft.VCRedist.*` package series are already available in the repository and can be specified as `PackageDependencies` in manifests. Also **MSVC** is the recommended compiler collection for _Microsoft Windows_.

Should I add them both and split them into 2, such as `ghdl.ghdl.mingw32` and `ghdl.ghdl.ucrt64`?

> [!NOTE]
>
> [_RipGrep_](https://github.com/BurntSushi/ripgrep) also available in the repository as 2 identifiers - [`BurntSushi.ripgrep.MSVC`](https://github.com/microsoft/winget-pkgs/tree/master/manifests/b/BurntSushi/ripgrep/MSVC) and [`BurntSushi.ripgrep.GNU`](https://github.com/microsoft/winget-pkgs/tree/master/manifests/b/BurntSushi/ripgrep/GNU).

Xcodo, 2024-12-11:
The recommendation from the project gitter channel was to use UCRT64. There are different capabilities in the different packages.

Paebbels, 2024-12-11:
Nowadays, also VC++ should link against Universal C Runtime (UCRT), thus MSYS2 compiled code and VC++ compiled code link against the same libraries/symbols.

Until now, I haven't seen any issues using libghdl.dll build with MSYS2/UCRT64 and CPython on Windows (build by VC++).

Here you see Python 3.13 side-by-side with libghdl 5.0.0-dev:  
[image]

ecstrema, 2025-02-04:
Very nice! Unfortunately, a clean installation doesn't find the openieee libraries.

```
Analyzing /../virtual_fuse.vhdl
~\AppData\Local\Microsoft\WinGet\Links\ghdl.exe:warning: ieee library directory '/GHDL\lib/ghdl\ieee\v08\' not found  
~\AppData\Local\Microsoft\WinGet\Links\ghdl.exe:error: cannot find "std" library
```

This is because it runs the symlink created in the `~/AppData/Local/Microsoft/WinGet/Links/` folder. If I run it directly from the executable in `~/AppData/Local/Microsoft/WinGet/Packages/ghdl.ghdl.ucrt64_Microsoft.Winget.Source_8wekyb3d8bbwe/GHDL/bin` then it works perfectly fine.

Does GHDL change its working directory to `../../`?

There might be some configuration needed to tell winget to not create a link and instead add /bin to path.

Xcodo, 2025-02-24:
Good point. I actually thought this would be an issue, but the person who made the `winget` manifest didn't follow my instructions and I didn't check properly.

I understand how to resolve it with a change to the manifest and just created an example I can use to prove it. I'll try to get it changed.

Xcodo, 2025-03-19:
@ecstrema I fixed the library problem with v4.1.0.

I've also just added v5.0.1. The identifier has changed to `ghdl.ghdl.ucrt64.mcode`, so you might need to uninstall and reinstall rather than just upgrade the package.