ghdl/ghdl #2776: Synthesis: missing Verilog signal types

stnolting, 2024-10-07:
Hey there!

I am using the `synth` feature to convert a rather large design to a single Verilog file. Everything worked fine so far, but today I discovered a problem. I am not really sure yet if this is really caused by a GHDL bug or by my input VHDL code. 🤔

Anyway, here is the description:

This is the workflow that is failing: https://github.com/stnolting/neorv32-verilog/actions/runs/11222793796
The generated Verilog file can be downloaded as artifact ("debug"). In that file there are several syntax errors like this:

```verilog
  wire neorv32_cpu_cp_fpu_f2i_inst_n16990;
[31:0] neorv32_cpu_cp_fpu_f2i_inst_result_o;
[4:0] neorv32_cpu_cp_fpu_f2i_inst_flags_o;
neorv32_cpu_cp_fpu_f2i_inst_done_o;
  wire [1:0] n16998_o;
```

The first and the last signal declarations are fine but the three inner ones are missing an actual type (`wire`/`reg`).

This is the script I am using for "conversion": https://github.com/stnolting/neorv32-verilog/blob/main/src/convert.sh
These are the important parts of the script:

```bash
# Import and analyze sources
ghdl -i --std=08 --work=neorv32 --workdir=build -Pbuild $NEORV32_HOME/rtl/core/*.vhd $SRC_FOLDER/$TOP.vhd
ghdl -m --std=08 --work=neorv32 --workdir=build $TOP
# Synthesize Verilog
ghdl synth --std=08 --work=neorv32 --workdir=build -Pbuild --out=verilog $TOP > $SRC_FOLDER/$TOP.v
```

The failing GitHub worklfow uses your GitHub action:

```yml
    - name: '📦 Install GHDL'
      uses: ghdl/setup-ghdl-ci@nightly
```

Which currently results in this specific GHDL version:

```
GHDL 5.0.0-dev (4.1.0.r253.geeab69c29) [Dunoon edition]
 Compiled with GNAT Version: 9.4.0
 static elaboration, mcode JIT code generator
Written by Tristan Gingold.
```

On my local machine I am using an older version of GHDL which **does not produce this error at all** when fed with the **same** code:

```
GHDL 4.0.0-dev (3.0.0.r823.g9e2b6fc95) [Dunoon edition]
 Compiled with GNAT Version: 9.4.0
 static elaboration, mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Maybe you could have a look? I'll also check my VHDL code base.

Thanks in advance! :wink:

tgingold, 2024-10-08:
Should now be fixed.  Was a mistake in a previous change.

Thank you for the issue!

closed 2024-10-08 by tgingold

stnolting, 2024-10-08:
It works! Thanks for fixing this so fast!

commit d2ddeae5 (2024-10-08): netlists-disp_verilog.adb: fix previous patch

Fix #2776