ghdl/ghdl #2777: Crash when exporting the waveforms, access check failed

marzoul, 2024-10-08:
**Description**
GHDL crashes with error message `access check failed` when  exporting the waveforms with option `--wave=wave.ghw`
There is no crash when not using the option and the simulation finishes without issue.
The issue seems to happen at init of the simulation or elaboration, not during simulation itself.

**Expected behaviour**
GHDL is expected to run the simulation with no error even when exporting the waves.

**How to reproduce?**
The reproducer design is too large to be attached. Here is a public link to the archive :
https://cloud.univ-grenoble-alpes.fr/s/3QToQr5gxJb5aAi

To reproduce : type `make`.

I have not yet found what part of the design triggers the issue.

**Context**
Please, provide the following information:

- OS: Archlinux
- Origin: Built from sources: `commit eeab69c29b68eb3f7fd51e6337eedb924d7be829` done on Oct 6th, 2024.

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r253.geeab69c29) [Dunoon edition]
Compiled with GNAT Version: 14.2.1 20240910
Target: x86_64-pc-linux-gnu
/home/prostboa/Projets/nnawaq/simu/simu-nnawaq-axi4lite/bugreport-ghdl-null/
Command line:
ghdl -r --std=08 --workdir=work -Punisim nnawaq_axi4lite_bench --wave=wave.ghw
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-avhpi.adb:593 access check failed
******************************************************************
```

marzoul, 2024-10-08:
ghdl backend used : mcode (not tried other backends)

tgingold, 2024-10-08:
You'd better to use `--fst=` instead of `--wave=`.  The capability is now the same, but currently fst has a better support (with the mcode backend).

I need to rework the ghw dumper.

marzoul, 2024-10-08:
FST works, thank you for the advice !

closed 2024-11-04 by tgingold