ghdl/ghdl #2784: GHDL Bug preceded by dodgy index out of bounds message

Araneidae, 2024-10-22:
**Description**
The attached code generates an error of the form
```
ghdl:error: index (23743296) out of bounds (23743296
```
followed by a "GHDL Bug occurred" message.  Trivial variations of the code generate a variety of interesting alternative messages, including:
```
ghdl:error: declaration of a too large object (1662104 kB > --max-stack-alloc=128 kB)
```
and
```
ghdl:error: index (0) out of bounds (0 to 0) at ../../src/ieee2008/std_logic_1164-body.vhdl:1059
```

**Expected behaviour**
The original code was accepted by ModelSim without demur.

**How to reproduce?**
The code below generates the large index out of bounds failure:

```vhd :file: testbench.vhd
library ieee;                                                                   
use ieee.std_logic_1164.all;                                                    
use ieee.numeric_std.all;                                       
use std.textio.all;                                           
                        
entity testbench is
end entity;

architecture a of testbench is
    type vector_array is array(natural range <>) of std_ulogic_vector;          
    type vector_array_array is array(natural range <>) of vector_array;         
    subtype phy_data_t is vector_array(63 downto 0)(7 downto 0);                
    subtype phy_edc_t is vector_array(7 downto 0)(7 downto 0);                  
                                                                                
    signal clk : std_ulogic := '0';                                             
    signal data : phy_data_t;                                                   
--     signal edc : phy_edc_t;                                                  
                                                                                
    constant test_pattern : vector_array_array(open)(0 to 8)(7 downto 0) := (   
        (X"00", X"FF", X"00", X"00", X"00", X"00", X"00", X"00", X"00"),        
        (X"00", X"FF", X"00", X"00", X"00", X"00", X"00", X"00", X"FF"),        
        (X"00", X"FF", X"00", X"00", X"00", X"00", X"00", X"00", X"02")         
    );                                                                          
                                                                                
    subtype TEST_RANGE is natural range test_pattern'RANGE;                     
    signal test_index : TEST_RANGE := test_pattern'LOW;                         
                                                                                
begin                                                                           
    clk <= not clk after 5 ns;        
    process (clk)                                                               
        variable linebuffer : line;                                             
    begin                                                                       
        if rising_edge(clk) then                                                
            if test_index < test_pattern'HIGH then                              
                test_index <= test_index + 1;                                   
            else                                                                
                test_index <= test_pattern'LOW;                                 
            end if;                                                             
                                                                                
            write(linebuffer, to_hstring(test_pattern(test_index)(8)));     
            writeline(output, linebuffer);   
        end if;                                                                 
    end process;
end;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 testbench.vhd
ghdl -r --std=08 testbench
```

The `data` and `edc` declarations can be added or removed with the following permutations of error messages:

| Declarations | Result |
| --- | --- |
| both removed | declaration of a too large object (1662104 kB > --max-stack-alloc=128 kB), no Bug |
| both present | "index (0) out of bounds (0" followed by GHDL Bug |
| `data` only | "index (27667952) out of bounds ..." followed by GHDL Bug |
| `edc` only | index (0) out of bounds (0 to 0) at ../../src/ieee2008/std_logic_1164-body.vhdl:1059, no Bug |

**Context**
Please, provide the following information:

- OS: RHEL8
- Origin:
  - [ ] Built from sources: `commit SHA`  Sorry, @EmilioPeJu, can you fill this in please, again?!  Fairly recent, anyway.

The `GHDL Bug occurred` for the pasted code is included below:

```
ghdl:error: index (22953456) out of bounds (22953456
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 12.4.0
Target: x86_64-unknown-linux-gnu
/home/mga83/targetOS/IFC_1412/gddr6/sim/crc/do/ghdl-bug/
Command line:
/home/hir12111/.nix-profile/bin/ghdl -r --std=08 testbench
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-lib.adb:220 invalid data
******************************************************************
```

EmilioPeJu, 2024-10-22:
Commit "testsuite/gna: testcase for #2778"
SHA1: 99c84540ff2c096dfb5f72aab5bf761602d06b4a

tgingold, 2024-10-22:
I can reproduce the crash.  It works with the llvm backend, so it's an issue during elaboration.  I will fix it.

commit 7bb99359 (2024-10-25): simul-vhdl_compile: fix handling of constants

Fix #2784
closed 2024-10-25 by tgingold