ghdl/ghdl #2785: ghdl-mcode crash in simulation init

marzoul, 2024-10-22:
**Description**
GHDL (mcode version at least) crashes with segfault at init of a simulation.

**Expected behaviour**
The simulation should be performed successfully.

**How to reproduce?**
For now I am not able to share the code because it involves simulation sources from Vivado.
With dedicated work I can create a less optimized variant of the design that only uses my code, not immediate though.

In the meantime I can also provide more details from gdb, just tell me is there is some stuff you would like to know the status of.

**Context**
Please, provide the following information:

- OS: Archlinux
- Origin: built from sources (AUR package), commit `ec95e91300e6ec2c38abc5618aa5da3b0dd41160` done just today

**Additional context**

Even if it's not really useful, here is the command line that I used :
`ghdl -r --std=08 --workdir=work -P../unisim-vivado -P../unimacro-vivado --warn-no-hide     --ieee=synopsys -fexplicit -frelaxed -O2 <top>`

Here is the stacktrace of the crash :

```
Program received signal SIGSEGV, Segmentation fault.
0x000055555558e246 in grt__signals__create_nets () at /usr/src/debug/ghdl-mcode-git/ghdl/src/grt/grt-signals.adb:2967
2967	            Propag : Propag_Array (1 .. Last_Off);  --  := (others => 0);
(gdb) bt
#0  0x000055555558e246 in grt__signals__create_nets () at /usr/src/debug/ghdl-mcode-git/ghdl/src/grt/grt-signals.adb:2967
#1  grt__signals__order_all_signals () at /usr/src/debug/ghdl-mcode-git/ghdl/src/grt/grt-signals.adb:3126
#2  __ghdl_simulation_init () at /usr/src/debug/ghdl-mcode-git/ghdl/src/grt/grt-processes.adb:1182
#3  0x000055555556902e in __ghdl_run_through_longjump (func=0x55555558db80 <__ghdl_simulation_init>) at /usr/src/debug/ghdl-mcode-git/ghdl/./src/grt/config/jumps.c:234
#4  0x000055555589c134 in simul__main__simulation () at /usr/src/debug/ghdl-mcode-git/ghdl/src/simul/simul-main.adb:108
#5  0x000055555599d590 in simul__vhdl_compile__simulation () at /usr/src/debug/ghdl-mcode-git/ghdl/src/simul/simul-vhdl_compile.adb:2195
#6  ghdlrun__run () at /usr/src/debug/ghdl-mcode-git/ghdl/src/ghdldrv/ghdlrun.adb:412
#7  0x0000555555880538 in ghdlcomp__perform_action__3 (cmd=..., args=...) at /usr/src/debug/ghdl-mcode-git/ghdl/src/ghdldrv/ghdlcomp.adb:175
#8  0x0000555555876c69 in ghdlmain__main () at /usr/src/debug/ghdl-mcode-git/ghdl/src/ghdldrv/ghdlmain.adb:679
#9  0x0000555555564d7c in _ada_ghdl_jit () at /usr/src/debug/ghdl-mcode-git/ghdl/src/ghdldrv/ghdl_jit.adb:42
#10 main (argc=<optimized out>, argv=<optimized out>, envp=<optimized out>) at /usr/src/debug/ghdl-mcode-git/ghdl/b~ghdl_jit.adb:1245
#11 0x00007ffff7c8ae08 in ?? () from /usr/lib/libc.so.6
#12 0x00007ffff7c8aecc in __libc_start_main () from /usr/lib/libc.so.6
#13 0x0000555555564eb5 in _start ()
(gdb) 
```

Interestingly, resident RAM usage just reached 2.1GB when the crash happened (overflow of 32b signed integer perhaps ?).
After the crash I can see that it has jumped to 5.8 GB.

tgingold, 2024-10-23:
Probably you have a large amount of net and you get a stack overflow.
If this is really what happens, that's easy to fix.

Can you try to increase the stack size to confirm this hypothesis ?

marzoul, 2024-10-24:
The issue was the stack size, thank you for the advice.
My design is not that large, the final design size is 3000 LUT6 (for Xilinx 7-series) and I can easily scale that 100x.

So this memory alloc can be very large and is design-dependent.
Could it be changed to an alloc on the heap instead of the stack ?
To avoid many people reporting this crash under normal system configuration ;-)
Or at least an error message indicating a stack allocation failed, and the stack size should be increased as workaround.

marzoul, 2024-10-24:
There is something really weird, I can't make the simulation run for more than a few clock cycles (I want 100k), memory usage is permanently increasing. I killed it at 20 GB virtual / 12 GB resident. It used to be very light for earlier iterations of the design.
So maybe there is root issue elsewhere.
I will provide the design with more generic code to replace the Unisim instances.

marzoul, 2024-10-24:
Complete reproducer available at following link :
https://cloud.univ-grenoble-alpes.fr/s/6SA3e992dDqpNHX
Type `make`
Or `make analyze` followed by `make run`

marzoul, 2024-10-24:
Oooops correction : due to a bug in the chair-keyboard interface, the main design (generated) was not updated within the simulation context, so the simulation was launched on a much larger design than expected.
Simulation does run till the end of the 100k cycles, good :+1: 

How may signals are supposed to trigger the stack memory issue, then ?

tgingold, 2024-10-25:
You have a huge number of signals.  There is currently no signal optimization in GHDL and therefore it requires a lot of memory.  Probably, you should reduce the size of you design.

marzoul, 2024-10-25:
Yes that's fixed on my side now.

Is there a plan to fix the segfault in ghdl for large designs ?
Switching to allocation on heap, or at least an error message to indicate an allocation failed while a large number of nets and increasing the stack size is recommended.

commit a938ee87 (2024-10-26): grt-signals.adb: allocate propagation array on the heap

Fix #2785
closed 2024-10-26 by tgingold