ghdl/ghdl #2787: Support running multiple units/combining multiple waveform files into one

DUOLabs333, 2024-10-24:
**Is your feature request related to a problem? Please describe.**
I have a file where multiple entities have each have their own testbench, and I want to test them all at once. However, currently, you can only run a single entity in an invocation.

**Describe the solution you'd like**
It would be nice if `elab-run`/`run` could run multiple units at a time.

**Describe alternatives you've considered**
Alternatively, I could run each testbench separately, ave each run to a different waveform file, and then combine them into one file, where run i+1 would immediately succeed run i in the new file.

**Additional context**
N/A

Paebbels, 2024-10-24:
the requested feature is not directly the duty of a simulator and/or compiler. You're requesting a test execution framework. Have a look at [VUnit](https://github.com/VUnit/vunit) or [OSVVM](https://github.com/OSVVM/OSVVM-Scripts). Both offer such features. There are others available too.

tgingold, 2024-10-24:
The simplest is to create a wrapper that instantiate all the units you want to test.

closed 2024-10-25 by tgingold

DUOLabs333, 2024-10-25:
@tgingold I worked around this by just creating a wrapper. Thanks for the suggestion.