ghdl/ghdl #2793: ghdl:error: exec error

mayureshw, 2024-11-17:
**Description**
When analyzing a large vhdl program "exec error" occurs.

**Expected behaviour**
Completing analysis without exec error.

**How to reproduce?**

At this point I am not able to share the program. It is around 250,000 lines of generated code by some tool in VHDL 08 standard. Could program size have something to do with the issue? Looking for preliminary help and inputs at this stage.

**Context**
Please, provide the following information:

- OS: NetBSD 9.1
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: c3a459e

**Additional context**

ghdl runs for over 10 minutes before exiting with "exec error". The machine has 8GB RAM and in "top" the ghdl process size appeared around 4G. Prime facie it does not appear that it fell short of memory.

tgingold, 2024-11-17:
Which backend are you using (ghdl -v) ?

Please, file the  generated source code so that the issue could be investigated.

mayureshw, 2024-11-17:
```
$ ghdl -v
GHDL 5.0.0-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 10.3.0
 llvm 18.1.8 code generator
```
I'll try and check if the test case can be shared.

tgingold, 2024-11-17:
You should try with the mcode backend.

mayureshw, 2024-11-18:
mcode backend worked. Thanks.

closed 2024-11-18 by mayureshw

Paebbels, 2024-11-18:
@mayureshw while it works for you, the llvm backend has still a problem/bug. Is it possible to provide the VHDL design / VHDL example so we can investigate and reopen the issue?

mayureshw, 2024-11-18:
I am afraid, difficult to share it as-is. I'll try and work on creating an equivalent test case that can be shared. Please bear with me.

mayureshw, 2024-11-19:
I rebuilt with LLVM backend for commit f1de234 and the issue is not present now. (It was for an earlier commit c3a459e )