ghdl/ghdl #2795: NULL access dereferenced

mayureshw, 2024-11-18:
**Description**
With mcode backend on NetBSD, running ghdl draws NULL access dereferenced. Test case doesn't matter. Even the test case in the bug report template can be used.

**Expected behaviour**
No "NULL access dereferenced" should occur.

**How to reproduce?**

```vhd :file: hello.vhd
entity ent is
end entity;

architecture a of ent is
begin
  process begin
    report "Hello world" severity note;
    wait;
  end process;
end;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

**Context**
Please, provide the following information:

- OS: NetBSD 9.2 amd64
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: f1de234

**Additional context**
```
$ ghdl --elab-run ent
ghdl:error: NULL access dereferenced
ghdl:error: simulation failed
```

tgingold, 2024-11-18:
Unfortunately, I don't have a netbsd machine.

Did you try to run under gdb ?  That's probably something obvious.

tgingold, 2024-11-18:
You can also to build using llvm but in jit mode using `configure --with-llvm-jit ...`

mayureshw, 2024-11-19:
> Did you try to run under gdb ? That's probably something obvious.

Yes. But this being a 'normal exit' it does not yield and back trace.

mayureshw, 2024-11-19:
> You can also to build using llvm but in jit mode using `configure --with-llvm-jit ...`

I switched back to llvm.

Earlier I faced issue #2793 with llvm with commit c3a459e

However with commit f1de234 that issue is not faced. llvm backend is known to work on NetBSD. So I'll stick to it for now.

The issue remains that mcode backend dose not work on NetBSD. Would you like to keep this issue to be open for further investigation? It's at least not a show stopper for me now.

mayureshw, 2024-11-19:
I'd suggest to incorporate above two patches in the first post. Let me know if you'd like them to be submitted on a separate issue.

tgingold, 2024-11-19:
Yes, please submit a PR.

tgingold, 2024-11-19:
> > Did you try to run under gdb ? That's probably something obvious.
> 
> Yes. But this being a 'normal exit' it does not yield and back trace.

Well no, there is apparently a sigsegv.  If you catch the signal (or set a breakpoint on the handler), you should know more.

mayureshw, 2024-11-19:
> Yes, please submit a PR.

Submitted #2797

tgingold, 2024-11-24:
I am closing this issue, as it requires investigation on NetBSD. 

You can reopen it if you have details.

closed 2024-11-24 by tgingold