ghdl/ghdl #2798: CONSTRAINT_ERROR : synth-vhdl_context.adb:427 invalid data

mayureshw, 2024-11-19:
**Description**

Following error occurs during synthesis of VHDL 2008 code, for which RTL simulation has worked.

`CONSTRAINT_ERROR : synth-vhdl_context.adb:427 invalid data`

**Expected behaviour**
Synthesis should succeed.

**How to reproduce?**

Will work on creating a small test case. Please consider providing other inputs, such as quick workarounds, if any, in the meantime.

**Context**
Please, provide the following information:

- OS: NetBSD
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources:  f1de234

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (tarball) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64--netbsd
/my/working/directory
Command line:

Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : synth-vhdl_context.adb:427 invalid data

...
******************************************************************
```

**Additional context**

Nothing specific for now.

tgingold, 2024-11-19:
I need a little bit of context (like a reproducer :-)

Did you get an error message before the crash ?

mayureshw, 2024-11-20:
A large number of `":warning: no assignment for port"` before the crash (could be genuine and may not be a reason for the crash).
`
"ERROR: vhdl import failed."` after the crash.

Trying to get a small test case (daunting with a large design and no clue...).

mayureshw, 2024-11-20:
Placed a breakpoint at the line shown in the bug report. Attached is the stack trace of the break point being hit for the last time. Will it be of some help?

[trace.log](https://github.com/user-attachments/files/17824731/trace.log)

mayureshw, 2024-11-20:
I do have a recursive function in the design - could it matter? I have data widths of not more than 128 bits.

```
    -- onehot bit to integer by binary search. Assumption: at least 1 bit is set
    function onehot2int( v : std_logic_vector; efflow : integer := 0 ) return integer is
        variable mid : integer := v'length/2;
        variable upper : std_logic_vector( v'length - mid - 1 downto 0 ) := v( v'length - 1 downto mid );
        variable lower : std_logic_vector( mid - 1 downto 0 ) := v( mid - 1 downto 0 );
    begin
        if v'length = 1 then
            return efflow;
        else
            if or_reduce(upper) = '1' then
                return onehot2int(upper,efflow+mid);
            else
                return onehot2int(lower,efflow);
            end if;
        end if;
    end function;

```

mayureshw, 2024-11-20:
UPDATE: Confirmed! Replacing this with iterative function makes synthesis work. FYI - this function synthesizes fine in Vivado, so should be possible in ghdl.

tgingold, 2024-11-20:
Can you create a reproducer ?  There are already some tests with recursion, so there is something different.

mayureshw, 2024-11-21:
```vhd :file: test.vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.std_logic_misc.all;
use ieee.numeric_std.all;

entity test is
    port (
        i : in  std_logic_vector( 127 downto 0 );
        o : out std_logic_vector( 6 downto 0 )
    );
    function onehot2int( v : std_logic_vector; efflow : integer := 0 ) return integer is
        variable mid : integer := v'length/2;
        variable upper : std_logic_vector( v'length - mid - 1 downto 0 ) := v( v'length - 1 downto mid );
        variable lower : std_logic_vector( mid - 1 downto 0 ) := v( mid - 1 downto 0 );
    begin
        if v'length = 1 then
            return efflow;
        else
            if or_reduce(upper) = '1' then
                return onehot2int(upper,efflow+mid);
            else
                return onehot2int(lower,efflow);
            end if;
        end if;
    end function;
end entity;

architecture test of test is
begin
    o <= std_logic_vector( to_unsigned ( onehot2int(i), o'length ) );
end architecture;
```
```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a test.vhdl
ghdl --synth test.vhdl -e test
```

commit 3d758c1c (2024-11-23): synth-vhdl_stmts: do not keep multiple return values

Fix #2798
closed 2024-11-23 by tgingold