ghdl/ghdl #2800: Show Memory Data

FlinkbaumFAU, 2024-11-20:
**Is your feature request related to a problem? Please describe.**
Commercial simulators, such as Modelsim, allow me to inspect memory data. E.g. it shows the value of a shared variable at the end of the simulation, making it possible to inspect RAM.

**Describe the solution you'd like**
An option which allows dumping shared variables at the end of a simulation.

**Describe alternatives you've considered**
Accessing the variable via external names and dumping them  with `report` statements. However this is a bit tedious and it only works with VHDL 2008.

Paebbels, 2024-11-20:
[OSVVM](https://github.com/OSVVM) offers a [sparse memory model](https://github.com/OSVVM/OSVVM/blob/main/MemoryGenericPkg.vhd) implemented as a protected type (PT). This PT can read files and write files. Thus, the memory content can be dumped in between and at the end of a simulation. The memory model is then also embedded in a Verification Component (VC) like [Axi4Memory](https://github.com/OSVVM/AXI4/blob/main/Axi4/src/Axi4Memory.vhd) offering an [AXI4](https://github.com/OSVVM/AXI4) subordinate providing "endless" memory. The PT can also be access via external names, thus a control process can prefill, check, consume, manipulate and dump memory contents.

If the memory is deep inside of the DUT, also an implementation with a virtual transaction interface (VTI) can be used.

-----
/cc @jimlewis

JimLewis, 2024-11-20:
@Paebbels  - almost endless memory.    The package produces warnings at 34 bits of address and supports up to 40 bits of address (hard limit internally).    Note some simulators will crash at some point in this range of addresses - our biggest goal was to produce a warning before they did - mission accomplished.   

Note the data structure for the OSVVM memory model is created in a package.   It is looked up by an ID.   The ID can be searched for by name.   Hence, for the Axi4Memory model, you can look up the ID by name in the testbench and do any memory operation on the memory - including write out the memory contents when the simulation finishes.