ghdl/ghdl #2802: error: cannot find "std" library

GTP95, 2024-11-21:
Greetings,

I'm trying to synthetize an [AES implementation](https://github.com/hgrosz/aes-dom) using Yosys, GHDL and the ghdl-yosys-plugin. I tried bot in an Ubuntu installation inside WSL2 (Windows 11) and in a Kubuntu VM running on Virtualbox. In both cases, I got the following output:

```
$ yosys -m ghdl -p 'ghdl aes_top; synth_ice40 -json aes_top.json'

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.33 (git sha1 2584903a060)

-- Running command `ghdl aes_top; synth_ice40 -json aes_top.json' --

1. Executing GHDL.
warning: ieee library directory '/usr/lib/ghdl/gcc/vhdl/ieee/v93/' not found
error: cannot find "std" library
ERROR: vhdl import failed.
```
I also noticed that I have some files that could be related in the following path:
```
$ ls /usr/lib/ghdl/src/ieee/v93
numeric_bit-body.vhdl  numeric_bit.vhdl  numeric_std-body.vhdl  numeric_std.vhdl  std_logic_1164-body.vhdl  std_logic_1164.vhdl
```
Therefore, in the case of Ubuntu running under WSL2, I tried making those available where GHDL is looking for the library by creating the missing directories and making a symlink, to no avail. Again the case of WSL2, I also tried installing GHDL by compiling it from source, but I still ran into the same problem. I didn't try these steps inside the Virtualbox VM as I suspect these aren't the right solution. Could you please help me in getting GHDL to work? 
In the case of the Virtualbox VM, I have 
```
$ ghdl -v
GHDL 4.1.0 (Ubuntu 4.1.0+dfsg-0ubuntu2) [Dunoon edition]
 Compiled with GNAT Version: 13.2.0
 static elaboration, mcode JIT code generator
```
while under WSL2, after installing from source, I have 
```
$ ghdl -v
GHDL 5.0.0-dev (4.1.0.r299.g8798f1195) [Dunoon edition]
 Compiled with GNAT Version: 13.2.0
 static elaboration, mcode JIT code generator
```

tgingold, 2024-11-21:
Can you analyze your files using ghdl ? (probably yes)
What is the output of `ghdl --dispconfig` ?

tgingold, 2024-11-21:
Also, within yosys, try: `ghdl --dispconfig`

tgingold, 2024-11-22:
Can you try under yosys: `ghdl --disp-config` ? (Sorry for the typo)

GTP95, 2024-11-25:
No problem, here's the output:

```
$ yosys -m ghdl

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.33 (git sha1 2584903a060)

yosys> ghdl --disp-config
1. Executing GHDL.
command line prefix (--PREFIX): (not set)
warning: ieee library directory '/usr/lib/ghdl/gcc/vhdl/ieee/v93/' not found
environment prefix (GHDL_PREFIX): (not set)
exec prefix (from program name): (not found)

library prefix: /usr/lib/ghdl/gcc/vhdl
library directory: /usr/lib/ghdl/gcc/vhdl
default library paths:
 /usr/lib/ghdl/gcc/vhdl/ieee/v93/
 /usr/lib/ghdl/gcc/vhdl/
yosys plugin compiled on Apr 18 2024 11:46:11, git hash:unknown
```

tgingold, 2024-11-26:
So it looks like you have installed ghdl with mcode but you are using the ghdl plugin for gcc backend.
Try to set `export GHDL_PREFIX=/usr/lib/ghdl/mcode/vhdl` before running yosys.

GTP95, 2024-11-26:
It worked! Thank you!