ghdl/ghdl #2804: Unable to emit a warning only with locally non-static choice

am9417, 2024-11-22:
**Description**
We have a code that violates(?) VHDL standard (see example). Other synthesis and simulator tools emit a only a warning here so we're trying to compile the design with GHDL nevertheless, using `-Wstatic` analyze option.

Looks like GHDL provides a [way to emit a warning](https://ghdl.github.io/ghdl/using/InvokingGHDL.html#cmdoption-ghdl-Wstatic) instead of an error in this case.

**Expected behaviour**
Warning should be emitted, but instead an error is raised and design won't compile.
```
example.vhd:63:14:error: choice must be locally static expression
        when IDLE =>
             ^
```

**How to reproduce?**
Compile the minimal reproducible example below.

Constants IDLE, RUN, and RESTART are locally non-static in the case statement choices.

```vhd :file: example.vhd
entity example is
  port(
    Clk        : in std_logic;
    Start_in   : in std_logic;
    Restart_in : in std_logic;
    State_out  : out std_logic_vector(1 downto 0)
  );
end example;

architecture Rtl of example is

  constant IDLE    : std_logic_vector := "00"; --0
  constant RUN     : std_logic_vector := "01"; --1
  constant RESTART : std_logic_vector := "11"; --3
  signal state     : std_logic_vector(1 downto 0) := IDLE;

begin
  state_process : process(Clk)
  begin
    if rising_edge(Clk) then

      case state is
        when IDLE =>
          if Start_in = '1' then
            state <= RUN;
          end if;

        when RUN =>
          if Restart_in = '1' then
            state <= RESTART;
          end if;

        when RESTART =>
          state   <= IDLE;

        when others =>
          state <= state;

      end case;
    end if;
  end process;
  State_out <= state;
end Rtl;

```

Trying to analyze with `-Wstatic` flag:
Also tried unsuccessfully with -Wno-static

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl analyze -v --std=08 -Wno-shared -Wno-library -Wno-hide -Wstatic -frelaxed -fsynopsys example.vhd
```

**Context**
Running on ubuntu 22.04 (Subsystem for Windows).

GHDL -v (using latest released GHDL)
```
GHDL 4.1.0 (4.0.0.r39.g7188e92cf) [Dunoon edition]
 Compiled with GNAT Version: 10.5.0
 GCC 12.1.0 code generator
Written by Tristan Gingold.
```

**Additional context**

While writing the issue, I started to feel like the `-Wstatic` is applied only to emit a warning with locally non static **expressions**, and not **choices** like in this case. Anyway, looks like the support for latter is missing.

tgingold, 2024-11-22:
Yes, in case statements, the choices must be static.  Otherwise, the analyzer cannot check the choices are distinct and complete.

am9417, 2024-11-25:
Ok! It sounds like adding the support for emitting only a warning and analyzing the file even with non-static choices is hard or impossible.

Should the documentation or error message (in case of a locally non-static choice) be updated? From the error message I got mislead that it could be downgraded into a warning with `-Wstatic`:

>example.vhd:63:14:error: choice must be locally **static expression**

>-Wstatic: Emit a warning when a non-static expression is used at a place where the standard requires a **static expression**.

diogratia, 2024-12-04:
There are two classes of static expressions, locally and globally static. Your case statement choices in question are globally static because assignment during elaboration determines the bounds the constant declarations subtype indications lacking a constraint.

Those vendors only issuing a warning in lieu of a black and white semantic requirement have an implementation that places portability at risk. They likely have command line options for strict standard compliance to insure portability instead of in this case evaluating the constants value expression at analysis time.

The set of value expressions whose length can be determined at analysis time is a subset of the possible value expressions and here the standard does not require evaluating value expressions at analysis time when the constraints are not supplied.

You might imagine such noncompliance with the standard by default could be used for market lock-in where customers are reluctant to under go validation again after modifying their design models for portability relying on synthesis tools that rely on elaborated models and not VHDL simulation for formal proofs.

When you get multiple vendors supporting semantic changes to break down a walled city market of a dominant vendor it’s an indication there should be vendor participation in the standardization process potentially requiring additional semantic complexity (which could increase analyzer complexity greater than distinctly supporting multiple standard revisions).

As for ghdl the lack of guidance on the use of -Wstatic is likely predicated on the skill level expected of a VHDL user familiar with the standard. Here the ghdl documentation uses the term of art *static* found in the standard and meaning either locally or globally static. Filling the documentation with excerpts or references might be fraught with implying the effort to keep the documentation up to date for detail the user of such a flag might be expected to be familiar with or know how to find in the standard. 

You could note *locally static* is separately defined in the standard derailing the reasoning you espoused in the preceding comment.

am9417, 2024-12-10:
Thanks for the comment.

There are static expressions in VHDL and my initial assumption was that `-Wstatic` switch would suppress the error in these cases.  Apparently that is not the case. According your comment, the switch should be removed altogether?

The choice in a case statement is actually an expression, too, so I got mislead by the error messages. Anyway, both locally static and globally static are defined in the standard (9.4.2 and 9.4.3 respectively), so I don't understand what the note that *locally static is separately defined* means.

I don't know what skill level of VHDL standard is expected before I can start using GHDL and can't find that anywhere. However, [GHDL welcomes contributions from the community](https://github.com/ghdl/ghdl?tab=readme-ov-file#contributing), hence I opened issue with what I think is a mismatch between what's documented and expected and actual behaviour. From my experience, developing real-life applications with VHDL is based on relying that tools follow the standard and flag code violating the standard (which has not happened with the code that I've pasted) rather than the designer knowing details of the standard. And of course, it's only positive that user is forced to follow the standard - or make deliberate actions to violate - when using GHDL to ensure compability with other vendors' tools.

tgingold, 2024-12-11:
@am9417 Obviously the need of locally static expressions is required for consistency reasons (at least in some place).  For example, it is required when it can change the type of an expression.

In some other places, this is not exactly true and `-Wstatic` can relax this requirement.

For case statements, if GHDL checked for completeness after elaboration, it could have relaxed this requirement.

For sure the documentation could be improved to clarify this point.  But you cannot expect magic.

Most vendors relax the check for case statements, which is a violation of the standard.

closed 2024-12-14 by tgingold