ghdl/ghdl #2806: GNA: Issue2005 not working

Paebbels, 2024-11-23:
**Description**
When #2718 was developed, GNA issue2005 needed to be disabled on all macOS platforms.  
From original understanding, it needed to be deactivated just on `arm64` platforms.

The error reports a duplicate rpath:
> ld: duplicate LC_RPATH '/Users/runner/work/ghdl/ghdl/gnat/lib' in '/Users/runner/work/ghdl/ghdl/install/lib/libghdl-5_0_0_dev.dylib'

```
  + c_compiler_is_available
  + '[' -z ']'
  + which gcc
  + CC=gcc
  + which gcc
  ++ /Users/runner/work/ghdl/ghdl/install/bin/ghdl --libghdl-name
  + lib=libghdl-5_0_0_dev.dylib
  ++ /Users/runner/work/ghdl/ghdl/install/bin/ghdl --libghdl-library-path
  + lib_path=/Users/runner/work/ghdl/ghdl/install/lib/libghdl-5_0_0_dev.dylib
  + '[' '!' -f /Users/runner/work/ghdl/ghdl/install/lib/libghdl-5_0_0_dev.dylib ']'
  + lib=ghdl-5_0_0_dev.dylib
  + lib=ghdl-5_0_0_dev
  ++ dirname /Users/runner/work/ghdl/ghdl/install/lib/libghdl-5_0_0_dev.dylib
  + libdir=/Users/runner/work/ghdl/ghdl/install/lib
  ++ /Users/runner/work/ghdl/ghdl/install/bin/ghdl --libghdl-include-dir
  + incdir=/Users/runner/work/ghdl/ghdl/install/include
  ++ uname -s
  ++ uname -m
  + '[' Darwin = Darwin -a x86_64 = arm64 ']'
  ++ /Users/runner/work/ghdl/ghdl/install/bin/ghdl --libghdl-library-path
  + gcc -I/Users/runner/work/ghdl/ghdl/install/include test.c /Users/runner/work/ghdl/ghdl/install/lib/libghdl-5_0_0_dev.dylib
  ld: duplicate LC_RPATH '/Users/runner/work/ghdl/ghdl/gnat/lib' in '/Users/runner/work/ghdl/ghdl/install/lib/libghdl-5_0_0_dev.dylib'
  clang: error: linker command failed with exit code 1 (use -v to see invocation)
  
  gna test failed (issue2005 )
```

Looking at the outputs of `otool`:
```
./install/bin/ghdl:
	@rpath/libgcc_s.1.1.dylib (compatibility version 1.0.0, current version 1.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1336.61.1)
./install/bin/ghdl1-llvm:
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1319.100.3)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.11)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 1500.65.0)
./install/lib/libghdl-5_0_0_dev.dylib:
	@rpath/libghdl-5_0_0_dev.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libgnat-14.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libgcc_s.1.1.dylib (compatibility version 1.0.0, current version 1.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 13[36](https://github.com/Paebbels/ghdl/actions/runs/11903414967/job/33170495976#step:9:37).61.1)
./install/lib/libghdlvpi.dylib:
	@rpath/libghdlvpi.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1336.61.1)
```

then yes, `otool` seams to report self-references for all dynamic libraries generated by the Ada toolchain / linker.

**Expected behavior**
This code changed could be reverted:  
https://github.com/ghdl/ghdl/pull/2718/files#diff-b019cd33dfa480fe26382ddf3ff5d62f2c003d32f664db51d7b151077b1d9098L20

**How to reproduce?**
It's part of the CI pipeline on GitHub.

gna issue2005: https://github.com/Paebbels/ghdl/actions/runs/11903414967/job/33170495976#step:11:629
otool output: https://github.com/Paebbels/ghdl/actions/runs/11903414967/job/33170495976#step:9:34

**Context**
Please, provide the following information:

- OS: macOS 13 (x64_64) - GitHub Action Runner
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: 04c5f4e9

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

Paebbels, 2024-12-02:
@kms1212 any ideas on this one?

closed 2024-12-16 by tgingold

mayureshw, 2025-01-27:
-R breaks the build on NetBSD. libgnat resides in something like /usr/pkg/gcc10-aux/lib/gcc/x86_64--netbsd/10.3.0/adalib. Hence rpath is necessary.