ghdl/ghdl #2814: GHA: GHDL is currently not compiled with GCC backend

Paebbels, 2024-11-28:
A follow up from #2718.

**Description**
I couldn't figure out how to compile GHDL with GCC backend. The Ubuntu job template is prepared for it, but at 50% of the execution it can't find a make target.

Failing job: https://github.com/Paebbels/ghdl/actions/runs/12074253218/job/33671916814

```
make -j4 -f ../../libraries/Makefile.inc LIBSRC_DIR="../../libraries" LIBDST_DIR="lib/ghdl" enable_gplcompat="false" LN="ln -s" CP="cp" MKDIR="mkdir" GHDL="../../install/bin/ghdl" GHDL_FLAGS=" " vhdl.libs.all libs.vhdl.standard
make[1]: warning: -j4 forced in submake: resetting jobserver mode.
make[1]: *** No rule to make target '../../install/bin/ghdl', needed by 'lib/ghdl/std/v8[7](https://github.com/Paebbels/ghdl/actions/runs/12074253218/job/33671916814#step:13:8)/std-obj87.cf'.  Stop.
make[1]: Entering directory '/home/runner/work/ghdl/ghdl/build/gcc'
make[1]: Leaving directory '/home/runner/work/ghdl/ghdl/build/gcc'
make: *** [Makefile:333: libs.vhdl.gcc] Error 2
```

[image]

**Expected behavior**
Also a Ubuntu + gcc backend variant should be build in CI.

**How to reproduce?**
Activate the commented out GCC matrix combination(s).

**Context**
Please, provide the following information:

- OS: Ubuntu 22.04 / 24.04
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: latest

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

Paebbels, 2024-11-29:
I tried to ignore/remove/reduce :) the complexity of also specifying the GCC version in addition to backend and OS version. I would advise to not multiply the current matrix times gcc versions. The pipeline performance is limited by GitHub by the number of concurrent jobs.

If you think I should check other GCC versions, I can have a look to specify the GCC version different from the default GCC version. Then we can e.g. select a GCC version for the GCC backend, which isn't the default GCC. But I wouldn't build e.g. the mode backend for different GCC versions.

OR:  
Do you think there is a certain risk in various GCC and LLVM versions, so we should check them by CI?

Paebbels, 2024-11-29:
Building with GCC-12 leads to the same results: https://github.com/Paebbels/ghdl/actions/runs/12090660016/job/33717974039

----
Oh wait, did you mean:
1. compile GHDL and GCC with gcc-12 (compiler to compile both tools) or did you mean
2. compile for gcc-12, so any compiler, but build GHDL and GCC by using GCC version 12 sources?

Paebbels, 2024-11-29:
Also GCC 12.1.0 sources create the same problem: https://github.com/Paebbels/ghdl/actions/runs/12090950120/job/33718682663

tgingold, 2024-11-30:
At some point 'ghdl' binary is installed:
https://github.com/Paebbels/ghdl/actions/runs/12090950120/job/33718682663#step:11:666

tgingold, 2024-11-30:
I fear the prefix used for gcc and for ghdl are not the same.

Paebbels, 2024-11-30:
Oh, there are these errors:

> [...]  
> mkdir: cannot create directory ‘/home/runner/work/ghdl/ghdl/gcc-objs/gcc/bin’: File exists  
> [...]  
> mkdir: cannot create directory ‘/home/runner/work/ghdl/ghdl/gcc-objs/gcc/lib’: File exists  
> [...]  

GHDL has a prefix, but GCC has not

**Condensed view on configure and make:**
```
cd build/gcc
../../configure --prefix=../../install
make copy-sources

cd -

cd gcc-objs
../gcc-srcs/configure
```

**Directory structure:**
```
/
o- gcc-srcs
|  o- configure
o- gcc-objs
o- build
|  o- gcc
|     o- Makefile
o- install
o- configure
```

tgingold, 2024-11-30:
I think I almost got it.

But there is a problems with permissions: they are not kept with upload/download artifact.  We should probably create a tar/zip before.

Paebbels, 2024-11-30:
> But there is a problems with permissions: they are not kept with upload/download artifact. We should probably create a tar/zip before.

You refer to the artifact published as assets?

Now as you mention it, yes, that's a bug of GitHub. In the pipeline, I handle it manually. Let me think about a solution how to resolve this issue, so I can keep a generic NightlyRelease job template. If not, we can copy that template as a YAMl file to GHDL and manually restore access rights.

Btw. that template now also supports tar.zst files.

closed 2024-12-01 by tgingold

Paebbels, 2024-12-02:
I looked at your commit. It looks like the `chmod` settings you did are within the same VM, right? So not an artifact problem.

tgingold, 2024-12-02:
Maybe the one for the docker image is not necessary.

commit 88a44cc8 (2024-12-01): CI: enable gcc build

Fix #2814