ghdl/ghdl #2819: No CI build uses OpenIEEE packages

Paebbels, 2024-11-29:
A follow up from https://github.com/ghdl/ghdl/pull/2718.

**Description**
Currently, no GHDL build in the CI pipeline builds GHDL with GPL compatible IEEE libraries (OpenIEEE packages) auto-generated from Python code.

**Expected behavior**
At least one CI build should auto-generate the `std` and `ieee` library files and run the testsuite using a GHDL installation utilizing these auto-generated files.

**How to reproduce?**
This feature is missing in the new CI pipeline and was present in the old pipeline.

**Context**
Please, provide the following information:

- OS: Ubuntu
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: latest

**Additional context**
Please clarify if these auto-generated files are still needed, or if GHDL's needs can be fulfilled from backported VHDL-2019 package files.

Paebbels, 2024-11-30:
OK.

Do you want to remove the code base for it (e.g. make entries, configure), or just no CI?

closed 2024-11-30 by Paebbels

tgingold, 2024-11-30:
no CI.  And it should be probably removed from configure.