ghdl/ghdl #2820: ******************** GHDL Bug occurred ***************************

BLangOS, 2024-12-01:
**Description**
I want to synthesize a generated vexriscv-Processor with the following commands: 
ghdl -a ./Prozessor/VexRiscv.vhd
ghdl --synth --out=verilog VexRiscv  1>_GHDL_/VexRiscv.v

**Expected behaviour**
I expected to get a synthesized vhdl version of the processor description.

**How to reproduce?**
The bug happens each time when I synthesize the VexRiscv.vhd (second call of ghdl)

```vhd :file: ent.vhd
entity ent is
end entity;

architecture a of ent is
begin
  process begin
    report "Hello world" severity note;
    wait;
  end process;
end;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS: Windows 
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0 (2.0.0.r1417.g7de967c51) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: i686-w64-mingw32
L:\Eigene_Dateien\Home\2024-2025_WS\Projekte\Open_Source_Synthese\04_ICE40_VexRiscv\Source\
Command line:
L:\tools\ghdl-MINGW32\GHDL\bin\ghdl.exe --synth --out=verilog VexRiscv
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : elab-vhdl_values.adb:30 invalid data
Load address: 0xaf0000
Call stack traceback locations:
0xd6584f 0xd65d89 0xd6661d 0xd745bd 0xd74934 0xd5b66f 0xd5b88a 0xd1d109 0xdf4660 0xdd304a 0xe96ed1 0xaf2dea 0xaf12ee 0x766dfcc7 0x7786809c 0x7786806c
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.
[VexRiscv.zip](https://github.com/user-attachments/files/17970119/VexRiscv.zip)

BLangOS, 2024-12-01:
Sorry, I did forget to delete the in the defaults in the section "Expected behaviour".

Paebbels, 2024-12-01:
@BLangOS please adjust your issue title. You can update the title and content at any time.

Paebbels, 2024-12-01:
Version `2.0.0.r1417` is quite old. Can you upgrade to 4.1 or recompile from sources (your version is a development release)

BLangOS, 2024-12-01:
OK, I will try it. Thanks for the info.

BLangOS, 2024-12-01:
After updating to the latest GHDL, everything now works. Many thanks for your help.

closed 2024-12-01 by BLangOS