ghdl/ghdl #2822: Coverage JSON is very broken

BenWiederhake, 2024-12-02:
**Description**
Trying to generate coverage JSON files is broken in many ways:
- crash with `CONSTRAINT_ERROR : trans-coverage.adb:76`
- crash with `CONSTRAINT_ERROR : ghdlcovout.adb:109`
- invalid JSON

**Expected behaviour**
I expected `rm -f work-obj93.cf && ghdl -a mre.vhd && ghdl -r --coverage a` to produce some kind of JSON output in `coverage-DATETIME.json` that tells me something about coverage, [as documented](https://ghdl.github.io/ghdl/using/Simulation.html#cmdoption-ghdl-coverage). In particular, I expect ghdl to not crash, and the JSON file to be valid JSON.

**How to reproduce?**

- Case 1:

```vhd :file: ent1.vhd
entity a is end entity;
architecture b of a is begin end b;
```

… results in:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r398.g8ef5a830f) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: x86_64-linux-gnu
/root/mre-eo/
Command line:
/root/ghdl/build/ghdl_mcode -r --coverage a
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-coverage.adb:76 range check failed
******************************************************************
```

- Case 2:

```vhd :file: ent2.vhd
entity c is end entity;
architecture d of c is
signal foo: bit;
begin
foo <= '1';
end d;
```

… results in:

```
ghdl:error: NULL access dereferenced
ghdl:error: simulation failed

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r398.g8ef5a830f) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: x86_64-linux-gnu
/root/mre-eo/
Command line:
/root/ghdl/build/ghdl_mcode -r --coverage c
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : ghdlcovout.adb:109 access check failed
******************************************************************
```

- Case 3:

```vhd :file: ent3.vhd
entity e is end entity;
architecture f of e is
signal foo: bit;
signal bar: bit;
begin
foo <= '1';
bar <= '0';
end f;
```
… results in:

```console
# cat coverage-20241202175708.243.json 
{
 'app': 'GHDL coverage output file',
 'version': '1.0.0',
 'testcase': 'unknown',
 'timestamp': '20241202175708.243',
 [
  {
   'file': 'mre.vhd',
   'dir': '.',
   'sha1': '049de034341a683815a69473935e01d4f6754635',
   'mode': 'stmt',
   'max-line': 7,
   'result': [
    '6': 1,
    '7': 1
   ]
  }
 ]
}
```

This is not JSON for three separate reasons:
- it uses `'` instead of `"`
- the data in the line after `timestamp` have no key, despite being in an object/dict
- the data to `result` are clearly meant to be an object/dict, despite using object syntax

One way to make it actual JSON would be:

```json
{
  "app": "GHDL coverage output file",
  "version": "1.0.0",
  "testcase": "unknown",
  "timestamp": "20241202175708.243",
  "outputs": [
    {
      "file": "mre.vhd",
      "dir": ".",
      "sha1": "049de034341a683815a69473935e01d4f6754635",
      "mode": "stmt",
      "max-line": 7,
      "result": {
        "6": 1,
        "7": 1
      }
    }
  ]
}
```

As a temporary workaround, I'm using the following script to convert the "coverage JSON" files to actual JSON:
```
sed -Ee "s-'-\"-g" \
    -e 's,^ \[,"outputs": [,' \
    -e 's,"result": \[,"result": {,' \
    -e 's,^   \],},'
```

With the above files, the following "script" should be 

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent1.vhd ent2.vhd ent3.vhd
ghdl -r --coverage a # ent1, crashes
ghdl -r --coverage c # ent2, crashes
ghdl -r --coverage e # ent3, returns with exitcode 0 ("no error"), but produces invalid output
cat coverage* | jq . # Errors out because "JSON" coverage files are not valid JSON.
```

**Context**
Please, provide the following information:

- OS: Debian
- Origin:
  - [X] Package manager: `GHDL 4.1.0 (Debian 4.1.0+dfsg-2+b1) [Dunoon edition]`
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `commit 8ef5a830f031e08b2cbfa80cb86cc7de3773c772`, self-reports as `GHDL 5.0.0-dev (4.1.0.r398.g8ef5a830f) [Dunoon edition]`

I used the current `master` to generate the above reports, just to verify that the bug hasn't already been fixed.

Since these "three" bugs are closely linked together, I reported them as one. Let me know if you'd prefer to see three separate issues for this.

Paebbels, 2024-12-02:
@tgingold which code coverage JSON dialect is this? We could add a schema check to the CI pipeline.

1. checking for valid JSON
2. validating the content against a JSON schema

tgingold, 2024-12-03:
Thank you.  I have just fixed the output, following your comments.
But this output is not supposed to be read by users (also you can).

I will investigate the crashes.

@Paebbels I am not sure we want to go that far!

Paebbels, 2024-12-03:
Is this JSON a common JSON coverage format or "invented" by GHDL?

I know some frameworks have a conversion to JSON (`Coverage.py`) but I'm not aware or a common standard format as JSON like Cobertura in XML. But if it's a standard format, we could easily check it.

tgingold, 2024-12-03:
No, it's tailored for GHDL.

Paebbels, 2024-12-04:
@tgingold Do you have a format description? I would like to add it to pyEDAA.Reports (potential christmas project) so we can convert it to e.g. Cobertura for usage with Coveralls, CodeCov, Codacy and GitLab.

tgingold, 2024-12-05:
It can output lcov, gcov or gcovr files.  The latter is a JSON file.

Those files are for the final output, while the JSON file in this issue is about raw results, before merging.

Paebbels, 2024-12-05:
Ok, then I don't need to have a format reader for that, because I can use lcov, gcov or gcovr files. Or is there any information in the JSON format that's not present in these 3 file formats?

closed 2024-12-06 by tgingold

tgingold, 2024-12-06:
@Paebbels yes, we can put more information in the intermediate file (like testcase name...).

BenWiederhake, 2024-12-08:
Thank you for these quick fixes! :D

commit ab4189f4 (2024-12-06): coverage: fix crashes

Fix #2822