ghdl/ghdl #2829: casting to like types in a port map does not propogate constrained width

awschult002, 2024-12-12:
**Description**
GHDL exhibits different behavior than other simulators when working with unconstrained array types in port maps. My specific issue is that I have an entity with unconstrained SLV and the parent caller sends in a fully constrained `unsigned`. The unsigned is casted to an SLV; but GHDL generates an error that the vector "must be fully constrained". Neither Questa/Modelsim, nor XSim, exhibit this behavior.

This behavior is observed in GHDL 3.0.0/4.1.0/5.0.0-dev

**Expected behaviour**
I would expect the fully constrained width of the `unsigned` type to propagate down to the SLV in the port description, thus constraining the port and preventing any errors.

**How to reproduce?**
Here is a link to a MWE on eda playground
https://www.edaplayground.com/x/s6Yp

tgingold, 2024-12-14:
Doesn't LRM2008 5.3.2.2 applies here ?

For an interface object or subelement whose mode is out, buffer, inout, or linkage, if the
formal part includes a conversion function or a type conversion, then the parameter sub-
type of that function or the type mark of the type conversion shall define a constraint for
the index range corresponding to the index range of the object, and the index range is
obtained from that constraint; otherwise, the index range is obtained from the object
denoted by the actual designator.

diogratia, 2024-12-14:
Casting isn't a term of art in VHDL. It's type conversion here, the semantics are subtly different.

EDAplayground requires a login to see both your design units. The casual and wily reader without a login can remove the overlay window but not scroll through the complete architecture of mwe_tb.

With a little help from screenshots and MacOS Preview app text conversion plus a bit of editing and completion a test case might look like:

```vhdl
library ieee;
use ieee.std_logic_1164.all; 
-- use ieee.numeric_std.all;

entity mwe is
    port (
        data : in std_logic_vector;
        data_out : out std_logic_vector
    );
end mwe;

architecture rtl of mwe is begin
    data_out <= data; -- CHANGED WAS data_in on RHS
end architecture;

Library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity mwe_tb is
end mwe_tb;

architecture tb of mwe_tb is 
    signal data_in : unsigned(7 downto 0); 
    signal data_out : unsigned(7 downto 0);
begin
uut_fail : 
    entity work. mwe 
    port map (
        data => std_logic_vector(data_in),
       -- data_out => std_logic_vector(data_out) -- CHANGED
        unsigned(data_out) => data_out -- type convert formal for mode out
    );
wrap: 
    block
        signal slv_data_in : std_logic_vector(data_in'range);
        signal slv_data_out : std_logic_vector(data_out'range);
    begin
    end block;
end architecture;
```

Note there were two errors in the image of your original design units. One in mwe with an undeclared signal and one with a mode out port type conversion on the actual instead of formal. Where to use type conversion is specified in IEEE Std 1076-1993 3.2.1.1 Index constraints and discrete ranges (or 5.3.2.2 Index constrains and discrete ranges for -2008).

Those have been corrected and the block statement foreshortened and architecture end provided shows the error on the mode in and warning on mode out ports (the out mode  port can generate a detectable error at elaboration):

```none
%: ghdl -a mwe_tb.vhdl
mwe_tb.vhdl:30:33:error: type of actual conversion must be fully constrained
        data => std_logic_vector(data_in),
                                ^
mwe_tb.vhdl:32:17:warning: type of formal conversion must be fully constrained [-Wport]
        unsigned(data_out) => data_out
                ^
%: 
```

If we look a little deeper in in 3.2.1.1 we see this is expected behavior:

>If the index ranges for an interface object, or member of an interface object, are obtained from the corresponding association element (when associating in whole) or elements (when associating individually), then they are determined either by the actual part(s) or by the formal part(s) of the association element(s), depending upon the mode of the interface object, as follows:  
>  
>> -- For an interface object or member of an interface object whose mode is **in**, **inout**, or **linkage**, if the actual part includes a conversion function or a type conversion, then the result type of that function or the type mark of the type conversion must be a constrained array subtype, and the index ranges are obtained from this constrained subtype;otherwise, the index ranges are obtained from the object or value denoted by the actual designator(s).  
>>  
[... 46 lines cut ...]
        signal slv_data_out : std_logic_vector(data_out'range);
    begin
    end block;
end architecture;
```

You can provide a conforming type mark to the type conversions.

So where does the idea come from that it's okay to use type conversion as you say several vendors do?  We see an incomplete semantic description of using type conversion , in -2008:

6.5.7 Association lists  
6.5.7.1 General  

>The formal part of a named association element may be in the form of a function call, where the single argument of the function is the formal designator itself, if and only if the formal is an interface object, the mode of the formal is **out**, **inout**, **buffer**, or **linkage**, and if the actual is not **open**. In this case, the function name shall denote a function whose single parameter is of the type of the formal and whose result is the type of the corresponding actual. Such a conversion function provides for type conversion in the event that data flows from the formal to the actual.  
>  
>Alternatively, the formal part of a named association element may be in the form of a type conversion, where the expression to be converted is the formal designator itself, if and only if the formal is an interface object, the mode of the formal is **out**, **inout**, **buffer,** or **linkage**, and if the actual is not **open**. In this case, the base type denoted by the type mark shall be the same as the base type of the corresponding actual. Such a type conversion provides for type conversion in the event that data flows from the formal to the actual. It is an error if the type of the form\al is not closely related to the type of the actual. (See 9.3.6.)

(Similar story for -1993 4.3.2.2 Association lists.)
These semantic descriptions omit the restriction on the type mark found in 3.2.1.1/5.3.2.2.

There appear to be multiple vendors who for whatever reason have never enforced the semantic requirement on the type mark found in the subsection Index constraints and discrete ranges. That they can may indicate the issue is ripe for a change in a future revision. (And I'd want to be convinced their implementations have no unintended consequences first.)

[... cut ...] icularly as default behavior. I'm also reminded of the made up game Fizzbin from Star Trek (the original series). Make the rules complex enough and they can mean anything you want them to mean. To prevent that VHDL uses terms of art (in italics) defined in the standard itself. The term cast appears in -2008 21. VHPI callbacks and doesn't refer to VHDL type conversion.

Note that `must` and `shall` have mandatory weight in semantic descriptions of the two revisions of the standard mentioned here.

awschult002, 2024-12-15:
Thank you so much for the reply. This is exactly what I was hoping to see. 

I am not an expert in reading the language standard. But if I understand what you have provided correctly. When doing a type conversion, the range is taken from the type itself and not from the object of that type? So when performing a type conversion of the unsigned to SLV, it tries to take the range of the type "unsigned"... Which is unresolved by definition. Because of which, it tries then to take the range of the port member... Which was also unresolved. Thus the error. 

This also gives reason to why you have created a subtype with the explicit range in your example. 

Am I understanding that correctly?

awschult002, 2024-12-19:
I am going to assume that by closing this (I agree with) you are confirming my understanding of your response?