ghdl/ghdl #2831: Schedule for releases ?

csantosb, 2024-12-14:
**Is your feature request related to a problem? Please describe.**
I [package](https://codeberg.org/guix-science/guix-science/src/master/guix-science/packages/electronics.scm#L129) releases of ghdl for guix package manager. Latest release 4.1.0, is from 8 months ago.

I wonder if there is a policy about releases, or a schedule about it ?

**Describe the solution you'd like**
Regular releases every few months, or a clear statement about this.

**Describe alternatives you've considered**
One may always package random releases for its own use, but this is far from satisfactory.

**Additional context**
When doing ci testing, having a release version of ghdl helps traceability of results, improving reproducibility.

Paebbels, 2024-12-14:
Final releases are created by @tgingold, but I can try to answer some questions.

Usually, GHDL gets a major release like `v3.0` and `v4.0` around February each year. So `v5.0` is expected in around 2 months. The current version is: `5.0.0-dev`.

Most users are building GHDL from latest sources, as GHDL is very stable on the main branch as well as it supports many platforms with the provided build flow. Package maintainers are free to deliver packages of any commit on the `main` branch.

GHDL itself provides nightly builds (there has been a major change in the GitHub pipeline some time ago to enhance that process.) These nightly builds are based on a `nightly` tag which is shifted from time to time (somehow around 1..4 times a month. This tag could be also interesting for package maintainers. So they don't rely on any arbitrary commit on the main branch. The results of the build pipeline are uploaded as assets with permanent links to a release page: https://github.com/ghdl/ghdl/releases/tag/nightly  
(Please note: the release page currently shows the date when the first nightly was published.)

Every nightly build also pushes new GHDL docker images published on Docker Hub.

For traceability, the `GHDL --version` command contains the Git hash. So in case an arbitrary commit was used, it can be traced back in the repository.

-----
The GitHub pipeline was reworked as we figured out that many packages download assets from the `v4.1.0` or `nightly` release page and install the files on the destination machines. These installations are many times broken, because of various reasons. My goal would be a `v4.2` release quite soon, so we can test the new pipeline and release flow before we have the `v5.0` release. This would allow me to create PRs for macOS to fix these homebrew casks (recipes). Currently it installs x86 code on aarch64 macOS.

-----
With more and more package managers supporting GHDL (also constantly supporting it), we should think about a quarterly release cycle or so. Like `v5.0` to `v5.3` every 3 months to gather bug fixes and improvements.

-----
Currently, the history with Debian/Ubuntu is an on-off story ...

csantosb, 2024-12-14:
>sam. 14 déc. 2024 at 13:03, Patrick Lehmann ***@***.***> wrote:

> Most users are building GHDL from latest sources, as GHDL is very stable on the main
> branch as well as it supports many platforms with the provided build flow. Package
> maintainers are free to deliver packages of any commit on the `main` branch.

Thanks for the reply.

If I understand it correctly, there is no meaning attached to
tags/releases, other than being a periodic snapshot, is that correct ?

> With more and more package managers supporting GHDL (also constantly supporting it), we
> should think about a quarterly release cycle or so. Like `v5.0` to `v5.3` every 3 months
> to gather bug fixes and improvements.

+1 !

--
Cayetano Santos
GnuPG Key:   https://meta.sr.ht/~csantosb.pgp
FingerPrint: CCB8 1842 F9D7 058E CD67 377A BF5C DF4D F6BF 6682

Paebbels, 2024-12-16:
> If I understand it correctly, there is no meaning attached to tags/releases, other than being a periodic snapshot, is that correct ?

Yes.

As the `master` branch is always stable and we see only 1..2 regressions per year, GHDL can be build and "released" from any version. There is no feature driven development. New features are added step by step without breaking existing code. These are mainly improvements to non-working VHDL-2008 code snippets. In between you'll find many fixes for reported problems by users.

csantosb, 2024-12-16:
Thanks for your answer.

closed 2024-12-16 by csantosb