ghdl/ghdl #2834: Invalid "(-1 downto 0)" Verilog syntax with an "integer range 0 to 0" inside a record that is passed from one entity to entity

eduardosm, 2024-12-15:
**Description**

When there is a zero-sized integer (`integer range 0 to 0`) inside a record, and such record is passed from one instance to another, `ghdl synth --out=verilog` will produce invalid Verilog syntax. Similar to https://github.com/ghdl/ghdl/issues/2826, but in this case, there is an entity with the record as output and another entity with the record as input, connected together.

**Expected behaviour**

The following Verilog is generated, where the `assign n7_o = sig (-1 downto 0);` is invalid. The `n7_o` signal could be omitted.

```verilog
module take_in
  (input  i_a,
   input  i_c);
endmodule

module gen_out
  (output o_a,
   output o_c);
  wire n12_o;
  wire n14_o;
  localparam [1:0] n15_o = 2'b00;
  assign o_a = n12_o; //(module output)
  assign o_c = n14_o; //(module output)
  assign n12_o = n15_o[0]; // extract
  assign n14_o = n15_o[1]; // extract
endmodule

module top_level
  (output o_a,
   output o_c);
  wire n1_o;
  wire n3_o;
  wire [1:0] sig;
  wire inst1_o_a;
  wire inst1_o_c;
  wire n6_o;
  wire n8_o;
  wire [1:0] n10_o;
  assign o_a = n1_o; //(module output)
  assign o_c = n3_o; //(module output)
  assign n1_o = sig[0]; // extract
  assign n3_o = sig[1]; // extract
  /* top_level.vhd:37:12  */
  assign sig = n10_o; // (signal)
  /* top_level.vhd:39:5  */
  gen_out inst1 (
    .o_a(inst1_o_a),
    .o_c(inst1_o_c));
  /* top_level.vhd:44:5  */
  take_in inst2 (
    .i_a(n6_o),
    .i_c(n8_o));
  assign n6_o = sig[0]; // extract
  assign n7_o = sig (-1 downto 0); // extract
  assign n8_o = sig[1]; // extract
  assign n10_o = {inst1_o_c, inst1_o_a};
endmodule
```

**How to reproduce?**

```vhd :file: ent.vhd
package defs is
    type rec is record
        a: integer range 0 to 1;
        b: integer range 0 to 0;
        c: integer range 0 to 1;
    end record;
end package;

entity gen_out is
    port (
        o: out work.defs.rec
    );
end entity;

architecture arch of gen_out is
begin
    o <= (0, 0, 0);
end architecture;

entity take_in is
    port (
        i: in work.defs.rec
    );
end entity;

architecture arch of take_in is
begin
end architecture;

entity top_level is
    port (
        o: out work.defs.rec
    );
end entity;

architecture arch of top_level is
    signal sig: work.defs.rec;
begin
    inst1: entity work.gen_out
    port map (
        o => sig
    );

    inst2: entity work.take_in
    port map (
        i => sig
    );

    o <= sig;
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl synth --std=93 --out=verilog ent.vhd -e top_level
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: 4526d7eacad6ba3119c080a6ba520e51cefc4390

tgingold, 2024-12-15:
Thank you for the issue and the testcase.  Let's see if this is the last one!

closed 2024-12-15 by tgingold

eduardosm, 2024-12-15:
This seems to be the last one :) Thanks for fixing all the cases!

tgingold, 2024-12-15:
Thank you for the status update!

commit 27cc2f19 (2024-12-15): netlists-cleanup: fix iteration on linked list

Deal with a removal.
Fix #2834