ghdl/ghdl #2838: Nightlies not updating

avelure, 2024-12-19:
I see the last updates to the nightlies was a month ago with sha a3b41ea25.
Is it intentional or just in progress of being fixed in @Paebbels restructuring of the CI?

Paebbels, 2024-12-19:
The restructuring should be now complete-complete-finally-complete :).

Last problems where related to GitHubs artifact actions, which didn't preserve executable bits. I created now my own artifact handling[^1]. I'm using the nightly builds from https://github.com/Paebbels/ghdl/releases/tag/testing (here called testing in my repo, because of tag-collisions) in my own pipelines for a few days. Everything works as expected - it just took me 2 month + another 2 month to workaround bugs in GitHub Actions.

Please give it a try in case you want to test it early. Otherwise, I think GHDL is ready to "move" the nightly tag again as we did in the past.

-----
[^1]: In case someone needs fixed artifact handling: [upload](https://github.com/pyTooling/upload-artifact)/[download](https://github.com/pyTooling/download-artifact))

Paebbels, 2025-01-05:
@avelure running: https://github.com/ghdl/ghdl/actions/runs/12618140156

stnolting, 2025-01-05:
It seems like the nightly tag is not accessible by the GHDL GitHub action: 🤔 

[image]

Paebbels, 2025-01-05:
@stnolting Oh. Thank's for giving us quick feedback.

By reorganizing the pipeline and what artifacts and assets were created by the pipeline, some asset names were cleaned up and renamed. I'll check were the action is located and see how I can update that action.

stnolting, 2025-01-05:
> I'll check were the action is located and see how I can update that action.

Awesome! Thank you so much for your work!

closed 2025-01-07 by Paebbels