ghdl/ghdl #2848: Allow selecting records by name

jobtijhuis, 2025-01-03:
**Is your feature request related to a problem? Please describe.**
Many simulators and synthesizers allow selecting part of a record type to query for example the width of that field directly on the type. GHDL does not allow this even when using the relaxed rules. It can be quite a hassle to rewrite a whole codebase that uses constructs similar to this is a bit of a blocker to using GHDL.

**Example:**
The following code compiles on both QuestaSim and Active-HDL/Riviera Pro by default. On Synopsys it also compiles by adding the `-xlrm` option to relax LRM rules. With GHDL this does not compile even with relaxed rules.
```vhdl
library ieee;
  context ieee.ieee_std_context;

package test_pkg is

  type t_master is
  record
    valid : std_ulogic;
    data  : std_ulogic_vector(7 downto 0);
    status : std_ulogic_vector;
  end record;

  constant DATA_RESET_VAL : std_ulogic_vector(t_master.data'range) := x"00";

end package;
```
This is a simplified example, the access of the record fields could be somewhere else than in the package.

GHDL gives the error:
``
type "t_master" cannot be selected by name
``

**Describe the solution you'd like**
It would be preferred if GHDL would allow this behaviour when using the relaxed rules.

**Alternatives**
 * An update to the VHDL standard that would allow this by default could solve the dispute of whether this is allowed or not.
 * You can rewrite the VHDL to define the ranges before the record and use that definition for both the record and the constant, but I consider this to be quite unwieldy.
 * You can rewrite the VHDL to make a dummy signal with the record as a type and take the width from there but this is also not ideal.

Paebbels, 2025-01-03:
Wasn't there a problem in the VHDL language itself, that it doesn't work for types, but for objects? So if you define a constant temp, it would work.

```vhdl
constant temp : t_master;
constant DATA_RESET_VAL : std_ulogic_vector(temp.data'range) := (others => '0');
-- constant DATA_RESET_VAL : std_ulogic_vector(t_master.data'range) := x"00";
```

jobtijhuis, 2025-01-03:
That is indeed one way to do it. However you cannot use a constant like that because you must assign it a value. So depending on the context it would have to be a signal or a variable, or the constant needs to be assigned a value. But assigning a value to the constant is not always possible when the type is not fully constrained. Especially tricky in subprograms.
So like you said the best solution would be to allow it for types.

I still think that the interpretation of the VHDL language as GHDL sees it is correct. I am just curious whether a similar relaxation is possible as other simulators and synthesizers have.

Paebbels, 2025-01-03:
I tried to dive into names in the LRM, but I can't see why it's not allowed or want sentence limits the usage for types. Pinging @diogratia :). But I'm pretty sure, there are more tools then just GHDL having that limit.

diogratia, 2025-01-04:
As I recall Questasim has a command line parameter that forces strict compliance with the standard. It'd be interesting to see the result with that invoked. The documentation for Active-HDL/Riviera Pro  is not openly available.

That ol' `-frelaxed` is a terrible weapon to leave lying around loose particularly when addressing all sorts of compatibility issues between vendors.  You'd hope that any real questions of semantics would have been answered or at least submitted as issues to VASG in the 20 years or so most of them have been around.

The impetus for adding compatibility between non-conforming implementations appears to be rooted in the desire to artificially boost portability while avoiding verification and validation repetition.

When you get stymied trying to search the standard you can always look in ghdl source for hints. 

I use a C-Shell alias that invokes a find command for a partial string from an error message while passing further command line arguments to an executed invocation of grep. In this case (and this point in time) that revealed the error message came from vhdl-sem_names.adb line 2523.  Scrolling up we find the subject is LRM93 6.3 which is more or less unchanged in -2008 8.3 Selected names. So ghdl claims it's a selected name/expanded name issue.

Note this matches Patrick's first comment in this issue.

8.3 Selected names

> selected name can denote an element of a record, an object designated by an access value, or a named
entity whose declaration is contained within another named entity, particularly within a library, a package,
or a protected type. Furthermore, a selected name can denote all named entities whose declarations are
contained within a library or a package.  
>  
>For a selected name that is used to denote a record element, the suffix shall be a simple name denoting an
element of a record object or value. The prefix shall be appropriate for the type of this object or value.  

A selected name denoting an element of a record type declaration is neither an object nor a value of that type.

Your example is doesn't teach a need for a semantic change:

```vhdl
library ieee;
  context ieee.ieee_std_context;

package test_pkg is

  type t_master is
  record
    valid : std_ulogic;
    data  : std_ulogic_vector (7 downto 0);
    status : std_ulogic_vector;
  end record;

  -- constant DATA_RESET_VAL : std_ulogic_vector(t_master.data'range) := x"00";
  constant DATA_RESET_VAL : std_ulogic_vector := x"00";
  -- where the range constraint is taken from the value expression and the 
  -- left bound and direction is taken from the index subtype
  
end package;

library ieee;
use ieee.std_logic_1164.all;
use work.test_pkg.all;

entity usage is
end entity;

architecture foo of usage is
    signal record_obj: t_master (status(0 to 0));
begin
    record_obj.data <= DATA_RESET_VAL;  -- implicit subtype conversion preserves
                                        -- left right associativity
end architecture;
```

(I don't get github notifications, I managed to notice there was a new issue here.)

tgingold, 2025-01-04:
So, yes this is completely invalid.

Now the question is: do we want to support that (in -frelax) ?

Paebbels, 2025-01-04:
I would like to see it working in any VHDL tool and I would like to see it fixed in the language too. I see no reason, why a selected name shouldn't work on types.

I also think it's a very important feature, because my code has many workarounds for this problem. Ugly code that could have been written in a nicer style.

-----
Pinging @JimLewis too.

tgingold, 2025-01-04:
Well, with vhdl 2008 you have `'subtype` and `'element`. So the feature already exists.

JimLewis, 2025-03-24:
@diogratia 
>> For a selected name that is used to denote a record element, the suffix shall be a simple name denoting an
element of a record object or value. The prefix shall be appropriate for the type of this object or value.

> A selected name denoting an element of a record type declaration is neither an object nor a value of that type.

So are you suggesting then that it is amended to say:    
For a selected name that is used to denote a record element, the suffix shall be a simple name denoting an
element of a record object, value, or type. The prefix shall be appropriate for the object, value, or type?

It should be noted that the prefix for A'range is defined as:    
Prefix: Any prefix A that is appropriate for an array object, or an alias
thereof, or that denotes an array subtype whose index ranges are
defined by a constraint.

So here at the LRM intends that the simple name of a type can be used with
an attribute, hence, it is inconsistent that a record element is not allowed - almost
if one part of the LRM was updated without considering the other parts of the LRM
or more complex use models of the same enhancement.

F2g-attocube, 2025-04-14:
After I just worked through code created by a member of the balloting committee that voted on the 2008 standard and found multiple occurences of this, I want to throw in my two cents and  advocate for allowing the use of selected names with types, at least in -frelaxed.

And now I am going to fix this in 20 different vhdl files...