ghdl/ghdl #2850: Synthesis dynamic Set/Reset seems not handled correctly in process sensitivity list

toitoise, 2025-01-06:
**Description**
I wrote a basic VHDL entity that got a dynamic async Set/Reset flip-flop, based on an input signal (signal : state)
The final stage of the process is to check a PSL assertion with YOSYS formal engine, in order to check values at reset.

**Expected behaviour**
When "reset" asserted (reset=1) the flip-flop should either be in Set or Reset Async state based on the value of "state", and the output "out1" receives this value.

** Observed behavior **
When reset is asserted, the subsequents togglings of "state" signal (causing either Async Set or Reset) is not seen on "out1". 

**How to reproduce?**
--File: issue.vhd
```
LIBRARY IEEE;
USE IEEE.std_logic_1164.ALL;
USE IEEE.numeric_std.ALL;
USE IEEE.std_logic_misc.ALL;

ENTITY issue IS
    PORT(
        clk                               :   IN    std_logic;
        reset                             :   IN    std_logic;
        state                             :   IN    std_logic;    				
        in1                               :   IN    std_logic;     
        out1                         	  :   OUT   std_logic          
    );
END issue;

ARCHITECTURE rtl OF issue IS
begin
    out1_pc: process (all)
    begin 
        if (reset='1') then 
            out1 <= state;	-- not working as expected with Synth
        elsif (clk'EVENT AND clk = '1') THEN
            out1 <= in1;
        end if;
    end process;
END rtl;
```
**Command**
```
ghdl --synth --std=08 issue.vhd -e issue
```

**Context**
- OS: RHEL7
- Origin:
  - [* ] Released binaries: from oss_cad_suite (2025-01-03)  
       GHDL 5.0.0-dev (4.1.0.r453.g85927f052.dirty) [Dunoon edition]

**Possible cause**
The synthesis command generates a netlist and inside this netlist a wrapper for the Async Flip-Flop is used, but for me a signal is missing in the sensitivity list : 
```
  process (wrap_clk, wrap_reset)                 --  <==== Here "state" should be added or use "all" 
  begin
    if wrap_reset = '1' then
      n6 <= wrap_state;
    elsif rising_edge (wrap_clk) then
      n6 <= wrap_in1;
    end if;
  end process;
```
Regards

tgingold, 2025-01-06:
Ok, I will fix that.
But does such a gate exist ?

toitoise, 2025-01-07:
The result of synthesis seem good to me, it correctly uses an async set/reset flip-flop, and outside a basic glue logic with 2 ANDs and an inverter  to activate either Set or Reset pins when external Reset signal is asserted.

toitoise, 2025-01-07:
[image]

toitoise, 2025-01-07:
Originally, my problem came from a property saying  (PSL flavor) :
    assert always ({reset='1'} |-> { out1 = state} );
that failed.
Since I'm using oss-cad-suite to perform formal proof, does it mean the issue is also in the GHDL plugin used by yosys ?

commit 4d5fe294 (2025-01-07): netlists-disp_vhdl: handle reset value sensitivity for adff

Fix #2850
closed 2025-01-07 by tgingold