ghdl/ghdl #2851: simul-vhdl_compile.adb:2163 explicit raise

Quuxplusone, 2025-01-06:
**Description**

Disclaimer: I don't know anything about VHDL. I was just noodling around on https://tapec.uv.es/pardo/hdlsim/ with my zero knowledge of the subject, and hit an ICE which it said to report here, and I don't see anything already filed that resembles this, so here it is.

When I give the (presumably invalid) input below, I see GHDL crash with this error message:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.1.0 (4.0.0.r39.g7188e92cf) [Dunoon edition]
Compiled with GNAT Version: 10.5.0
Target: x86_64-linux-gnu
/userapp/
Command line:
/userapp/ghdl/bin/ghdl -c -fsynopsys /circuit.vhdl testbench.vhdl -r test_circuito_hdlsim2024
Exception PROGRAM_ERROR raised
Exception information:
raised PROGRAM_ERROR : simul-vhdl_compile.adb:2163 explicit raise
******************************************************************
```

**How to reproduce?**

Go to https://tapec.uv.es/pardo/hdlsim/ and load a VHDL file with these contents.

```vhd :file: ent.vhd
library IEEE;
use IEEE.STD_LOGIC_1164.all;
package NUMERIC_STD is
  type UNSIGNED is array (natural range <>) of STD_LOGIC;
  function "+" (L : UNSIGNED; R :  NATURAL) return UNSIGNED;
end NUMERIC_STD;

library ieee;
use ieee.std_logic_1164.all;
use work.numeric_std.all;

entity circuit is
port (hex1: out std_logic_vector(6 downto 0));
end circuit;

architecture description of circuit is
  signal counter: unsigned(7 downto 0):=x"00";

begin
  process begin
    counter<=counter+1;
  end process;
end description;
```

Unfortunately I don't know what it's doing behind the scenes with that file, sorry. But maybe this is enough information to help you reproduce and reduce the bug.

tgingold, 2025-01-10:
I can reproduce the crash with the reproducer, I am investigating.

tgingold, 2025-01-10:
There is no package body for numeric_std, which is the issue.  It shouldn't crash but report that as an error.

commit f92d2e39 (2025-01-10): vhdl-configuration: always check for missing package body

Fix #2851
closed 2025-01-10 by tgingold