ghdl/ghdl #2855: compile-xilinx-vivado.sh --all; Script Error Unfiltered Line

F2g-git, 2025-01-13:
**Description**
I need the unisim library in a module that I want to simulate. Hence I use the provided compilation script : compile-xilinx-vivado.sh.
When running the script I get the Error: 
```bash
ERROR: /tools/Xilinx/Vivado/2023.2/data/vhdl/src/unisims/secureip/BITSLICE_CONTROL.vhd:572:34:error: no declaration for "bitslice_control_v" (due to conflicts)
SCRIPT ERROR: Unfiltered line
Write ( Message, string'(BITSLICE_CONTROL_V'PATH_NAME));
SCRIPT ERROR: Unfiltered line
^                                         
```
for various lines of the BITSLICE_CONTROL.vhd file.

**Expected behaviour**
Successfull compilation of the xilinx vivado libraries

**How to reproduce?**
I would not know, I have no second PC for testing.

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS: Ubuntu 22.04 in as WSL2
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x ] Built from sources: commit ea6a2fb29855ca72d7bc3a3c34d528535625d62b 

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
I am using vivado 2023.2
I am using the mcode backend

tgingold, 2025-01-16:
Do you have a reproducer ?  I don't have vivado installed on my machine.

Paebbels, 2025-01-17:
I'm checking with Vivado 2023.1 and GHDL 5.0.0-dev in UCRT64 in Bash.

What I see so far:
* The GHDL output format of warnings (and errors) has changed, thus, the filtering/coloring rules need to be adjusted.
* The files are still containing VHDL coding bugs.
  * WARNING: C:/Xilinx/Vivado/2023.1/data/vhdl/src/unisims/primitive/SYSMONE4.vhd:1536:44:warning: prefix of array attribute must be an object name [-Wattribute]
  * WARNING: C:/Xilinx/Vivado/2023.1/data/vhdl/src/unifast/secureip/GTXE2_CHANNEL.vhd:34:1:warning: entity "gtxe2_channel" was also defined in file "C:/Xilinx/Vivado/2023.1/data/vhdl/src/unisims/secureip/GTXE2_CHANNEL.vhd" [-Wlibrary]
  * WARNING: C:/Xilinx/Vivado/2023.1/data/vhdl/src/unifast/secureip/GTHE2_CHANNEL.vhd:29:1:warning: entity "gthe2_channel" was also defined in file "C:/Xilinx/Vivado/2023.1/data/vhdl/src/unisims/secureip/GTHE2_CHANNEL.vhd" [-Wlibrary]
  * ERROR: C:/Xilinx/Vivado/2023.1/data/vhdl/src/unisims/secureip/BITSLICE_CONTROL.vhd:909:34:error: no declaration for "bitslice_control_v" (due to conflicts)
  * ...

**Further details:**
> SCRIPT ERROR: Unfiltered line

This has no effect on GHDL's outputs. It just reports filtering and formatting issues to the authors of the output filter script, so they can identify changed formatting and a need for a script adjustment.

If you want to avoid this, you can run the script with `-n`/`--no-filter` and/or in addition with `-N`/`--no-warnings`.

> ERROR: C:/Xilinx/Vivado/2023.1/data/vhdl/src/unisims/secureip/BITSLICE_CONTROL.vhd:909:34:error: no declaration for "bitslice_control_v" (due to conflicts)

This file has then no result from GHDL. If you want to get this fixed, you need to contact AMD (former Xilinx) and request them to write proper VHDL code :). OTOH, files like `BITSLICE_CONTROL.vhd` are either way from the encrypted `secureip` library and can't be simulated by GHDL.

-------
I'll look into updating the filtering algorithm.

F2g-git, 2025-01-23:
thanks for looking into it. I have also observed now, that this issue was not related to my other errors and is just an eyesore, albeit one that is not caused by ghdl directly.

Paebbels, 2025-02-17:
@F2g-git this will be fixed via #2896.

PR #2896 (2025-02-17): Reworked pre-compile Scripts
# New Features
* Implemented new Bash procedures for formatting and error checking to reduce code duplication.

# Changes
* Fixed color codes in `ansi_color.sh`.
* Transformed usage of `echo` to `printf`.

# Deprecations
* Removed pre-compile scripts for OSVVM.  
  &rarr; OSVVM requires complex handling of TCL code files. This can't be solved by a Bash-based file parser.
  &rarr; OSVVM is shipped with its own scripting environment to solve this usecase.
* Removed pre-compile scripts for Xilinx ISE.  
  &rarr; Xilinx ISE go its last update in September 2013 (ISE 14. [... cut]
closed 2025-02-17 by tgingold