ghdl/ghdl #2858: Accessing an alias to an external name fails quietly

am9417, 2025-01-16:
**Description**
External names (VHDL-2008) are not supported in GHDL. However, if an alias is created to point to an external name, the simulation fails quietly when running.

**Expected behaviour**
Error message in analyze, elaborate or run step to indicate what's wrong. Instead, nothing is printed and simulation stops.

**How to reproduce?**
Run the following minimal simulation where two aliases are created to point into a signal in the architecture and also one to an external name.

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity ghdl_sim is
end entity;

architecture rtl of ghdl_sim is
    signal test : std_logic := '0';
begin
end architecture;

library ieee;
use ieee.std_logic_1164.all;

entity example is
end example;

architecture sim of example is
    impure function get_and_print_some_string return string is
    begin
        report "get_and_print_some_string()";
        return "teststr";
    end;
    constant C_SOMETHING : string := get_and_print_some_string;
    signal arch_signal : std_logic := '1';
begin

    inst: entity work.ghdl_sim;
    test_p: process
        alias test_alias_ext is << signal .example.inst.test : std_logic >>;
        alias test_alias_int is arch_signal;
    begin
        report "Running";
        wait for 20 ns;
        --test_alias_ext <= '1';
        test_alias_int <= '0';
        assert false report "READY" severity failure;
    end process;

end sim;
```

Output before enabling line `test_alias_ext <= '1'`: works as expected
```
ghdl analyze --std=08 ghdl_sim.vhd; ghdl elaborate --std=08 work.example; ghdl run --std=08 work.example
ghdl_sim.vhd:21:9:@0ms:(report note): get_and_print_some_string()
ghdl_sim.vhd:33:9:@0ms:(report note): Running
ghdl_sim.vhd:37:9:@20ns:(assertion failure): Ready
./example:error: assertion failed
in process .example(sim).test_p
./example:error: simulation failed
```

After removing comment from that line.
```
ghdl analyze --std=08 ghdl_sim.vhd; ghdl elaborate --std=08 work.example; ghdl run --std=08 work.example
ghdl_sim.vhd:21:9:@0ms:(report note): get_and_print_some_string()
```

**Context**
`ghdl -v`

> GHDL 4.1.0 (4.0.0.r39.g7188e92cf) [Dunoon edition]
>  Compiled with GNAT Version: 10.5.0
>  GCC 12.1.0 code generator
> Written by Tristan Gingold.
> 
> Copyright (C) 2003 - 2024 Tristan Gingold.
> GHDL is free software, covered by the GNU General Public License.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

tgingold, 2025-01-17:
external names are only supported with mcode backends.

closed 2025-01-26 by tgingold

JvWesterveld, 2025-02-22:
I also ran into this. GHDL as built from last week's master branch still silently exits, at least with the LLVM back end, even though this issue is marked as completed.

temoTxt, 2026-10-11:
Re-tested on current builds (2026-10-11). Sharing a standalone repro in case it is useful. This looks like the same known limitation as #2610 / #3131 (external names unsupported by the LLVM and GCC code generators), and it matches the "fails quietly" symptom here. I am not asking for external-name support in those backends, but for a diagnostic instead of a silent abort.

**Result** (`--std=08`, official `ghdl/ghdl` Docker Hub images, Ubuntu 24.04, Linux x86_64)

| image tag (digest) | `ghdl --version` | `tb_control` | `tb_rel_proc` (relative name aliased in process) | `tb_abs_proc` (absolute name aliased in process) | `tb_rel_conc` (relative name, concurrent assignment) |
|---|---|---|---|---|---|
| `7.0.0-dev-llvm-ubuntu-24.04` (`sha256:272503cfdb68217ecfa1027bf9e9a072a6bd00fbfeca1409600f20c94ed0165b`) | 7.0.0-dev (6.0.0.r535.g2c451de76), LLVM | rc 0 | rc 255, no output | rc 255, no output | rc 1, `NULL access dereferenced`, `instance: .tb_rel_conc(sim).P0`, `error during elaboration` |
| `7.0.0-dev-gcc-ubuntu-24.04` (`sha256:ee0b964db2d4272189612faa03177c4c564113eaf49301954a5d95ee7d2ae238`) | same, GCC backend | rc 0 | rc 255, no output | rc 255, no output | rc 1, NULL access dereferenced |
| `6.0.0-llvm-ubuntu-24.04` (`sha256:8b3ec37c3873b2eee9387759e66c50830c15ae5b7b533badaa97ce007a0f8022`) | 6.0.0 (6.0.0.r0.ge589c698c), LLVM | rc 0 | rc 255 | rc 255 | rc 1 |
| `5.1.1-llvm-ubuntu-24.04` (`sha256:f24e71328d582b0cb0b8c75eba6ff795b01471957278c9ef7e8e1d8c245c38e6`) | 5.1.1 (4.1.0.r774.g8fb617d74), LLVM | rc 0 | rc 255 | rc 255 | rc 1 |
| `7.0.0-dev-mcode-ubuntu-24.04` (`sha256:a20ebdfaec1e568d6a0d3510b97fd48ec18e56887f789a9e7c6ba173c47691b9`) | 7.0.0-dev, mcode | rc 0 | rc 0, `sees 0101` | rc 0, `sees 0101` | rc 0, `sees 0101` |
| `7.0.0-dev-llvm-jit-ubuntu-24.04` (`sha256:22f1deb62ad7aeb61235302b40469f720cffebefb82ae99e4f56228e4cf2dbee`) | 7.0.0-dev, LLVM JIT | rc 0 | rc 0, `sees 0101` | rc 0, `sees 0101` | rc 0, `sees 0101` |
| `6.0.0-mcode-ubuntu-24.04` (`sha256:ee042751cf7ed4bf1d84fd0ef9278504012f69dad503214c99819b4f9388a4dd`) | 6.0.0, mcode | rc 0 | rc 0, `sees 0101` | rc 0, `sees 0101` | rc 0, `sees 0101` |

So the code is accepted by analysis, works with mcode and the LLVM JIT, and aborts with the LLVM and GCC code generators. The aliased-in-process case (the one in this issue) exits with status 255 and prints nothing at all, not even an error line. Expected output for the failing testbenches, as seen with mcode: `sees 0101` at 0 ms, then `done` at 1 ns.

**Reproduce**

```
ghdl -a --std=08 --workdir=$w leaf.vhd tb_rel_proc.vhd
ghdl --elab-run --std=08 --workdir=$w tb_rel_proc
echo $?
```

`leaf.vhd`:

```vhdl
library ieee; use ieee.std_logic_1164.all;
entity leaf is port (clk : in std_logic); end entity;
architecture rtl of leaf is
  signal cnt : std_logic_vector(3 downto 0) := "0101";
begin
end architecture;
```

`tb_rel_proc.vhd`:

```vhdl
library ieee; use ieee.std_logic_1164.all;
entity tb_rel_proc is end entity;
architecture sim of tb_rel_proc is
  signal clk    : std_logic := '0';
  signal mirror : std_logic_vector(3 downto 0);
begin
  dut : entity work.leaf port map (clk => clk);
  mon : process(clk)
    alias x is << signal dut.cnt : std_logic_vector(3 downto 0) >>;
  begin if rising_edge(clk) then report "sees " & to_string(x); end if; end process;
  stim : process begin clk <= '1'; wait for 1 ns; report "done"; wait; end process;
end architecture;
```

`tb_abs_proc.vhd` is identical except for the entity/architecture names and `alias x is << signal .tb_abs_proc.dut.cnt : std_logic_vector(3 downto 0) >>;`.

`tb_rel_conc.vhd` replaces the `mon` process with:

```vhdl
  mirror <= << signal dut.cnt : std_logic_vector(3 downto 0) >>;
  mon : process(clk) begin if rising_edge(clk) then report "sees " & to_string(mirror); end if; end process;
```

`tb_control.vhd` is the same testbench with a `mon` process that only reports `"tick"` (no external name); it runs fine everywhere.

**Suggestion.** Since the LLVM and GCC code generators do not support external names, could elaboration under those backends report a clear error (for example "external names are not supported by this code generator, use the mcode or JIT backend") instead of exiting silently with 255 or raising a NULL access? That would save users a lot of debugging, as the comments on #3131 and #2858 suggest.