ghdl/ghdl #2860: Review of ghdl/docker

Paebbels, 2025-01-16:
@tgingold: I'm collecting here my review on `ghdl/Docker` and if it's still needed.

**Build Containers**
GHDL is build locally in GitHub Actions without any Docker "overhead". It's a quick GCC + Ada + LLVM installation, which is installed once. Then the output results a generated and saved as an artifact for multiple deployments. On deployment is into a Ubuntu container.

As the build environment is used only once, there is no great advantage in using a compiler inside of Docker. Also, we don't want to preserve the build environment for next runs.

&rArr; not needed anymore

**Runtime Containers**
When GHDL build on Ubuntu is save in an artifact, the needed requirements are save in a file along side the GHDL files. This is used inside the container to install GHDL's dependencies. Again, this is done only once.

&rArr; not needed anymore

**Read-to-Use GHDL Container**
GHDL this container is provided by the new pipeline.

&rArr; replaced by direct containers builds in the main build and test pipeline of GHDL.

**Read-to-Use Extended GHDL Container**
Other containers containing VUnit, extended tooling and Co-Simulation tools/lib. While this is nice to have, I don't see the responsibility and maintenance work at GHDL. This is a container in the scope of `github.com/hdl`, because multiple EDA tools are integrated into one container.

&rArr; out of scope, should be transferred to e.g. `github.com/hdl`

-------------
How to proceed?  
What are your thoughts?

tgingold, 2025-01-17:
So, we can deprecate ghdl/docker, cannot we ?

Paebbels, 2025-01-17:
I think so too.

We could declare the repository as archived for 6..12 months, in case someone want to look at the Dockerfiles.

Maybe we find a way to move the vunit container. I'll ask in the vunit community.