ghdl/ghdl #2862: GHDL 5.0.0-dev: "exception raised: unknown reason" when generating --waves or --vcd

sideprojectslab, 2025-01-19:
**Description**
GHDL 5.0.0-dev mcode for win64: simulation of VHDL-2008 code runs like charm on its own but crashes with unknown exception when using the --waves or --vcd option.

I am launching GHDL from python (invoke) like this:

# this runs and passes the test
ctx.run(f"ghdl -c --std=08 -frelaxed {' '.join(SRC)} -r {TOPLEVEL}")

# this crashes with the only output: "exception raised: unknown reason"
ctx.run(f"ghdl -c --std=08 -frelaxed {' '.join(SRC)} -r {TOPLEVEL} --wave=waves.ghw")

This command works on other simulations of mine, but crashes on a specific one with this unknown exception.
I don't really know what is triggering the issue, I would like to know primarily how can I debug further, or get better error messages.

**Context**
- Version: 5.0.0-dev (mcode)
- OS: Windows 11
- Origin:
  - [ ] Package manager: `version`
  - [x ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

tgingold, 2025-01-26:
This is a known problem, use `--fst=` dumper.

dalex78, 2026-09-04:
# Issue #2862 — "exception raised: unknown reason" with `--wave`/`--vcd`

Analysis by an AI agent.

## High-level summary

A VHDL-2008 simulation that runs fine on its own crashed as soon as `--wave=` or `--vcd=` was added, and on Windows with the mcode backend the crash surfaced only as

```
exception raised: unknown reason
```

The reporter narrowed the trigger themselves — *"the problem seems to go away if I remove all the conditional generate statements"* — then edited the report to say that running the same thing on Linux gave a usable message and identified the real defect as #2640. Their own summary is that what is left specific to this issue is the unhelpful wording, not the crash.

tgingold's reply is the rest of the thread: *"This is a known problem, use `--fst=` dumper."*

## Current status: the crash is fixed

#2640's reproducer — an entity with `x: if FOO < 9 generate else generate end generate;` and nothing else — runs cleanly on current master, on all three backends, with `--wave=`, `--vcd=` and `--fst=`. The exit status is 0 and the dump files are written.

Since the crash is gone, the "unknown reason" wording has nothing left to report: it was the Windows mcode rendering of *that* crash, not a separate defect in the diagnostic machinery.

## What this commit does

`testsuite/gna/issue2862/` carries #2640's reproducer and simulates it with each of the three dumpers on every backend, so that the fix does not silently regress. There is no source fix here: there is nothing left to fix.

## Recommendation

Close #2862 as fixed (with #2640).

closed 2026-09-05 by tgingold