ghdl/ghdl #2865: Crash on PSL assert with long boolean AND equation

amb5l, 2025-01-20:
GHDL crashes when I use a PSL assert with a long boolean AND equation, but does not when the components of the equation are separated into individual assertions:
```
synth_psl_expr: cannot handle N_IMP_BOOL

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r479.gc319b79b5.dirty) [Dunoon edition]
Compiled with GNAT Version: 10.5.0
Target: x86_64-linux-gnu
/mnt/d/work/tyto2/test/memac/tb_memac_rmii_rx/sby/memac_rmii_rx/src/
Command line:
/tools/oss-cad-suite/libexec/ghdl --synth --std=08 memac_rmii_rx.vhd memac_rmii_rx.psl -e memac_rmii_rx
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : psl-errors.adb:37
******************************************************************
```

Basically, this works...
```
  assert always rst = '1' -> s1_crs_dv = '0';
  assert always rst = '1' -> s1_er     = '0';
  assert always rst = '1' -> s1_d      = "00";
```
...and this does not:
```
  assert always rst = '1' ->
    s1_crs_dv = '0'   and
    s1_er     = '0'   and
    s1_d      = "00"  and ...
```

VHDL and PSL vunit sources attached in [memac_rmii_rx.zip](https://github.com/user-attachments/files/18478042/memac_rmii_rx.zip).

To reproduce: `ghdl --synth --std=08 memac_rmii_rx.vhd memac_rmii_rx.psl -e memac_rmii_rx`

Commenting out lines 41-71 of `memac_rmii_rx.psl` resolves the error.

This is low priority because I have a workaround (see lines 7-36 of `memac_rmii_rx.psl`).

I am running GHDL on WSL Ubuntu 24.04.1 LTS as bundled with [oss-cad-suite](https://github.com/YosysHQ/oss-cad-suite-build) release [2025-20-01](https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2025-01-20).

tmeissner, 2025-01-20:
Only a hint or tip, precisely spoken, this:

```vhdl
-- clock
default clock is rising_edge(clk);

-- asynchronous reset (multiple assertions)
assert always rst = '1' -> s1_crs_dv = '0';
```

checks for synchronous reset. I always use a simple VHDL assertion for asynchronous reset checks, for example:

```vhdl
process (all) is
begin
  if (rst) then
    assert s1_crs_dv = '0';
  end if;
end process;
```

However, I never checked if the resulting netlist in yosys is different to the check with PSL and default clock ...

closed 2025-01-20 by tgingold

amb5l, 2025-01-20:
@tgingold Thankyou for such a fast response!

@tmeissner Thanks for your input, I have seen some of your work with GHDL and PSL. Your simple VHDL assert is a great idea and I will use it. I have found that my PSL assert is "good enough". For example, if **my_reg** has an asynchronous reset, then this assertion...
`assert always rst = '1' -> my_reg = '1';`
...will cause BMC to fail, producing a counterexample waveform showing a single clock cycle, with **rst** = '1' and **my_reg** = '0' throughout.

commit 5e2495b1 (2025-01-20): synth_psl_expression: handle N_Imp_Bool

Fix #2865