ghdl/ghdl #2868: Record fields not being written to?

F2g-git, 2025-01-24:
**Description**
I was playing around with record data types and realized that I can not properly write data to a record.
As a minimal example i implemented a counter and wrote its value to a record field.
The bits that were non zero in the counter value were represented as X in the record, while others were still 0, see my MWE.
**Expected behaviour**
The fields of a record should take the value assigned to them.

**How to reproduce?**

```vhd :file: basic_test.vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity basic_test is
end entity basic_test;

architecture rtl of basic_test is
 type axi_stream_32_type is record
  tdata : std_logic_vector(31 downto 0);
  tvalid : std_logic;
 end record axi_stream_32_type; 
 
 signal clk : std_logic := '0';
 signal tdata : std_logic_vector(31 downto 0 ) := (others => '0'); 
 signal tvalid : STD_LOGIC := '0';
 signal w_axi : axi_stream_32_type := (tdata => (others => '0'), tvalid => '0') ;
 signal start_counter : std_logic := '0';
begin
  clk <= not clk after 5 ns;
 proc_name: process
  begin
        wait until rising_edge(clk);
        wait for 1000 ns;
        w_axi.tdata <= (others => '0');
        wait for 300 ns;
  end process proc_name; 
  
  tdata_counter : process(clk)
  begin
    if rising_edge(clk) then
      start_counter <= '1';
      tdata <= std_logic_vector( unsigned(tdata) + 1);
      if tdata(3) = '1' then
        tvalid <= '1';
      else 
        tvalid <= '0';
      end if;
    end if;
  end process;
  w_axi.tdata <= tdata;
  w_axi.tvalid <= tvalid;
  checker_process : process(clk)
  begin
    if rising_edge(clk) then
      if w_axi.tvalid = '1' then
      assert w_axi.tdata = tdata report "The record field did not match its source" severity FAILURE;
      end if;
    end if;
  end process;

end architecture rtl;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 basic_test.vhd   
ghdl -r --std=08  basic_test --wave=test.ghw --stop-time=1500ns 
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ x ] Built from sources: ea6a2fb29855ca72d7bc3a3c34d528535625d62b

diogratia, 2025-01-24:
A concurrent assignment statement represents an equivalent process. Each process with an assignment to a signal has a driver for that signal. Within the scope of the declaration of a signal the value of the signal is the resolved value of the drivers of that signal and any ports representing contributions to the value of the net the signal belongs to, dangling participles aside.

Here you have drivers for the elements of w_axi.tdata  in process proc_name and in the concurrent assignment statement.

F2g-git, 2025-01-27:
Hey thank you for picking that one up. I screwed up when destilling my test case down to an MWE. I made a slightly less minimal one and updated my post above to reflect that.

diogratia, 2025-01-27:
Here's what fixing the slave  s_axi.tdata and m_axi_tdata **inout** issues gives us:

[image]

Something similar needs to be done for the tvalid elements.

This has to do with using mode **inout** and  interfacing an entire record as well as how nets are resolved. A port signal that's of a record type has all the elements of that record type. The out portion of an **inout** mode port will have a default value. When not driven an **inout** port default value of 'Z' can be resolved to the value of any remaining ports and drivers on the net.

Without intending criticism your **Expected behaviour** is faulty here. I'm not yet awake enough in my time zone to provide references to the standard to explain the behavior (different dictionary spelling aside).

closed 2025-05-13 by tgingold