ghdl/ghdl #2869: -R in Makefile breaks build on NetBSD

mayureshw, 2025-01-27:
> -R breaks the build on NetBSD. libgnat resides in something like /usr/pkg/gcc10-aux/lib/gcc/x86_64--netbsd/10.3.0/adalib. Hence rpath is necessary. 

 _Originally posted by @mayureshw in [#2806](https://github.com/ghdl/ghdl/issues/2806#issuecomment-2616107952)_

Paebbels, 2025-01-27:
@mayureshw please be more specific.

What is not working and was is expected?  
Please provide error logs.

mayureshw, 2025-01-28:
The error is the following. It comes from pkgsrc - the packging system on NetBSD. But it may not convey the full story.

```
=> Checking for missing run-time search paths in ghdl-dev-5.0.0.9860
ERROR: lib/libghdl-5_0_0_dev.so: missing library: libgnat-10.so
```
The path to libgnat being, sort of `unusual' (see first post), it needs to be embedded in the library (.so) using rapth. Use of rpath is a very common practice in many build systems. Addition of '-R' prevents that from happening. I think this is too aggressive a fix for #2806.