ghdl/ghdl #2872: [Feature/Question] compile-*.sh scripts - can the run in parallel?

axman6, 2025-01-29:
**Is your feature request related to a problem? Please describe.**
I'm currently writing a Nix wrapper to compile Xilinx's unisim library using the `compile-xilinx-vivado.s` script. It currently takes a very long time to compile the output files so I was wondering if there was any way to rewrite the [`Compile`](https://github.com/ghdl/ghdl/blob/master/scripts/vendors/shared.sh#L308) to run the analysis step in parallel.

**Describe the solution you'd like**
`xargs` feels like the simplest solution, where instead of just calling GHDL directly in the script, the arguments would be written to a file and then that would be fed into xargs. The main problem I see here is how to deal with the error handling in that's currently done in `AnalyzeVHDL`. This could possible be alleviated by having each call to GHDL write its exit code to a unique template file, and then checking the files after the xargs call.

**Describe alternatives you've considered**
GNU Parallel is also designed for exactly this sort of task, but it's less likely to be installed on most systems.

It might also be possible to do this with build in shell job control, but managing parallelism/resource usage would be much harder (my initial thought was to just do something like:

```shell
...
Compile() {
	local SourceDirectory=$1
	local VHDLLibraries=$2

	for VHDLLibrary in $VHDLLibraries; do
		local LibraryName="${VHDLLibrary}_LibraryName"; local LibraryName=${!LibraryName}
		local LibraryPath="${VHDLLibrary}_LibraryPath"; local LibraryPath=${!LibraryPath}
		local VHDLVersion="${VHDLLibrary}_VHDLVersion"; local VHDLVersion=${!VHDLVersion}
		local Files="${VHDLLibrary}_Files[*]";          local Files=${!Files}

		echo -e "${ANSI_LIGHT_CYAN}Analyzing library '$LibraryName'...${ANSI_NOCOLOR}"

		(CreateVHDLLibrary $LibraryName $LibraryName $VHDLVersion && \
		    AnalyzeLibrary $LibraryName "$SourceDirectory" "$LibraryPath" "$Files") &
	done
}
 but that would spawn far too many jobs at once).

tgingold, 2025-01-29:
If you use the mcode backend, it would be much faster!

Don't try to do analysis in parallel, it will either corrupt the library or forget some files.

Paebbels, 2025-01-29:
Besides ghdl writing in parallel into the *.cf files (not thread safe as Tristan said), vhdl files and more over entities/packages have dependencies. We barely know the linear order. But a full dependency graph is unknown. Without it you can't know what could run in parallel.

Maybe whole libraries could run in parallel like unisim vs. xpm. This could be checked by running each library independently in a fresh directory. If no complaints, then there are no cross-references.

tgingold, 2025-01-30:
Please, don't try to combine `.cf` files: the format will change, and there are indexes which should be unique.

I am not sure that simulating with llvm will be much faster than the mcode backend.  If you have testcases, I am very interested.

The correct way to do parallel compilation would be to first analyze without code generation (but update of the `.cf` library file) and then, in parallel, generating the corresponding code (without touching the library file).  But this is not implemented!