ghdl/ghdl #2873: Return error code on failure

ecstrema, 2025-01-29:
## Description

When you run a testbench with failing assertions, I would expect the command to exit with a non-zero exit code. It does not.

## Expected behaviour
Return 1, or the error code

## Reproduction

```vhd :file: ent.vhd
entity ent is
end entity;

architecture a of ent is
begin
  process begin
    report "Error world" severity error;
    wait;
  end process;
end;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
echo $?
```
- OS: Windows
- Origin:
  - [x] Package manager(winget): ghdl 4.1.0
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

ecstrema, 2025-01-29:
Edit: I see the way this is handled in VHDL is to stop simulation on the first error.

Can I keep simulating and report at the end whether or not there was an error?

I'll change the issue title consequently

diogratia, 2025-01-30:
The assertion level used to end simulation can be modified on the run time command line:

[\--assert-level=\<LEVEL\>](https://ghdl.github.io/ghdl/using/Simulation.html#cmdoption-ghdl-assert-level)  

 >Select the assertion level at which an assertion violation stops the simulation. LEVEL is the name from the severity_level enumerated type defined in the standard package or the none name.

 >>By default, only assertion violation of severity level failure stops the simulation.

 >>For example, if LEVEL was warning, any assertion violation with severity level warning, error or failure would stop simulation, but the assertion violation at the note severity level would only display a message.

 >>Option [--assert-level=none](https://ghdl.github.io/ghdl/using/Simulation.html#cmdoption-ghdl-assert-level) prevents any assertion violation from stopping simulation.

By default only assert levels of failure will end simulation. An example of setting the ghdl exit status:

[Simulation Options](https://ghdl.github.io/ghdl/using/Simulation.html#simulation-options)

>In most system environments, it is possible to pass CLI options while invoking a program. Contrary to most programming languages, there is no standard method in VHDL to obtain the arguments or to set the exit status. However, the GHDL runtime behaviour can be modified with some options:

>>・It is possible to pass parameters to your design through the generic interfaces of the top entity.

 >>・It is also possible to stop simulation after a certain time. The exit status of the simulation is EXIT_SUCCESS (0) if the simulation completes, or EXIT_FAILURE (1) in case of error (assertion failure, overflow or any constraint error).

The [ghdl documentation](https://ghdl.github.io/ghdl/index.html) provides the answers. The first bit on assertion level was found after invoking `ghdl --help-run` and using the search term *--assert-level*. The second portion of the documentation found with the search term *exit status*.

An example implementation to provide a failure after accumulating errors:

```vhdl

entity ent is
end entity;

architecture a of ent is
begin
    process
         variable num_errors:  natural;  -- default 'left 0
    begin
        for i in 0 to  3 loop
            wait for 10 ns;
            num_errors := num_errors + 1;
            report "number of errors = " & integer'image(num_errors) 
                severity error;
        end loop;
        wait for 100 ns;
        if num_errors /= 0 then
            report "errors = " &  integer'image(num_errors)
            severity FAILURE;
        end if;
    end process;
end architecture;
```

which produced:
```none
%: ghdl -r ent
ent.vhdl:14:13:@10ns:(report error): number of errors = 1
ent.vhdl:14:13:@20ns:(report error): number of errors = 2
ent.vhdl:14:13:@30ns:(report error): number of errors = 3
ent.vhdl:14:13:@40ns:(report error): number of errors = 4
ent.vhdl:19:13:@140ns:(report failure): errors = 4
ghdl:error: report failed
ghdl:error: simulation failed
%: 
```

Note the invocation of the simulation run time did not modify the assertion level.

ecstrema, 2025-01-30:
Oh I see. So the testbenches have to record if an error ocurred, and then report a failure at the end if not. 

Although cumbersome, it is definetely a worthy workaround.

ecstrema, 2025-01-30:
(Thank you for the detailed answer by the way!)

diogratia, 2025-01-31:
There is a mechanism to count syntactical errors during analysis and defer exit based on a threshold account and any syntactical or semantic error insures there is no update to analyzed design units stored in design libraries. There is no current  mechanism in the simulation kernel to defer the effects of detecting severity level in assertion or report statements. It'd also be complicated by the ability to not continue execution based on severity level. So maybe along the lines of `--exit-failure-on-severity-threshold=N` along with counting severity level report occurrences specified by `--assert-level=LEVEL` and a value of 0 for N would give the behavior you desire. 

You could make this a feature request otherwise your **Expected  behaviour** doesn't match the current ghdl implementation. 

The author of ghdl likely expected  the command line severity level would be used to limit the volume of diagnostic output during debugging allowing attention to each severity level in turn.  The behavior is outside the scope of the standard and is implementation detail. See for example IEEE Std 1076-2008 10.4 Report statement the NOTE and the preceding paragraph (notes are informative and not normative):

>An implementation should continue execution of a model after displaying a report message in which the severity level is NOTE, WARNING, or ERROR.  

>NOTE—An implementation may choose whether or not to continue execution of a model after execution of report statements with various severity levels. It may also give tool users ability to control simulator actions for report statements of various severity levels via mechanisms not specified by this standard.

I'd expect your desired usage to be particular to your workflow. `--exit-failure-on-severity-threshold=N` and `--assert-level=LEVEL` together might make the feature a bit more generally useful. I come from a generation of UNIX users who might instead grep for ERROR messages in console output and use the return status from that for control flow (and noting error message formats might be implementation dependent but that can be managed too).

ecstrema, 2025-01-31:
grep'ing ERROR messages is definetely doable, though once again, cumbersome. It has the advantage over the solution above that you write it once and do not need to add specific code in every testbench.

`--exit-failure-on-assert-level=<severity>` is, you nailed it, what I would need.