ghdl/ghdl #2876: Wrong assignment during synthesis in array of complex records

egorxe, 2025-02-01:
**Description**
In some cases when synthesizing code with arrays of records wrong record member is assigned. Same code works fine in behavior simulation, but fails if simulated after passing through ghdl --synth or through yosys plugin.

**Expected behaviour**
In provided example after synthesis v(0).x is reassigned in STATE1 when it should keep its value.

**How to reproduce?**

Provided VHDL code runs fine in simulation mode:
```
> ghdl -a --std=08 ./record_bug.vhd record_bug_tb.vhd && ghdl elab-run --std=08 work.record_bug_tb
record_bug_tb.vhd:30:5:@100ns:(report note): Test OK!
simulation finished @100ns
```

But it fails if simulated after synthesis:
```
> ghdl -a --std=08 ./record_bug.vhd && ghdl --synth --std=08 record_bug > record_bug_synth.vhd && ghdl -a --std=08 ./record_bug_synth.vhd record_bug_tb.vhd && ghdl elab-run --std=08 work.record_bug_tb
record_bug_tb.vhd:29:5:@100ns:(assertion failure): Wrong test signal value BB
./record_bug_tb:error: assertion failed
  instance: .record_bug_tb(arch).P2
./record_bug_tb:error: simulation failed
```

Also it fails the same way if converted to Verilog using Yosys and yosys ghdl plugin and simulated by Verilog simulator.

Problematic source for synthesis (record_bug.vhd):
```vhd :file: record_bug.vhd
library ieee;
use ieee.std_logic_1164.all;

entity record_bug is
    port (
        clk_i       : in  std_logic;
        rst_i       : in  std_logic;
        test_o      : out std_logic_vector(7 downto 0)
    );
end record_bug;

architecture arch of record_bug is

constant ZERO       : std_logic_vector(7 downto 0) := (others => '0');
constant ELEMENTS   : integer := 2; -- bug happens for array elements number > 1 && < 6

type array_type is array (0 to ELEMENTS-1) of std_logic_vector(7 downto 0);

type record_type is record
    x   : std_logic_vector(7 downto 0);
    a   : array_type;
end record;

type record_array_type is array (0 to ELEMENTS-1) of record_type;   

type state_type is (STATE0, STATE1, STATE2);

signal state    : state_type;
signal i        : integer range 0 to ELEMENTS-1;
signal j        : integer range 0 to ELEMENTS-1;
signal v        : record_array_type;

begin

test_o <= v(0).x;   -- should be AA after reset deassertion, but in synth mode BB is assigned in STATE1

process(clk_i)
begin
    if Rising_edge(clk_i) then
        if rst_i = '1' then
            state   <= STATE0;
            v       <= (others => (ZERO, (others => ZERO)));
            i       <= 0;
            j       <= 0;
        else
            case state is
                when STATE0 =>
                    v(0).x  <= X"AA";   -- assignment of other elements here does not affect buggy behavior
                    state   <= STATE1;
                    
                when STATE1 =>
                    v(i).a(j) <= X"BB"; -- v(0).x should not be touched here
                    
                    if i /= ELEMENTS-1 then
                        i <= i + 1;
                    else
                        if j /= ELEMENTS-1 then
                            j <= j + 1;
                            i <= 0;
                        else
                            state <= STATE2;
                        end if;
                    end if;
                    
                when STATE2 =>
                    null;
            end case;
        end if;
    end if;
end process;

end arch;
```

Testbench (record_bug_tb.vhd):
```vhd :file: record_bug_tb.vhd
library ieee;
use ieee.std_logic_1164.all;
use std.env.finish;

entity record_bug_tb is
end record_bug_tb;

architecture arch of record_bug_tb is

signal clk  : std_logic  := '0';
signal rst  : std_logic  := '1';
signal test : std_logic_vector(7 downto 0);

begin

uut : entity work.record_bug
    port map (
        clk_i   => clk,
        rst_i   => rst,
        test_o  => test
    );

clk <= not clk after 5 ns;
rst <= '0' after 20 ns;

process
begin
    wait for 100 ns;
    assert test = X"AA" report "Wrong test signal value " & to_hstring(test) severity failure;
    report "Test OK!";
    finish;
end process;

end arch;
```

Synth and test script:
```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 ./record_bug.vhd
ghdl --synth --std=08 record_bug > record_bug_synth.vhd
ghdl -a --std=08 ./record_bug_synth.vhd record_bug_tb.vhd
ghdl elab-run --std=08 work.record_bug_tb
```

**Context**

- OS: Debian 12
- Origin:
  - Built from sources: fails on both 4.1.0 release and latest master 5af853d.

egorxe, 2025-02-02:
Thank you for the prompt fix! But I'm afraid that offset calculation is still not correct. Here is slightly changed testcase (checking another record value) which fails after synth on 8546971. Testbench is the same.

```vhd :file: record_bug2.vhd
library ieee;
use ieee.std_logic_1164.all;

entity record_bug is
    port (
        clk_i       : in  std_logic;
        rst_i       : in  std_logic;
        test_o      : out std_logic_vector(7 downto 0)
    );
end record_bug;

architecture arch of record_bug is

constant ZERO       : std_logic_vector(7 downto 0) := (others => '0');
constant ELEMENTS   : integer := 3; 

type array_type is array (0 to ELEMENTS-1) of std_logic_vector(7 downto 0);

type record_type is record
    x   : std_logic_vector(7 downto 0);
    a   : array_type;
end record;

type record_array_type is array (0 to ELEMENTS-1) of record_type;   

type state_type is (STATE0, STATE1, STATE2);

signal state    : state_type;
signal i        : integer range 0 to ELEMENTS-1;
signal j        : integer range 0 to ELEMENTS-1;
signal v        : record_array_type;

begin

test_o <= v(0).a(1);   -- should be AA after STATE1 deassertion

process(clk_i)
begin
    if Rising_edge(clk_i) then
        if rst_i = '1' then
            state   <= STATE0;
            v       <= (others => (ZERO, (others => ZERO)));
            i       <= 0;
            j       <= 0;
        else
            case state is
                when STATE0 =>
                    v(0).x  <= X"BB";   
                    state   <= STATE1;
                    
                when STATE1 =>
                    v(i).a(j) <= X"AA";
                    
                    if i /= ELEMENTS-1 then
                        i <= i + 1;
                    else
                        if j /= ELEMENTS-1 then
                            j <= j + 1;
                            i <= 0;
                        else
                            state <= STATE2;
                        end if;
                    end if;
                    
                when STATE2 =>
                    null;
            end case;
        end if;
    end if;
end process;

end arch;
```

tgingold, 2025-02-03:
Not yet fully fixed.

egorxe, 2025-02-03:
> Not yet fully fixed.

Just in case, here is slightly more complex testcase which still fails and also covers first two.

To synth:
```vhd :file: record_bug3.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity record_bug is
    port (
        clk_i       : in  std_logic;
        rst_i       : in  std_logic;
        x_o         : out std_logic_vector(7 downto 0);
        a_o         : out std_logic_vector(7 downto 0);
        i_o         : out std_logic_vector(7 downto 0);
        j_o         : out std_logic_vector(7 downto 0)
    );
end record_bug;

architecture arch of record_bug is

constant ZERO       : std_logic_vector(7 downto 0) := (others => '0');
constant ELEMENTS   : integer := 3; 
constant ELEMENTS2  : integer := 5; 

type array_type is array (0 to ELEMENTS2-1) of std_logic_vector(7 downto 0);

type record_type is record
    x   : std_logic_vector(7 downto 0);
    a   : array_type;
end record;

type record_array_type is array (0 to ELEMENTS-1) of record_type;   

type state_type is (STATE0, STATE1, STATE2);

signal state    : state_type;
signal i        : integer range 0 to ELEMENTS-1;
signal j        : integer range 0 to ELEMENTS2-1;
signal v        : record_array_type;

begin

x_o     <= v(i).x;      -- should be i
a_o     <= v(i).a(j);   -- should be (i+1)*(j+1)
i_o     <= std_logic_vector(to_unsigned(i,8));
j_o     <= std_logic_vector(to_unsigned(j,8));

process(clk_i)
begin
    if Rising_edge(clk_i) then
        if rst_i = '1' then
            state   <= STATE0;
            v       <= (others => (ZERO, (others => ZERO)));
            i       <= 0;
            j       <= 0;
        else
            case state is
                when STATE0 =>
                    -- set all x values at the same time
                    for k in 0 to ELEMENTS-1 loop
                        v(k).x  <= std_logic_vector(to_unsigned(k,8)); 
                    end loop;
                    state   <= STATE1;
                    
                when STATE1 =>
                    -- cycle to fill all a values with (i+1)*(j+1)
                    v(i).a(j) <= std_logic_vector(to_unsigned((i+1)*(j+1),8));
                    
                    if i /= ELEMENTS-1 then
                        i <= i + 1;
                    else
                        if j /= ELEMENTS2-1 then
                            j <= j + 1;
                            i <= 0;
                        else
                            state <= STATE2;
                            j <= 0;
                        end if;
                    end if;
                    
                when STATE2 =>
                    -- cycle infinitely for verification
                    if i /= ELEMENTS-1 then
                        i <= i + 1;
                    else
                        if j /= ELEMENTS2-1 then
                            j <= j + 1;
                            i <= 0;
                        else
                            j <= 0;
                        end if;
                    end if;
            end case;
        end if;
    end if;
end process;

end arch;
```

Testbench:
```vhd :file: record_bug_tb3.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.env.finish;

entity record_bug_tb is
end record_bug_tb;

architecture arch of record_bug_tb is

signal clk  : std_logic  := '0';
signal rst  : std_logic  := '1';
signal x, a : std_logic_vector(7 downto 0);
signal i, j : std_logic_vector(7 downto 0);

constant PERIOD : time := 10 ns;

begin

uut : entity work.record_bug
    port map (
        clk_i   => clk,
        rst_i   => rst,
        x_o     => x,
        a_o     => a,
        i_o     => i,
        j_o     => j
    );

clk <= not clk after PERIOD/2;
rst <= '0' after PERIOD*4;

process
begin
    wait for PERIOD*100;
    for k in 0 to 100 loop
        assert x = i 
        report "Wrong 'x' signal value " & to_hstring(x) & " for " & to_hstring(i)
        severity failure;
        
        assert to_integer(unsigned(a)) = (to_integer(unsigned(i))+1)*(to_integer(unsigned(j))+1) 
        report "Wrong 'a' signal value " & to_hstring(a) & " for " & to_hstring(i) & "," & to_hstring(j) 
        severity failure;
        
        wait for PERIOD;
    end loop;
    report "Test OK!";
    finish;
end process;

end arch;

```

commit 0542b473 (2025-02-02): generate_muxes: fix offset for multidim dyn_inst

Fix #2876
commit 9375ae63 (2025-02-04): testsuite/synth: add a test for #2876

Close #2876
closed 2025-02-04 by tgingold