ghdl/ghdl #2878: Segmentation fault on Array of Array access

JimLewis, 2025-02-05:
**Description**
Segmentation fault on Array of Array access.   The error often does not manifest until calling std.env.stop.   If I remove the call to stop, it does not fail.    

**Expected behaviour**
Not error

**How to reproduce?**
```vhd :file: TbAlertLog_print_count_1.vhd
library IEEE ;
  use ieee.std_logic_1164.all ;
  use ieee.numeric_std.all ;

  use std.textio.all ;
  use ieee.std_logic_textio.all ;

library osvvm ;
  context osvvm.OsvvmContext ;

entity TbAlertLog_print_count_1 is
end TbAlertLog_print_count_1 ;
architecture t40_print_count of TbAlertLog_print_count_1 is

  -- From AlertLogPkg
  --   subtype  AlertLogIDType           is integer ;

  type AlertLogIDArrayType is array (integer range <>) of AlertLogIDType ;
  type AlertIDMatrixType is array (1 to 2) of AlertLogIDArrayType(1 to 4) ;
begin

  TestProc : process 
    variable ProcID : AlertIDMatrixType ; 
    constant AlertLevel : AlertType := ERROR ; 
  begin
  
    SetTestName("TbAlertLog_print_count_1") ; 
    SetLogEnable(PASSED, TRUE) ; 

    ProcID(1)(1) := NewID("Proc_1 ", ALERTLOG_BASE_ID) ; 
    ProcID(1)(2) := NewID("P1_2 FE", ALERTLOG_BASE_ID) ; 
    
--    SetAlertPrintCount(ProcID(1)(2), AlertLevel, 2) ;  
--    AffirmIfEqual(GetAlertPrintCount(ProcID(1)(2), AlertLevel ), 2,             "P1.2  " & to_string(AlertLevel)) ;  
--    AffirmIfEqual(GetAlertPrintCount(ProcID_12, AlertLevel ), 2,             "P1.2  " & to_string(AlertLevel)) ;  
--    Alert(ProcID(1)(2), "P1.2    Max: 2,  " & to_string(AlertLevel),  AlertLevel) ; 

    AffirmIf(TRUE, "Token Pass") ; 
   
    EndOfTestReports (ExternalErrors => (ERROR => 0, others => 0)) ; 

    std.env.stop; 
    wait ;  
    
  end process TestProc ;
end t40_print_count ;
```
If I create the AlertLogIDs individually with individual variables and then assign them using 
```
ProcID := := ( (ProcID_11, ProcID_12, ProcID_13, ProcID_14), (ProcID_21, ProcID_22, ProcID_23, ProcID_24)) ; 
```
Then this kicks the can down the road until the `AffirmIfEqual(GetAlertPrintCount(...)`.   Here the one that uses indicies causes the segmentation fault - again at std.env.stop (and not without it).  However, if I replace the indicies with ProcID_12, then it works.  I am at a loss to understand.    

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a TbAlertLog_print_count_1.vhd
ghdl --elab-run TbAlertLog_print_count_1 
```

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

GHDL 5.0.0-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 14.2.0
 static elaboration, mcode JIT code generator
Written by Tristan Gingold.

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Error: %%      0 ns    Log    PASSED    in Default,   Token Pass
Error: %%      0 ns    DONE   PASSED   TbAlertLog_print_count_1  Passed: 1  Affirmations Checked: 1
Error: simulation finished @0ms
Error: child killed: segmentation violation...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2025-02-05:
On which OS ?

JimLewis, 2025-02-05:
MSYS2 on windows 10.   Running yesterday's nightly:  mingw-w64-x86_64-ghdl-mcode-5.0.0.dev-1-any.pkg.tar.zst

JimLewis, 2025-02-05:
I removed the std.env.stop from 5 of my regression tests last night and all of them ran correctly.   So it is something about the usage of the array of array and then calling std.env.stop.

tgingold, 2025-02-06:
Can you try again with a nightly build which includes commit c749342 ?  I have fixed a very similar regression.

(Not sure why it worked before...)

Paebbels, 2025-02-06:
I moved `nightly` to c74934214c1d9ad4ef9deaf1b2fb19de007379ef. It should finish in 30..40 minutes.

Pipeline: https://github.com/ghdl/ghdl/actions/runs/13173204513
Release: https://github.com/ghdl/ghdl/releases/tag/nightly (when pipeline finished)

JimLewis, 2025-02-06:
Curiously if change ProcID to a signal, it no longer fails.   It was bugging me that more of my regressions were not doing this, so I had to look further at it.

tgingold, 2025-02-08:
For me this is fixed using head.  Please reopen if it is not working for you.

closed 2025-02-08 by tgingold