ghdl/ghdl #2879: Analyzed files from GHDL (LLVM backend) can't be moved via artifacts from one GitLab runner to another

Paebbels, 2025-02-06:
**Description**
The current state of investigation shows that generated *.o files or *.cf files can't be moved via GitLab artifacts from a pre-compile job to a new job (Docker based runner instance). GHDL reports that certain design units can't be found. For example package bodies are claimed to be not found.

*Pipeline Idea:*
1. OSVVM
   1. checkout main repository, wherein OSVVM and PoC are Git submodules
   2. analyze OSVVM files using GHDL (LLVM) and OSVVM's TCL scripting
   3. package working directory of OSVVM (`./temp`) as artifact
2. PoC Library
   1. checkout repository, wherein OSVVM and PoC are Git submodules
   2. download and restore artifact in `./temp`
   3. analyze PoC files (which reference OSVVM) using GHDL (LLVM) and OSVVM's TCL scripting
   4. package working directory of OSVVM + PoC (`./temp`) as artifact
3. DUT
   1. checkout repository, wherein OSVVM and PoC are Git submodules
   2. download and restore artifact in `./temp`
   3. analyze DUT files (which references PoC and OSVVM) using GHDL (LLVM) and OSVVM's TCL scripting
   4. run DUT's simulation using GHDL (LLVM) and OSVVM's TCL scripting
   5. package OSVVM's reports as artifact

When the DUT is simulated using OSVVM's scripts, GHDL complains a package body from OSVVM is missing.

*What was checked so far?*
* The content of the working directory (`./temp`) is properly transferred including `*.o` and `*.cf` files.  
  => directory structure looks as expected
* `ghdl` analyzes all files from OSVVM without errors.  
  => no errors
* `ls` and `ghdl --dir OSVVM` was used before artifact collection to check if the missing package body was analyzed.  
  => package body is listed
* `ghdl --dir OSVVM` was used after artifact restoration to check if the missing package body was transferred to the new place.  
  => package body is listed
* When running that sequence of OSVVM scripts on the same machine without transferring the pre-compiled files, the package body is found. (Sequence of scripts means: There are 3 TCL scripts with OSVVM script commands, launched in 3 independent tclsh sessions.)

*What's different for that package to others?*  
The package and the body reside in two VHDL files and are analyzed separately. Because of scripting and two independent GHDL calls, there is a tiny delay.

**Expected behavior**
Transferring pre-compiled libraries should work, so pipeline time can be reduced.

**How to reproduce?**
Before creating a full blown reproducer, I wanted to ask and clarify, if such a transfer should be possible and how contents in the `*.cf` files are effected.

*Possible differences between CI jobs:*
* The repository is checked out multiple times on different VMs / Docker instances.
* The working directory might not be identical between CI jobs.
* The file dates might not be identical due to Git checkouts at different times.
* The file dates of restored artifact contents might be different.

What aspects are used to calculate if a package body exists?  
Please note, the error message didn't report a design unit is outdated.

**Context**
Please, provide the following information:

- OS: Debian
- Origin:
  - [x] Package manager: nightly release via Docker Hub
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

------
/cc @weilanad

tgingold, 2025-02-06:
When the ieee library is built, the `.cf`` file and the object files are moved for installation.

So it's possible.  I think you must be sure not to use absolute paths.

Please, post the .cf file.

Paebbels, 2025-02-06:
There are entries like this:
* OSVVM
  ```
  v 4
  file "/builds/GitHub/PLC2/PoC/temp/GHDL/" "../../lib/osvvm/OsvvmContext.vhd" "14d30e255d4b67a7aa71fa190cbf1c94ab845b8d" "20250206180838.782":
    context osvvmcontext at 48( 1695) + 0 on 76;
  file "/builds/GitHub/PLC2/PoC/temp/GHDL/" "../../lib/osvvm/OsvvmSettingsPkg_default.vhd" "6fd8e1b6db1983aa7f7966659111bd8f1d74613f" "20250206180838.559":
    package body osvvmsettingspkg at 49( 1814) + 0 on 74;
  file "/builds/GitHub/PLC2/PoC/temp/GHDL/" "../../lib/osvvm/ReportPkg.vhd" "351d8269ff42d8890f3595a98aee2445e22b53d1" "20250206180838.416":
    package reportpkg at 52( 2178) + 0 on 71 body;
    package body reportpkg at 92( 3584) + 0 on 72;
  ```
* PoC
  ```
  v 4
  file "/builds/GitHub/PLC2/PoC/temp/GHDL/" "../../src/net/arp/arp_IPPool.vhdl" "6a12416f2107c619369575fc4d5ac2e3f470bdf2" "20250206181008.050":
    entity arp_ippool at 32( 1332) + 0 on 399;
    architecture rtl of arp_ippool at 75( 2522) + 0 on 400;
  file "/builds/GitHub/PLC2/PoC/temp/GHDL/" "../../src/net/eth/eth_GEMAC_TX.vhdl" "77c9348186fc5778472eca6d33771f8851c63dd5" "20250206181007.640":
    entity eth_gemac_tx at 32( 1332) + 0 on 395;
    architecture rtl of eth_gemac_tx at 68( 2111) + 0 on 396;
  ```
* IEEE from GHDL's installation:
  ```
  v 4
  file . "../../src/synopsys/v08/std_logic_misc-body.vhdl" "5f8cc78695408981a5b39a90eb6e3c07dc3e964c" "20250206063839.758":
    package body std_logic_misc at 18( 623) + 0 on 50;
  ```

I needed to rename `*.cf` to `*.cf.txt` for GitHub...  
[osvvm-obj08.cf.txt](https://github.com/user-attachments/files/18694380/osvvm-obj08.cf.txt)  
[poc-obj08.cf.txt](https://github.com/user-attachments/files/18694433/poc-obj08.cf.txt)

OSVVM calls GHDL like this:
```
set CurrentWorkingDirectory ../../lib/osvvm
source ../../lib/osvvm/osvvm.pro
library osvvm /builds/GitHub/PLC2/PoC/temp/GHDL/VHDL_LIBS/GHDL-5.0.0-dev
creating directory /builds/GitHub/PLC2/PoC/temp/GHDL/VHDL_LIBS/GHDL-5.0.0-dev/osvvm/v08
analyze IfElsePkg.vhd
ghdl -a --std=08 -Wno-library -Wno-hide --work=osvvm --workdir=/builds/GitHub/PLC2/PoC/temp/GHDL/VHDL_LIBS/GHDL-5.0.0-dev/osvvm/v08 -P/builds/GitHub/PLC2/PoC/temp/GHDL/VHDL_LIBS/GHDL-5.0.0-dev ../../lib/osvvm/IfElsePkg.vhd
analyze OsvvmTypesPkg.vhd
ghdl -a --std=08 -Wno-library -Wno-hide --work=osvvm --workdir=/builds/GitHub/PLC2/PoC/temp/GHDL/VHDL_LIBS/GHDL-5.0.0-dev/osvvm/v08 -P/builds/GitHub/PLC2/PoC/temp/GHDL/VHDL_LIBS/GHDL-5.0.0-dev ../../lib/osvvm/OsvvmTypesPkg.vhd
analyze OsvvmScriptSettingsPkg.vhd
ghdl -a --std=08 -Wno-library -Wno-hide --work=osvvm --workdir=/builds/GitHub/PLC2/PoC/temp/GHDL/VHDL_LIBS/GHDL-5.0.0-dev/osvvm/v08 -P/builds/GitHub/PLC2/PoC/temp/GHDL/VHDL_LIBS/GHDL-5.0.0-dev ../../lib/osvvm/OsvvmScriptSettingsPkg.vhd
analyze OsvvmSettingsPkg.vhd
ghdl -a --std=08 -Wno-library -Wno-hide --work=osvvm --workdir=/builds/GitHub/PLC2/PoC/temp/GHDL/VHDL_LIBS/GHDL-5.0.0-dev/osvvm/v08 -P/builds/GitHub/PLC2/PoC/temp/GHDL/VHDL_LIBS/GHDL-5.0.0-dev ../../lib/osvvm/OsvvmSettingsPkg.vhd
```

Here we see:
* The working directory (`--workdir`) is used with absolute paths.
* The file name (`...vhdl`) is used with relative paths.

I think that's also what we see in the `*.cf` file for arguments 1 and 2, right? I also added the `*.cf` content of the IEEE library. Here, the first argument is `.`.

-----------
/cc @JimLewis

JimLewis, 2025-02-06:
Which package body is reported as missing?  The settings packages analyzed are generated and not saved in the repository.

Paebbels, 2025-02-06:
The settings package body is reported.

JimLewis, 2025-02-06:
Is a missing VHDL source  settings package body making GHDL upset?

JimLewis, 2025-02-06:
Try removing, OsvvmScriptSettingsPkg_generated.vhd, from the .gitignore file.   You will need to check it in once.   But once you do, it should not update unless the settings change.   If you never want the settings to change, even if someone changes the settings, you can copy OsvvmScriptSettingsPkg_generated.vhd to OsvvmScriptSettingsPkg_local.vhd and then OSVVM will use the _local one and not generate one.   If you use the _local one, you need to remove it from the .gitignore also.

JimLewis, 2025-02-06:
You could just generate OsvvmScriptSettingsPkg_generated.vhd and analyze it into the osvvm library, and then run your regressions.   This would allow each CI to have different settings - that is the purpose of package bodies and deferred constants.

JimLewis, 2025-02-06:
Is there a way to tell GHDL here is a compiled library, assume it is up to date, please do not check the VHDL source dates against the library compile time stamps for a particular library?

tgingold, 2025-02-06:
@Paebbels Another possibility (if you cannot use a relative path for workdir) is to modify on the fly the .cf file.  But that's a hack.

@JimLewis, no.  Those are consistency checks.

JimLewis, 2025-02-06:
@tgingold  I get it that those are consistency checks.   I remember all tools used to do those.   However, I wonder if any other than GHDL do them now.

Paebbels, 2025-02-06:
@JimLewis can OSVVM use `cd` to change the directory, thus no `--workdir` is needed. Or alternatively, can a relative `--workdir` be used?

Being able to relocate pre-compiled libraries generated by OSVVM is very essential.

JimLewis, 2025-02-06:
By default, the files will be put in OsvvmLibraries/osvvm.   You just have to edit the .gitignore file as I explained above and check it into your library.   That way GHDL will have the copy you used.     You can even do this before running it in CI and make sure the file gets committed to your library.   

It will still autogenerate the file (unless you rename it to *_local.vhd), but if it is not different, it will not copy the generated file over the existing file (which has the *_generated.vhd name).   

You will need to be able to provide GHDL with file time stamps that are the same as they were when they were compiled.

Paebbels, 2025-02-06:
Modifying `osvvm` is not possible, because it will create a divergent code base (new commit, new hashes, ...). In theory it can be done, but in reality, it's too much of a burden in maintaining a new internal branch and keeping it up to date by manual git operations.

Actually, if such a variation branch scheme is needed, the original library has a bad design. Also the original code base shouldn't be modified (creating a dirty git repository) nor should files be added into the original directory structure.

Additional files, like auto generated source files should be stored in the simulators working directory. Here: `./temp/GHDL/...`. Thus it can be packaged as part of the artifact if needed.

If I got @tgingold answer/question correct, it's not about the missing source file, but about absolute paths in the `*.cf` files caused by OSVVM using absolute paths in `--workdir`. Independent of a potentially missing file, the generated library directories can't be relocated to a different destination because of absolute paths.

JimLewis, 2025-02-07:
@Paebbels  the path issue should be able to be addressed.   I think activehdl needs absolute paths, but Siemens relative paths.   So if ghdl needs relative paths that is just a small update.