ghdl/ghdl #2886: pyGHDL.dom crash with ieee.std_logic_arith

DrPiWork, 2025-02-12:
The following VHDL source file generates an exception when analyzed by pyGHDL :

```vhd :file: test.vhd
library ieee;
use ieee.std_logic_arith.all;

entity Test is
end entity;

architecture rtl of Test is
begin
end;
```

The python script to analyze the VHDL file is this one :
```python :file: pyGHDL_Test.py
from pathlib import Path
from pyGHDL.dom.NonStandard import Design, Document

design = Design()
design.LoadDefaultLibraries()
doc = Document(Path("./test.vhd"))
design.AddDocument(doc, design.GetLibrary("default_lib"))
design.Analyze()
```

The generated exception is this one :
```
Traceback (most recent call last):
  File "C:\Python312\Lib\site-packages\pyVHDLModel\__init__.py", line 1441, in LinkPackageReferences
    package = library._packages[packageIdentifier]
              ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^
KeyError: 'std_logic_arith'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "Z:\Tools\ghdl\Test\PyGHDL_Test.py", line 8, in <module>
    design.Analyze()
  File "C:\Python312\Lib\site-packages\pyGHDL\dom\NonStandard.py", line 131, in Analyze
    super().Analyze()
  File "C:\Python312\Lib\site-packages\pyVHDLModel\__init__.py", line 790, in Analyze
    self.AnalyzeDependencies()
  File "C:\Python312\Lib\site-packages\pyVHDLModel\__init__.py", line 838, in AnalyzeDependencies
    self.LinkPackageReferences()
  File "C:\Python312\Lib\site-packages\pyVHDLModel\__init__.py", line 1445, in LinkPackageReferences
    raise ex
pyVHDLModel.Exception.VHDLModelException: Package 'std_logic_arith' not found in library 'Ieee'.
Caused in design unit 'Entity: 'default_lib.Test(rtl)'' in file 'Document: 'test.vhd''.
```

- OS: Windows 11
- Origin: `pip install pyghdl`

__NOTE__ : This problem raises when using any `ieee` library originating from `synopsys`.

DrPiWork, 2025-02-12:
As a workaround I copied the synopsys file next to the analyzed files and added it to the list of documents.

Paebbels, 2025-02-12:
I release pyVHDLModel v0.30.0, which adds more Synopsys packages. Please update your code as follows until pyGHDL.dom supports `IEEEFlavor` too.

```python
design = Design()
design.LoadDefaultLibraries()
design.GetLibrary("ieee").LoadSynopsysPackages()
...
```

Later, this will be possible:
```python
design = Design()
design.LoadDefaultLibraries(IEEEFlavor.Synopsys)
...
```

DrPiWork, 2025-02-14:
That works. Thanks.

closed 2025-02-16 by tgingold

Paebbels, 2025-02-16:
@DrPiWork / @NicoPy  
I pushed a new nightly (needs ~45 minutes), so you can test it soon.

NicoPy, 2025-02-16:
Great !
I'll try it tomorrow.

DrPiWork, 2025-02-17:
Works perfectly.
+1

PR #2893 (2025-02-16): Added IEEE flavor option when setting up a design in pyGHDL.dom
# New Features

* Added an option to allow setting up different IEEE flavors when setting up a design with `pyGHDL.dom`.  
  `design.LoadDefaultLibraries(IEEEFlavor.Synopsys)`

# Changes
* Replace local implementations of `firstValue` by a common implementation from `pyTooling.Common`
* Bumped dependencies: pyVHDLModel to v0.30.0.

# Related Issues or PRs
* Fixes #2886
* Fixes #2887.