ghdl/ghdl #2889: Missing version information in builds for MSYS2/*

Paebbels, 2025-02-15:
**Description**
When installing nightly releases, at least some assets report no "correct" version, e.g. the Git hash.

**Expected behavior**
Binaries build from Git (locally, via CI as nightly, or real release) should contain the Git hash and maybe other information to ease error reporting and checking if GHDL can be updated.

I have the following rules in mind:
* If GHDL was downloaded as a ZIP file from GitHub  
  (or after a Git clone/checkout the `.git` directory was removed,  
   or if Git program is missing):  
  &rarr; It should report `tarball`.
* If a Git environment is present, useful version information from Git should be visible in the binaries version output.  
  &rarr; Git hash  
  &rarr; Commits since last tag  
  &rarr; Git reference (branch or tag)
* If the build is a nightly build, it should be noted too.

**How to reproduce?**
* Ubuntu 24.04:
  ```
  GHDL 5.0.0-dev (4.1.0.r540.g8e0a02a46) [Dunoon edition]
   Compiled with GNAT Version: 13.3.0
   static elaboration, mcode JIT code generator
  Written by Tristan Gingold.
  ```
  Pipeline: https://github.com/Paebbels/ghdl/actions/runs/13229282625/job/36924211113#step:10:14
* macOS-14:
  ```
  GHDL 5.0.0-dev (4.1.0.r540.g8e0a02a46) [Dunoon edition]
   Compiled with GNAT Version: 14.2.0
   llvm 19.1.7 code generator
  Written by Tristan Gingold.
  ```
  Pipeline: https://github.com/Paebbels/ghdl/actions/runs/13229282625/job/36924222254#step:10:14
* MinGW64:  
  ```
  GHDL 5.0.0-dev (tarball) [Dunoon edition]
   Compiled with GNAT Version: 14.2.0
   static elaboration, mcode JIT code generator
  Written by Tristan Gingold.
  ```
  Pipeline: https://github.com/Paebbels/ghdl/actions/runs/13229282625/job/36924258540#step:10:15
* UCRT64:  
  ```
  GHDL 5.0.0-dev (tarball) [Dunoon edition]
   Compiled with GNAT Version: 14.2.0
   static elaboration, mcode JIT code generator
  Written by Tristan Gingold.
  ```
  Pipeline: https://github.com/Paebbels/ghdl/actions/runs/13229282625/job/36924263125#step:10:15

As a conclusions:
1. It seams to be limited to builds in MSYS2
   Is this caused by rules in `make`?  
   Could is be caused by a missing Git executable in the MSYS2 environment?  
2. The version information lists since years (or a decade?) `Dunoon edition`.  
   What is this name about?  
   Are all releases we had and will have part of this edition?
3. In case of nightly releases, could this be marked in the version output?  
   Current: `GHDL 5.0.0-dev (4.1.0.r540.g8e0a02a46) [Dunoon edition]`  
   Proposals:  
   &rarr; `GHDL 5.0.0`    
   &rarr; `GHDL 5.0.0-dev (4.1.0.r540.g8e0a02a46 - master)`
   &rarr; `GHDL 5.0.0-dev (4.1.0.r540.g8e0a02a46 - nightly)`

**Context**
Please, provide the following information:

- OS: MSYS2/*
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: 8e0a02a46

**Additional context**
*None*

tgingold, 2025-02-15:
The problem is:
```
 /bin/sh: line 4: git: command not found
```

tgingold, 2025-02-15:
However, there is:
```
 "C:\Program Files\Git\bin\git.exe" version
```
So maybe git must be added to PATH.

Paebbels, 2025-02-15:
I'll provide a PR for it.

I finished compiling it locally an my PC in MSYS2 and here the information is available.

The reason is, we're inside of the MSYS2 environment and almost non of the external programs of the native Windows is visible to MSYS2 via PATH. That's on one because there is a MinGW64 and UCRT64 environment with different system bindings.

OTOH, I now noticed, the Git checkout action is running outside of MSYS2 on the native Windows' Git installation, the one you found. Aber then the make is inside of MSYS2. I just need to add Git to the MSYS2 dependencies for pacboy.

PR #2890 (2025-02-15): Install Git in MSYS2, so binaries contain Git version information.
This PR fixes the missing version information in MSYS2/* builds by installing Git in MSYS2.

Fixes #2889.
closed 2025-02-15 by tgingold