ghdl/ghdl #2892: GHDL should provide a `setup-pyGHDL` action or publish to PyPI for pyGHDL releases

Paebbels, 2025-02-16:
This issue is an idea collection and discussion.

**Is your feature request related to a problem? Please describe.**
Currently the setup of pyGHDL is quite complicated because the pyGHDL and libghdl version must match each other.
Nightly releases are providing latest platform specific `*.whl` files including all necessary libghdl files.

Unfortunately, these can't be installed easily via `pip`.
1. Download the wheel file
2. Install manually

**Describe the solution you'd like**
1. Elaborate if a `setup-pyGHDL` action similar to `setup-GHDL` could support users in GitHub action environments.
2. Elaborate if releases and nightly builds could be published to PyPI, so users can easily download/install AND upgrade pyGHDL.  
   * This process could be integrated into the release pipeline, because all platform specific files exist as artifacts and assets.  
   * It also obsoletes the need for `setup-pyGHDL` and maintaining another repository / script.
   * The only missing puzzle piece is a versioning scheme for nightly releases as Python and PyPI strictly require semantic versioning and a clean incremental pattern.

**Proposals for version numbers:**
> [**PEP 440 – Version Identification and Dependency Specification**](https://peps.python.org/pep-0440/)  
> 
> [...]  
> 
> Public version identifiers are separated into up to five segments:
> 
> * Epoch segment: N!
> * Release segment: N(.N)*
> * Pre-release segment: {a|b|rc}N
> * Post-release segment: .postN
> * Development release segment: .devN
>
> Any given release will be a “final release”, “pre-release”, “post-release” or “developmental release” as defined in the following sections.
> 
> [...]  

Lets list some possibilities:
1. GHDL usually uses major and minor numbers like `3.0.0`, `4.0.0`, `4.1.0`. The micro (patch) number is unused.  
   This number could be used by pyGHDL as an incrementing pattern.
2. Python has a numbering scheme for development releases: `X.Y.Z.devN`, where `N` is a positive integer.  
   The PEP has this note:
   > **Note**
   > 
   > While they may be useful for continuous integration purposes, publishing developmental releases of pre-releases to general purpose public index servers is strongly discouraged, as it makes the version identifier difficult to parse for human readers. If such a release needs to be published, it is substantially clearer to instead create a new pre-release by incrementing the numeric component.

&rArr; To me it sounds like, variant 1 is preferred over variant 2.

How to increment that release number?
GitHub offers an API to read and update Action variables (like secrets). The algorithm could be as follows:
1. every nightly release reads the GitHub variable and increments it by 1
2. the incremented number is updated in GitHub Action, when the publishing to PyPI was successful, otherwise the variable stays unmodified to allow pipeline reruns.
3. If a full release is triggered (e.g. 5.0.0), the counter is cleared to 0

**Describe alternatives you've considered**
* A series of `curl` and `jq` could be used. Ideally abstracted by a GitHub Action, but this doesn't solve non-CI installations.

**Additional context**
*None*

Paebbels, 2025-02-16:
@tgingold should I work on a proposal to have pyGHDL published to PyPI using the described method?

tgingold, 2025-02-16:
Simply use the date YYYYMMDD as development number.  If needed, add HHMM.

Paebbels, 2025-02-16:
I'm not sure if PyPI accepts such a long number, we would need to test it.

Do you have a PyPI account?

tgingold, 2025-02-16:
No, but I can create one.