ghdl/ghdl #2894: pyGHDL doesn't recognize Record Subtype Indication? Missing feature or skill issue?

fisherdog1, 2025-02-16:
I call AddDocument on a package-containing file first, then on a file which uses a record type defined in that package. I get the following exception:
```
pyGHDL.dom.DOMException: Unknown kind 'Record_Subtype_Definition' for an subtype indication in a signal of `regdata`.
```
Offending declaration:
```
signal regdata : reg_state_t(field_states(BIT_FIELDS'range)) 
	:= get_reg_initial_value(BIT_FIELDS, RESET_VALUE);
```
reg_state_t is declared in the package and I have the appropriate library and use clause. Walking through some of what is generated, it seems it does parse the reg_state_t subtype correctly.

It seems like this is a missing feature but I cannot be sure. It's very possible I'm just adding files wrong. If I can rule that out I will take an initial crack at supporting this.

**Misc Info**

My test script:
```
import sys
sys.path.append("/home/azimuth/ghdl")

from loguru import logger
logger.add(sys.stderr, backtrace=True, diagnose=True)

import code
from pathlib import Path
import pyGHDL
from pyGHDL.dom.NonStandard import Design, Document, DOMException

sourceFiles = [
  Path("./regoop/regoop.vhd"), # reg_state_t defined in here
  Path("./regoop/regoop_reg.vhd")] # reg_state_t used here, no error if this file is not parsed

design = Design()
design.LoadDefaultLibraries()
library = design.GetLibrary("work")

try:
  for sourceFile in sourceFiles:
    design.AddDocument(Document(sourceFile), library)
except DOMException as e:
  logger.exception("DOMException")
```

Output formatted by loguru
```
__main__:<module>:25 - DOMException
Traceback (most recent call last):

> File "/home/azimuth/testvhd/test.py", line 23, in <module>
    design.AddDocument(Document(sourceFile), library)
    │      │           │        │            └ Library: 'work'
    │      │           │        └ PosixPath('regoop/regoop_reg.vhd')
    │      │           └ <class 'pyGHDL.dom.NonStandard.Document'>
    │      └ <function Design.AddDocument at 0x7fe1635d8820>
    └ <unprintable Design object>

  File "/home/azimuth/ghdl/pyGHDL/dom/NonStandard.py", line 178, in __init__
    self.translate()
    │    └ <function Document.translate at 0x7fe16335dcf0>
    └ Document: 'regoop/regoop_reg.vhd'

  File "/home/azimuth/ghdl/pyGHDL/dom/NonStandard.py", line 233, in translate
    architecture = Architecture.parse(libraryUnit, contextItems)
                   │            │     │            └ []
                   │            │     └ 2354
                   │            └ <classmethod(<function Architecture.parse at 0x7fe16335d3f0>)>
                   └ <class 'pyGHDL.dom.DesignUnit.Architecture'>

  File "/home/azimuth/ghdl/pyGHDL/dom/DesignUnit.py", line 187, in parse
    return cls(architectureNode, name, entitySymbol, contextItems, declaredItems, statements, documentation)
           │   │                 │     │             │             │              │           └ ''
           │   │                 │     │             │             │              └ <generator object GetConcurrentStatementsFromChainedNodes at 0x7fe1633409e0>
           │   │                 │     │             │             └ <generator object GetDeclaredItemsFromChainedNodes at 0x7fe163340c10>
           │   │                 │     │             └ []
           │   │                 │     └ EntitySymbol: 'regoop_reg' -> unresolved
           │   │                 └ 'rtl'
           │   └ 2354
           └ <class 'pyGHDL.dom.DesignUnit.Architecture'>

  File "/home/azimuth/ghdl/pyGHDL/dom/DesignUnit.py", line 169, in __init__
    super().__init__(identifier, entity, contextItems, declaredItems, statements, documentation)
                     │           │       │             │              │           └ ''
                     │           │       │             │              └ <generator object GetConcurrentStatementsFromChainedNodes at 0x7fe1633409e0>
                     │           │       │             └ <generator object GetDeclaredItemsFromChainedNodes at 0x7fe163340c10>
                     │           │       └ []
                     │           └ EntitySymbol: 'regoop_reg' -> unresolved
                     └ 'rtl'

  File "/home/azimuth/.local/lib/python3.10/site-packages/pyVHDLModel/DesignUnit.py", line 513, in __init__
    ConcurrentDeclarations.__init__(self, declaredItems)
    │                      │        │     └ <generator object GetDeclaredItemsFromChainedNodes at 0x7fe163340c10>
    │                      │        └ <unprintable Architecture object>
    │                      └ <function ConcurrentDeclarations.__init__ at 0x7fe16359fbe0>
    └ <class 'pyVHDLModel.Concurrent.ConcurrentDeclarations'>
  File "/home/azimuth/.local/lib/python3.10/site-packages/pyVHDLModel/Concurrent.py", line 246, in __init__
    for item in declaredItems:
                └ <generator object GetDeclaredItemsFromChainedNodes at 0x7fe163340c10>

  File "/home/azimuth/ghdl/pyGHDL/dom/_Translate.py", line 805, in GetDeclaredItemsFromChainedNodes
    yield objectParseMethod(objectItem, furtherIdentifiers)
          │                 │           └ []
          │                 └ 2356
          └ <bound method Signal.parse of <class 'pyGHDL.dom.Object.Signal'>>

  File "/home/azimuth/ghdl/pyGHDL/dom/Object.py", line 188, in parse
    subtypeIndication = GetSubtypeIndicationFromNode(signalNode, "signal", name)
                        │                            │                     └ 'regdata'
                        │                            └ 2356
                        └ <function GetSubtypeIndicationFromNode at 0x7fe16335c4c0>

  File "/home/azimuth/ghdl/pyGHDL/dom/_Translate.py", line 301, in GetSubtypeIndicationFromNode
    return GetSubtypeIndicationFromIndicationNode(subtypeIndicationNode, entity, name)
           │                                      │                      │       └ 'regdata'
           │                                      │                      └ 'signal'
           │                                      └ 2366
           └ <function GetSubtypeIndicationFromIndicationNode at 0x7fe16335c550>

  File "/home/azimuth/ghdl/pyGHDL/dom/_Translate.py", line 327, in GetSubtypeIndicationFromIndicationNode
    raise DOMException(f"Unknown kind '{kind.name}' for an subtype indication in a {entity} of `{name}`.")
          └ <class 'pyGHDL.dom.DOMException'>

pyGHDL.dom.DOMException: Unknown kind 'Record_Subtype_Definition' for an subtype indication in a signal of `regdata`.
```
Just in case, the vhdl files are [here](https://gitlab.com/tough-steves-house-of-criminal-intent/regoop).

Paebbels, 2025-02-16:
GHDL can handle this, but the transformation from GHDL's IIR to pyVHDLModel is not yet complete.

It's on my agenda to fix it, so more features used in source files are supported.  
Any help in proposing transformations are welcome. (I know it's not easy to get into that topic, as it needs understanding of GHDL's IIR and pyVHDLModel's hierarchy.) But in case you want to support us/me, I'll help you understand details.

-----
Please do not call your libraries `work`, as this is a special name in VHDL.  
`library = design.GetLibrary("work")`

fisherdog1, 2025-02-16:
Good catch, I have this bad habit from Vivado. I'll return with questions.

Paebbels, 2025-02-16:
It's similar to #2779.

fisherdog1, 2025-02-17:
I made a tiny bit of progress here: https://github.com/ghdl/ghdl/commit/0f23aba5dce62bf298ef37c871330bfb68f44398
That at least gets me to where I can pretty print using dom.py
I may be able to figure out record subtypes based on how it's gone so far. Are there any major chasms I should avoid falling down?

fisherdog1, 2025-02-18:
What's the difference between ghdl/pyGHDL/dom/Symbol and pyVHDLModel/Symbol. Does the former just import classes from the latter? Implementing what I infer should be named "GetRecordConstraintsFromSubtypeIndication"

fisherdog1, 2025-02-18:
Figured previous comment out. Stopping on this question for today:
https://github.com/fisherdog1/ghdl/commit/23c63b7eb56d842cd7cae51da61622032ea84413#r152666921

Paebbels, 2025-02-19:
pyVHDLModel is "just" an object model describing how language elements are related to each other to compose an instance of a VHDL document. It's like a DOM for XML (or HTML), hence we call it a CodeDOM. It has no parser, so other frontends can generate an instance of pyVHDLModel.

One of these potential frontends is pyGHDL.dom. So it uses pyGHDL.libghdl to call into libghdl...so/dll, traverses GHDL's IIR and generates the model instance.

pyVHDLModel defines a `Symbol` class. pyGHDL.dom derives from that symbol and adds own and addition fields. One added field is usually the IIR node ID and/or the position in the source code document.
E.g. the name `myEntity` of an entity declaration like `entity myEntity is ...` is an identifier, while the reference to that entity in an architecture like `architecture rtl of myEntity is ...` is a symbol. At the beginning of parsing, it's a name referring to another item. When the symbol was resolved, it will point via a pointer directly to the pyVHDLModel's object, so no need to recursively search for it by a string. Symbols print an additional `?` when not yet resolved.

-----------------
`use ieee.std_logic_1164.all, ieee.textio.all;`

This use-clause (`UseClause`) is made of a list of symbols (`PackageReferenceSymbol`), the symbols itself are made of a chain of names, which refer to other elements in the model.

In record constraints or expressions, symbols are referring either to record elements or to constants, signals, functions, etc.

Paebbels, 2025-02-27:
I pushed this branch: https://github.com/Paebbels/ghdl/tree/refs/heads/paebbels/pyGHDL-record

It's not perfect, but is solves 2 big issues:
1. the code doesn't print on unimplemented items. It uses a new experimental feature from pyTooling to collect warnings.
2. all VHDL files like `all08.vhdl` delivered with GHDL can be read without an exception.

fisherdog1, 2025-02-27:
I will take a look at this. This test file is also very good to know about.

Paebbels, 2025-03-08:
@fisherdog1 is this issue solved?

fisherdog1, 2025-03-08:
Let me test it this weekend.

fisherdog1, 2025-03-08:
This works, and I have cherry-picked some code I used to [pretty-print the new feature.](https://github.com/fisherdog1/ghdl/commit/5a6dfa6fece5ac754249fd4cf9f95200d6f4aa66)

Paebbels, 2025-03-08:
Then I'll close this issue for now.

I'll check the commit you mentioned.

closed 2025-03-08 by Paebbels