ghdl/ghdl #2895: Drop support for some precompile scripts

Paebbels, 2025-02-16:
I would like to drop the support for some precompile scripts:
* **Xilinx ISE**  
  The last release of Xilinx ISE is from September 2013, thus 11 years ago.
* **OSVVM**  
  1. OSVVM now delivered with its own scripting environment, thus it can be easily pre-compiled.
  2. The OSVVM pro files have become very complicated using lots of embedded TCL instructions. Therefore, these files can't be parsed by Bash files.
  
  /cc @JimLewis

Any objections for these two libraries?

Needs further checks:
* [ ] UVVM
* [ ] Altera
* [ ] Intel (might be dropped too, as Intel PSG got renamed to Altera again)
* [ ] Lattice

JimLewis, 2025-02-17:
When GHDL is delivered, is there currently a copy with pre-compiled OSVVM libraries?    If yes, I can probably provide a set of scripts that should be bash compatible - just ghdl commands.  It is probably past time that I verify that capability anyway.

Paebbels, 2025-02-17:
It only ships with a script, no sources or pre-compiled files.

JimLewis, 2025-02-17:
OSVVM scripts may be sufficient.   However, it may be a good idea to tune up the OSVVM capability to generate the scripts.

Paebbels, 2025-03-15:
Xilinx ISE and OSVVM has been dropped.

OSVVM itself provides powerful scripting to generate all necessary files for GHDL.

closed 2025-03-15 by Paebbels