ghdl/ghdl #2900: macOS-15 + GNAT 14.2.0

Paebbels, 2025-02-17:
**Description**
Compiling GHDL using GNAT 14.2.0 on macOS-15 fails.

The current investigation indicates 2 options:
* The GNAT compiler from https://github.com/alire-project/GNAT-FSF-builds doesn't contain a definition of `FILE*` in `cstdio.h`.
* The GNAT compiler's `cstdio.h` don't contain `FILE*` itself and references a `_stdio.h`. Maybe this file is provided by the operating system, which has changed from macOS-14 to macOS-15.

```
gcc -c  -g -Wall -g -o grt-cstdio.o /Users/runner/work/ghdl/ghdl/build/llvm/../../src/grt/grt-cstdio.c
In file included from /Users/runner/work/ghdl/ghdl/build/llvm/../../src/grt/grt-cstdio.c:17:
/Users/runner/work/ghdl/ghdl/gnat/lib/gcc/aarch64-apple-darwin23.6.0/14.2.0/include-fixed/stdio.h:83:8: error: unknown type name 'FILE'
   83 | extern FILE *__stdinp;
      |        ^~~~
/Users/runner/work/ghdl/ghdl/gnat/lib/gcc/aarch64-apple-darwin23.6.0/14.2.0/include-fixed/stdio.h:81:1: note: 'FILE' is defined in header '<stdio.h>'; this is probably fixable by adding '#include <stdio.h>'
   80 | #include <sys/_types/_seek_set.h>
  +++ |+#include <stdio.h>
   81 | 
```

**Expected behaviour**
GHDL should also compile on macOS-15.

**How to reproduce?**
It can be reproduced by activating macOS-15 builds in the CI pipeline.

**Context**
Please, provide the following information:

- OS: macOS-15 
- GNAT: 14.2.0 (prebuild from https://github.com/alire-project/GNAT-FSF-builds)

**Additional context**
*None*

Paebbels, 2025-02-17:
@tgingold do we need to report this at **alire-project/GNAT-FSF-builds** or **actions/runner-images**?

Paebbels, 2025-02-18:
Is this issue solved by the change made in 47c5f8f0dfba4a946d25a78f6f51b822dc084fec ?

tgingold, 2025-02-18:
Yes.

closed 2025-02-18 by tgingold

quigleyj97, 2025-03-10:
Does this change need to be made to the build-from-source instructions as well?

https://ghdl.github.io/ghdl/development/building/index.html

I wasn't able to build from source locally until I added that clang override to my environment before running configure, eg `CC=clang ../configure --local-prefix=/usr/local`