ghdl/ghdl #2904: Feature Synth: Expand multiple levels of records into structured port names

JJ-Gaisler, 2025-02-19:
When GHDL synthesizes ports with complex types into Verilog these typically end up becoming a packed array of bits.
My main pain point is having to unpack these bits manually which is quite error prone.

GHDL already handles this problem for simpler types (records with std_logic(_vector)) where it takes the port name and then appends the field name. I would like to expand this so it works for more than one level of records/arrays.

Consider the following:
```vhdl
  type axi_mosi_type is record
    aw  : axi_aw_mosi_type;
    w   : axi_w_mosi_type;
    b   : axi_b_mosi_type;
    ar  : axi_ar_mosi_type;
    r   : axi_r_mosi_type;
  end record;

```
```vhdl
  type axi_aw_mosi_type is record
    id     : std_logic_vector (AXI_ID_WIDTH-1 downto 0);  -- awid
    addr   : std_logic_vector (31 downto 0);              -- awaddr
    len    : std_logic_vector (3 downto 0);               -- awlen
    size   : std_logic_vector (2 downto 0);               -- awsize
    burst  : std_logic_vector (1 downto 0);               -- awburst
    lock   : std_logic_vector (1 downto 0);               -- awlock
    cache  : std_logic_vector (3 downto 0);               -- awcache
    prot   : std_logic_vector (2 downto 0);               -- awprot
    valid  : std_logic;                                   -- awvalid
  end record;
```

If I were to create a port of type axi_somi_type on my vhdl top level once synthesized into Verilog this is converted into a packed array of bits. This makes it rather difficult to interact with the port since all bits have to be unpacked manually.

If I instead create the aw, w, b, ar and r ports individually they are turned into multiple signals with reasonable names. This allows me to use macros to easily handle the port and no unpacking needs to be done. Port names would become:
```
aw_id
aw_addr
aw_len
...
<port>_<field>
```

What I would like to see is for GHDL to do what it already does for a single level and apply that to multiple levels of structs.
I.e. a port `axi : out axi_mosi_type` would be expanded to:
```
axi_aw_addr
axi_aw...
axi_w_data
...
axi_r_ready
<port>_<field>_<field>
````

Furthermore if an array is used on the top level the array dimension could also be included in the name:
```
axi_0_aw_addr
<port>_<idx>_<field>_<field>...
```
A multidim array could become:
```
<port>_<idx_dim0>_<idx_dim1>_<field>
```

It would be enough to expand names like this one the top level since that is where all the interaction is done.

tgingold, 2025-02-23:
Yes, this looks a reasonable feature...

tgingold, 2026-04-03:
Indeed, this is now implemented.
Do we need another issue for signals ?

closed 2026-04-03 by tgingold