ghdl/ghdl #2914: Reference signals in submodules in PSL VUnit

efykse, 2025-03-05:
**Is your feature request related to a problem? Please describe.**
When working with PSL VUnit files for formal verification, I often need a way to write assertions and assumptions about signals inside VHDL blocks or even submodules.

**Describe the solution you'd like**
Is there support for this already? Or could GHDL support some way to reference these signals, for example:
```
vunit formal_vunit (my_module(rtl)) {
    assert always block_label.my_signal_b = '0';
    assert always submodule_instance_label.my_signal_a = '0';
}
```

**Describe alternatives you've considered**
Would an alternative be to be able to bind VUnits to submodules or blocks? I understand this should be supported by the language? For this to work I would need several VUnits at the same time for the formal verification proof.

```
vunit formal_vunit (my_module(rtl).my_vhdl_block) {
```

**Additional context**
Any other solution would be great, as long as it allows to reference the signals further down in the hierarchy in some way.

efykse, 2025-03-05:
I tried a few different ways to access signals inside a block, but unfortunately I could not get it to work. Maybe I'm missing something or doing something wrong? (b_block is a VHDL block inside my_module)

Here is what I tried with the error message written above each attempt:
```vhdl
vunit formal_test (my_module(rtl)) {
    default clock is rising_edge(clk_i);

    -- raised ADA.ASSERTIONS.ASSERTION_ERROR : vhdl-sem_decls.adb:72
    signal fsm_rdy_tb : std_logic;
    fsm_rdy_tb <= << signal b_block.fsm_rdy : std_logic >>;

    -- raised ADA.ASSERTIONS.ASSERTION_ERROR : vhdl-sem_decls.adb:72
    assert always << signal b_block.fsm_rdy : std_logic >> = '0';

    -- exec_pathname_concurrent_region: cannot handle IIR_KIND_VUNIT_DECLARATION
    -- raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
    alias fsm_rdy_tb is << signal b_block.fsm_rdy : std_logic >>;

    -- an expanded name is only allowed within the construct
    alias fsm_rdy_tb is b_block.fsm_rdy;
}
```

tgingold, 2025-03-09:
I can investigate.  Do you have a small reproducer ?

efykse, 2025-03-10:
Sure! Thanks for taking a look.

module.vhd:
```vhd :file: module.vhd
library ieee;
use ieee.std_logic_1164.all;

entity module is
   port (
      clk_i    : in  std_logic;
      input_i  : in  std_logic;
      output_o : out std_logic
      );
end entity module;

architecture rtl of module is
begin
   b_block : block
      signal my_signal : std_logic;
   begin
      process(clk_i)
      begin
         if rising_edge(clk_i) then
            my_signal <= input_i;
         end if;
      end process;
      output_o <= my_signal;
   end block;
end architecture;
```
vunit.psl (I included 3 different ways of achieving what I want. But only need one)
```vhd :file: vunit.psl
vunit formal_module (module(rtl)) {
    default clock is rising_edge(clk_i);

    alias a_my_signal is << signal b_block.my_signal : std_logic >>;
    assert always a_my_signal = '0';

    assert always << signal b_block.my_signal : std_logic >> = '0';

    assert always b_block.my_signal = '0';
}
```

Command:
```
# ghdl --synth --std=08 vunit.psl module.vhd -e module
```

commit 90b2af64 (2025-03-13): vhdl: handle external names in vunit

Fix #2914
closed 2025-03-13 by tgingold